openwrt/openwrt #12048: openssl fails to build on powerpc targets

MoetaYuko, 2023-02-22:
### Describe the bug

build fails after upgrading to 3.0:

https://downloads.openwrt.org/snapshots/faillogs/powerpc_464fp/base/openssl/
https://downloads.openwrt.org/snapshots/faillogs/powerpc_8548/base/openssl/

cc: @cotequeiroz 

### OpenWrt version

snapshot

### OpenWrt target/subtarget

powerpc

### Device

official sdk

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

cotequeiroz, 2023-02-22:
Hi. Thanks for the report. I mixed up Linux arch with OpenWrt `s/ppc/powerpc/`.

 I’m running a build for ppc at the moment—in case there’s something else causing a failure—and will send a fix in about 9 hours. Arc is not building either, and I finished testing the fix for it already.

pkgadd, 2023-02-22:
https://github.com/openwrt/openwrt/pull/12046

PR #12049 (2023-02-22): openssl: openssl: fix powerpc & arc libatomic dependencies ❤️ 
PowerPC had ppc instead of powerpc, when selecting libatomic. Arc needs to be built with libatomic as well.

Fixes #12048 

This was compile-tested for arc_archs and powerpc_464fp

Ping @Ansuel
closed 2023-02-22 by openwrt-bot