openwrt/openwrt #12082: ubnt_unifi-ap-lr: Bad magic number

selfhoster1312, 2023-02-25:
### Describe the bug

I have a few [Unifi AP-LR](https://openwrt.org/toh/ubiquiti/unifi_ap) running OEM firmware which is a nightmare to reconfigure. I wanted to flash one with OpenWRT as i have so far only had very good experiences with OpenWRT-supported devices.

This specific model doesn't have locked bootloader, and boots OpenWRT with some workarounds, but wifi then doesn't work.

The "different MTD partition scheme" workaround documented on the wiki, with factory.bin, refuses to boot with `Bad Magic Number`:

<details>

```
U-Boot unifi-v1.6.17.296-g1af7670c (Apr 22 2019 - 11:05:55)

DRAM:  64 MB
Base:0x80000000, Top:0x84000000, Res logbuf:0xa3ff3800, log_magic:0x5a5a5a5a kseg: 0xa0000000
Flash:  8 MB                                                                    
PCIe WLAN Module found (tries: 1).                                              
Net:   eth0, eth1                                                               
Board: Copyright Ubiquiti Networks Inc. 2014                                    
Hit any key to stop autoboot:  0                                                
ar7240> tftpboot 0x80000000 openwrt-ath79-generic-ubnt_unifi-ap-lr-squashfs-factory.bin 
Using eth0 device                                                               
TFTP from server 192.168.42.2; our IP address is 192.168.42.253                 
Filename 'openwrt-ath79-generic-ubnt_unifi-ap-lr-squashfs-factory.bin'.         
Load address: 0x80000000                                                        
Loading: #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         #################################################################      
         ###################################                                    
done                                                                            
Bytes transferred = 5833116 (59019cerase 0x9f050000 +0x00760000                 
...............................................................................e
Erased 90 sectors
ar7240> cp.b 0x80000000 0x9f050000 0x59019c                                       
Copy to Flash... write addr: 9f050000                                           
done
ar7240> reset                                                                     
                                                                                
Resetting...                                                                    
                                                                                
U-Boot unifi-v1.6.17.296-g1af7670c (Apr 22 2019 - 11:05:55)                     
                                                                                
DRAM:  64 MB                                                                    
Base:0x80000000, Top:0x84000000, Res logbuf:0xa3ff3800, log_magic:0x5a5a5a5a ks0
Flash:  8 MB                                                                    
PCIe WLAN Module found (tries: 1).                                              
Net:   eth0, eth1                                                               
Board: Copyright Ubiquiti Networks Inc. 2014                                    
Hit any key to stop autoboot:  0                                                
Board: Ubiquiti Networks AR7241 board (e512-20.0101.002e)                       
 0. Name = u-boot, offset = 0, start_addr=9f000000, size=262144,start_sector=0, 
 1. Name = u-boot-env, offset = 40000, start_addr=9f040000, size=65536,start_se 
 2. Name = kernel, offset = 50000, start_addr=9f050000, size=7733248,start_sect 
 3. Name = cfg, offset = 7b0000, start_addr=9f7b0000, size=262144,start_sector= 
 4. Name = EEPROM, offset = 7f0000, start_addr=9f7f0000, size=65536,start_secto 
UBNT application initialized                                                    
[... 218 lines cut ...]
These are the results with the latest development builds for the AP-LR variant, but i've also obtained the same results with the 22.03 release [listed on the wiki](https://openwrt.org/toh/ubiquiti/unifi_ap#installation).

So i'm wondering if maybe there should be two different images for the different partition schemes? Or if there's something i can do to make wifi work from sysupgrade image?

I'm willing to help and learn but i've never done OS/firmware development so i'm going to need a lot of guidance. I'm OK if someone else wants to fix this for me. Let me know how i can help.

### OpenWrt version

r22130-524704e677

### OpenWrt target/subtarget

ath79/generic

### Device

Ubiquiti UniFi AP LR

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

selfhoster1312, 2023-02-26:
I did more tests, it appears the error is random. Sometimes wifi actually gets initialized. Out of 16 reboots, i had: 6 working, 9 not working, and one kernel panic.

The kernel panic looked like this:

```
[   15.951690] Loading modules backported from Linux version v6.1-rc8-0-g76dcd734eca2
[   15.959430] Backport generated by backports.git v5.15.81-1-41-g02e352527db5
[   16.008434] urngd: v1.0.2 started.
[   17.045479] PPP generic driver version 2.4.2
[   17.067287] NET: Registered PF_PPPOX protocol family
[   17.214298] ath: phy0: Ignoring endianness difference in EEPROM magic bytes.
[   18.207615] Data bus error, epc == 803160a8, ra == 81981ce0
[   18.213258] Oops[#1]:
[   18.215548] CPU: 0 PID: 558 Comm: kmodloader Not tainted 5.15.94 #0
[   18.221868] $ 0   : 00000000 00000001 deadc0de 81981cac
[   18.227160] $ 4   : b0008048 00008048 00000000 00000020
[   18.232453] $ 8   : 81bd0690 00000000 000000ff bf43c5e8
[   18.237746] $12   : 00000000 000025cc 00000000 807be098
[   18.243039] $16   : 00000020 00000000 00008048 81a6d6c0
[   18.248332] $20   : 81800964 00023f60 00000006 00000001
[   18.253624] $24   : 00000018 81981d1c                  
[   18.258909] $28   : 81a14000 81a15a20 81a700a8 81981ce0
[   18.264203] Hi    : 0000052f
[   18.267098] Lo    : 1d360000
[   18.269992] epc   : 803160a8 ioread32+0xc/0x14
[   18.274518] ra    : 81981ce0 ath_start_rfkill_poll+0x1e0/0x66c [ath9k]
[   18.281338] Status: 1000d403 KERNEL EXL IE 
[   18.285566] Cause : 1080001c (ExcCode 07)
[   18.289604] PrId  : 00019374 (MIPS 24Kc)
[   18.293554] Modules linked in: ath9k(+) ath9k_common pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet ath9k_hw ath ppc
[   18.364884] Process kmodloader (pid: 558, threadinfo=(ptrval), task=(ptrval), tls=77ec5df0)
[   18.373290] Stack : 81a7029c 81b9ee88 00000000 00600000 000186a0 81a70080 00000000 81a704e4
[   18.381757]         00000000 81b8281c 00000000 81800964 00023f60 81a70080 81a70080 81b88424
[   18.390224]         00000000 81a704e4 18810c78 b8262626 26000000 00000000 20000000 00000000
[   18.398691]         00000000 00000011 0c8964f6 ac262624 00000000 00000011 0c8cd07d 00000000
[   18.407149]         81a70080 807b0000 81a6c480 81a70080 81a70164 00000000 81a700a8 00000004
[   18.415616]         ...
[   18.418085] Call Trace:
[   18.420543] [<803160a8>] ioread32+0xc/0x14
[   18.424699] [<81981ce0>] ath_start_rfkill_poll+0x1e0/0x66c [ath9k]
[   18.431199] 
[   18.432699] Code: 0000000f  8c820000  0000000f <03e00008> 00000000  0000000f  8c820000  0000000f  7c0210a0 
[   18.442579] 
[   18.444131] ---[ end trace 69efab1077a51179 ]---
[   18.448798] Kernel panic - not syncing: Fatal exception
[   18.454060] Rebooting in 1 seconds..
```

This is a working boot with debug level 4:

<details>

```

U-Boot unifi-v1.6.17.296-g1af7670c (Apr 22 2019 - 11:05:55)

DRAM:  64 MB
Base:0x80000000, Top:0x84000000, Res logbuf:0xa3ff3800, log_magic:0x03e00008 kseg: 0xa0000000
Flash:  8 MB
PCIe WLAN Module found (tries: 1). 
Net:   eth0, eth1
Board: Copyright Ubiquiti Networks Inc. 2014
Hit any key to stop autoboot:  0 
Board: Ubiquiti Networks AR7241 board (e512-20.0101.002e)
 0. Name = u-boot, offset = 0, start_addr=9f000000, size=262144,start_sector=0, end_sector=3 
 1. Name = u-boot-env, offset = 40000, start_addr=9f040000, size=65536,start_sector=4, end_sector=4 
 2. Name = kernel, offset = 50000, start_addr=9f050000, size=7733248,start_sector=5, end_sector=122 
 3. Name = cfg, offset = 7b0000, start_addr=9f7b0000, size=262144,start_sector=123, end_sector=126 
 4. Name = EEPROM, offset = 7f0000, start_addr=9f7f0000, size=65536,start_sector=127, end_sector=127 
UBNT application initialized 
[... 1839 lines cut ...]
[   14.567752] procd: - init -
[   14.573369] procd: Parsing inittab - ::sysinit:/etc/init.d/rcS S boot
[   14.580231] procd: Parsing inittab - ::shutdown:/etc/init.d/rcS K shutdown
[   14.587503] procd: Parsing inittab - ::askconsole:/usr/libexec/login.sh
[   14.595520] procd: Launched new askconsole action, pid=478
[   14.605461] procd: running /etc/rc.d/S* boot
Please press Enter to activate this console.
[   15.604066] random: jshn: uninitialized urandom read (4 bytes read)
[   15.768076] random: ubusd: uninitialized urandom read (4 bytes read)
[   16.612446] kmodloader: loading kernel modules from /etc/modules.d/*
[   17.652027] Loading modules backported from Linux version v6.1-rc8-0-g76dcd734eca2
[   17.659757] Backport generated by backports.git v5.15.81-1-41-g02e352527db5
[   17.694133] urngd: v1.0.2 started.
[   18.776817] PPP generic driver version 2.4.2
[   18.798367] NET: Registered PF_PPPOX protocol family
[   18.941086] ath: phy0: Ignoring endianness difference in EEPROM magic bytes.
[   18.969438] ieee80211 phy0: Atheros AR9287 Rev:2 mem=0xb0000000, irq=13
[   18.993805] random: crng init done
[   18.997363] random: 24 urandom warning(s) missed due to ratelimiting
[   19.117106] kmodloader: done loading kernel modules from /etc/modules.d/*
WARNING: Variable 'wlan' does not exist or is not an array/object

``` 

</details>

Djfe, 2023-02-26:
> The "different MTD partition scheme" workaround documented on the wiki, with factory.bin, refuses to boot with Bad Magic Number:

The wiki actually tells you which flashing variant to use for which firmware.
You could've avoided it by finding out that your device is on firmware 4.xx or later :)

Looking at the mtd partitions: it's not that different, ubiquiti just merged `kernel`, `rootfs` and `board_config` into one partition.
The rest (especially EEPROM) stayed the same (size and offset).

I don't think that's why your wifi doesn't work. You therefore should rename this issue probably.
The only issue I can see here is the broken initialization of your wifi chip.
(The rest (bad magic number) is ubiquiti trying to make it more cumbersome for users to flash open firmwares)

Just an idea: maybe the issue is caused by your eeprom being different from other variants. But I don't know how to debug that.
You could export/backup the partition via LuCI though and look at it with a hex editor.
The offset of the wifi calibration data is set here (maybe your offset is different):
https://github.com/openwrt/openwrt/blob/master/target/linux/ath79/dts/ar7241_ubnt_unifi-ap.dtsi#L85
This message is why I'm wondering about your eeprom partition:
`ath: phy0: Ignoring endianness difference in EEPROM magic bytes.`

Djfe, 2023-02-26:
what is the output of `lspci`? (when it works, and when it doesn't)
https://github.com/openwrt/openwrt/issues/8527#issuecomment-1040394667

selfhoster1312, 2023-02-26:
>  You could've avoided it by finding out that your device is on firmware 4.xx or later :)

I did, i was just posting this here in case it was not expected (i will amend the wiki to say it's normal).

> Just an idea: maybe the issue is caused by your eeprom being different from other variants. But I don't know how to debug that.

Maybe someone else has an idea?

> what is the output of lspci? (when it works, and when it doesn't)

In both cases: `00:00.0 Network controller [0280]: Qualcomm Atheros AR9287 Wireless Network Adapter (PCI-Express) [168c:002e] (rev 01)`

Also worth noting, when the wifi "works" i've tried to enable an AP in the config and then i get kernel panics when wifi initialization succeeded. Here's a sample (they all look alike):

``` 
[  554.184472] ath: phy0: Unable to reset hardware; reset status -22 (freq 2412 MHz)
[  554.193839] Data bus error, epc == 803160a8, ra == 81ca1c78
[  554.199461] Oops[#1]:
[  554.201754] CPU: 0 PID: 1152 Comm: hostapd Not tainted 5.15.94 #0
[  554.207899] $ 0   : 00000000 00000001 deadc0de 81a756c0
[  554.213191] $ 4   : b0008088 81e17ba0 81e17bb4 00000005
[  554.218484] $ 8   : 00000024 80346c98 00000000 81e17bbe
[  554.223778] $12   : 000000a8 81e17a64 000000a8 807be098
[  554.229070] $16   : 81e17bb4 81e17ba4 00000001 00000005
[  554.234364] $20   : 81a78080 81964c80 81a76b88 00000000
[  554.239657] $24   : 00000018 00000000                  
[  554.244941] $28   : 81e16000 81e17b68 82ca2100 81ca1c78
[  554.250234] Hi    : 00000084
[  554.253129] Lo    : b61f0000
[  554.256024] epc   : 803160a8 ioread32+0xc/0x14
[  554.260541] ra    : 81ca1c78 ath_start_rfkill_poll+0x178/0x66c [ath9k]
[  554.267159] Status: 1000d403 KERNEL EXL IE 
[  554.271388] Cause : 1080001c (ExcCode 07)
[  554.275426] PrId  : 00019374 (MIPS 24Kc)
[  554.279376] Modules linked in: ath9k ath9k_common pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet ath9k_hw ath pppoxc
[  554.350454] Process hostapd (pid: 1152, threadinfo=428a9728, task=42ec665f, tls=77e3ddf0)
[  554.358694] Stack : 00000000 00000000 819f3754 81a756c0 81a7937c 00000000 81a78080 81a761c0
[  554.367160]         800047a4 81b9ce2c 00000140 81a78080 81a756c0 81a78080 00008088 0000808c
[  554.375628]         00008090 00008094 00008098 00000000 81a78080 00000000 00000002 81b9d5d8
[  554.384094]         81a78080 81b9cf20 81a756c0 81a756c0 81a78080 00000000 81a78080 81b9e588
[  554.392562]         2b364eaa 81a756c0 00000000 81a756c0 00000000 81a756c0 00000000 81ca5384
[  554.401029]         ...
[  554.403498] Call Trace:
[  554.405956] [<803160a8>] ioread32+0xc/0x14
[  554.410103] [<81ca1c78>] ath_start_rfkill_poll+0x178/0x66c [ath9k]
[  554.416371] 
[  554.417867] Code: 0000000f  8c820000  0000000f <03e00008> 00000000  0000000f  8c820000  0000000f  7c0210a0 
[  554.427746] 
[  554.429316] ---[ end trace bd551aee7e6ee82c ]---
[  554.433979] Kernel panic - not syncing: Fatal exception in interrupt
[  554.440371] Rebooting in 3 seconds..
```

Now lately i'm no more getting kernel panic but i get the uninitialized device error on every boot (sample: 20 reboots).

Maybe my wifi card is just defect? I should try on another AP-LR to be sure but they are still being used as we speak so this will have to wait...