openwrt/openwrt #12083: ubnt_unifi-ap-lr: eth0 should be WAN, not LAN

selfhoster1312, 2023-02-25:
### Describe the bug

Trying to flash Unifi AP-LR. On first boot eth0 is considered a LAN interface where DHCP server is enabled. However, this device only has a single RJ45 port so treating it as LAN and serving DHCP requests will pollute the existing network with broken DHCP answers.

As the device only has one port, it should be treated as WAN and not respond to DHCP queries. I don't know if that's standard for other boards, but eth0 should probably be configured as DHCP client by default so that it can be reached/configured on the network after flashing.

### OpenWrt version

r22130-524704e677

### OpenWrt target/subtarget

ath79/generic

### Device

Ubiquiti UniFi AP LR

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

obsy, 2023-02-26:
All devices with one port are treated as lan. Since Wi-Fi is disabled, setting this port as wan would make it impossible to access the router. Just set it up as you need. 
Not a bug.

selfhoster1312, 2023-02-26:
That makes sense. I'll have to build my own images then. Thanks!

closed 2023-02-26 by selfhoster1312