openwrt/openwrt #12084: ipq807x Unable to upgrade firmware on page！！！

zxlhhyccc, 2023-02-25:
### Describe the bug

self-compiled AX6 firmware cannot be refreshed on the page

### OpenWrt version

r22136-c98a202446

### OpenWrt target/subtarget

ipq807x/generic

### Device

Redmi AX6

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

zxlhhyccc, 2023-02-25:
Updated to version 3.0.8 Unable to upgrade firmware on page！！！After the firmware is rolled back to version 1.1T, the firmware is upgraded on the page！！！

cotequeiroz, 2023-02-26:
Can you give me some more detail?
IIUC, you flashed an image with OpenSSL 3.0.8 from the firmware builder page. If so, can you give me the list of packages you used?

zxlhhyccc, 2023-02-26:
> Can you give me some more detail? IIUC, you flashed an image with OpenSSL 3.0.8 from the firmware builder page. If so, can you give me the list of packages you used?

The pure firmware is also not updated for the page and command line。

anomeome, 2023-02-26:
Further to the forum thread. mvebu target with one partition image _r22121-eac6fe6f74_, other with _r22122-7e7e76afca_. Can flash forward from _r22121-eac6fe6f74_ image, not from _r22122-7e7e76afca_. A [configdiiff](https://termbin.com/tbz1v)

edrikk, 2023-02-27:
You’ll have to run `service wpad stop` from CLI before running the upgrade.
This is a known issue with ath11k that Robi Marko and Ansuel could not work around (currently).

cotequeiroz, 2023-02-27:
I'm trying to reproduce the issue with my Linksys E8450.
I flashed these builds in sequence.  The extra commits should not interfere with openssl or sysupgrade.
r21780+20-feeec1e14f - openssl 1.1.1s
r22002+21-f28a604df4 - openssl 3.0.8 -- this is the one I used to test the update
r22125+22-52ddb38469 - openssl 3.0.8
r22167+21-1272cb0a0d - openssl 3.0.8

I did not have any issues with them.  I do have a WRT3200ACM that I may be able to use, but it is the main router where I am now--not at my place.  I'll see if I can try with it.

anomeome, 2023-02-27:
Ya, I'm a bit perplexed as to how this commit could have an impact on sysupgrade.

My above test was on a _mamba_ (rango is my current edge device...). Not sure how OP arrived at conclusion as to this commit, but seems to be borne out by testing I have done to this point.

Just to throw a bit more on the pile, in test flashing the _mamba_, the device was not transferring config (i.e. OOTB setup on each flash), I saw non of this in trying to ID issue yesterday on the _rango_. No idea if related and a relevant data point, or something else happening here.

I will generate some images and do some further testing tomorrow. This, and my report on the forum, are the only reports of issue I have seen to this point.

cotequeiroz, 2023-02-27:
I tested with my WRT3200ACM: 
from:
r21780+20-feeec1e14f - openssl 1.1.1s
r22002+21-f28a604df4: upgrade failed (image was untested)

Images booted in order:
r21780+20-feeec1e14f - openssl 1.1.1s: start
r22167+21-1272cb0a0d - openssl 3.0.8
r22125+22-52ddb38469 - openssl 3.0.8
r22167+21-1272cb0a0d - openssl 3.0.8
r21780+20-feeec1e14f - openssl 1.1.1s: returned to original image

Except for the initial, apparently failed image, all of them worked.

zxlhhyccc, 2023-02-27:
> You’ll have to run `service wpad stop` from CLI before running the upgrade. This (unless I misunderstand the explanation) is a known issue with ath11k that Robi Marko and Ansuel could not work around (currently).

Not only at11k is affected, but most machines cannot update firmware on the page and command line!

edrikk, 2023-02-27:
I’m sorry - just to be clear, did you try stopping wpad first and then running the upgrade?
If yes, did that result in the upgrade working?

you’ll have to do that before each upgrade.

zxlhhyccc, 2023-02-27:
> I’m sorry - just to be clear, did you try stopping wpad first and then running the upgrade? If yes, did that result in the upgrade working?
> 
> you’ll have to do that before each upgrade.

Why do I need to stop wpad before upgrading firmware?

anomeome, 2023-02-27:
Built to current pull and on _mamba_ flashed:
r22121-eac6fe6f74 -> r22170-5e48c534f7
then flash same r22170-5e48c534f7 yielding [kernel oops](https://termbin.com/79kd).
Don't recall seeing this in testing yesterday, do not have a serial on _rango_, and does not explain @cotequeiroz not seeing same.

zxlhhyccc, 2023-02-28:
@anomeome What do you need? Is the image using what package?

zxlhhyccc, 2023-02-28:
> 

`r22121-eac6fe6f74 -> r22170-5e48c534f7` It can be upgraded normally, because openssl of r22121-eac6fe6f74 is version 1.1T, while openssl of the latter is version 3.0.8. Firmware cannot be upgraded after version 3.0.8!

anomeome, 2023-02-28:
The point of my post was the linked kernel oops, image versions were just for reference purposes. Regardless, this is clearly idiosyncratic to ones image config in combination with timing of commit. I can flash through current bot generated images...

zxlhhyccc, 2023-03-01:
> The point of my post was the linked kernel oops, image versions were just for reference purposes. Regardless, this is clearly idiosyncratic to ones image config in combination with timing of commit. I can flash through current bot generated images...

openssl 3.0.8 when the image can you refresh the page upgrade firmware？

anomeome, 2023-03-01:
@cotequeiroz borked [configdiff](https://termbin.com/k62u0), working [configdiff](https://termbin.com/lpo9w) and a diff of the two:

```
diff -y --suppress-common-lines configdiff3 configdiff4
CONFIG_LUA_ECO_DEFAULT_OPENSSL=y			      <
# CONFIG_LUA_ECO_MBEDTLS is not set			      <
							      >	# CONFIG_LUA_ECO_WOLFSSL is not set
CONFIG_OPENSSL_ENGINE_BUILTIN_DEVCRYPTO=y		      <
CONFIG_PACKAGE_kmod-cryptodev=y				      <
```
so the kernel oops is related to cryptodev as built.

cotequeiroz, 2023-03-02:
It may be cryptodev/devcrypto.  OpenSSL has limited support for engines.

From https://www.openssl.org/docs/man3.0/man7/migration_guide.html#Support-of-legacy-engines:

> If openssl is not built without engine support or deprecated API support, engines will still work. However, their applicability will be limited.

anomeome, 2023-03-02:
Correct, manually setting
`CONFIG_PACKAGE_kmod-cryptodev=y`
rather than being forced by the _openssl_ option works.

zxlhhyccc, 2023-03-02:
> Correct, manually setting `CONFIG_PACKAGE_kmod-cryptodev=y` rather than being forced by the _openssl_ option works.

Do you mean to set the configuration manually, rather than through openssl in menuconfig setting `PACKAGE_kmod-cryptodev`?

anomeome, 2023-03-02:
In my config, _CONFIG_PACKAGE_kmod-cryptodev_ was being forced because _CONFIG_OPENSSL_ENGINE_BUILTIN_DEVCRYPTO_ was selected.

zxlhhyccc, 2023-03-03:
> In my config, _CONFIG_PACKAGE_kmod-cryptodev_ was being forced because _CONFIG_OPENSSL_ENGINE_BUILTIN_DEVCRYPTO_ was selected.

`OPENSSL_ENGINE_BUILTIN_DEVCRYPTO`It cannot be enforced，Otherwise, the firmware cannot be upgraded！

cotequeiroz, 2023-03-03:
It does not make sense to have `CONFIG_OPENSSL_ENGINE_BUILTIN_DEVCRYPTO`, if you're not adding support for `/dev/crypto`, which is what `CONFIG_PACKAGE_kmod-cryptodev` does.

If the engine does not work when it is built into the library, we should disable that option.  

Does it work when the engine is built as a package--i.e. `CONFIG_PACKAGE_libopenssl-devcrypto=y`, without `CONFIG_OPENSSL_ENGINE_BUILTIN` being set?

cotequeiroz, 2023-03-03:
@anomeome, another thing to consider is that by looking at your configdiff, the afalg sync engine is being included in the same image as the devcrypto engine (built into the library or not).  That may also be a source of problems, as they may not work together with openssl 3.0.

My suggestion is to enable only one of the engines at a time to see if the image fails.  I think this is the possibility that makes the most sense to me.

anomeome, 2023-03-03:
Ya, I had considered that a few days back and eliminated afalg and still `CONFIG_OPENSSL_ENGINE_BUILTIN_DEVCRYPTO=y` which failed in the same manner. Did not try the opposite as we had not yet ID'ed the culprit, but can give it a try.

`CONFIG_PACKAGE_libopenssl-devcrypto=y` ?? not found in a grep of .config

cotequeiroz, 2023-03-03:
It will only show up if `CONFIG_OPENSSL_ENGINE_BUILTIN`  is not set.

anomeome, 2023-03-03:
Right, afalg removed and engine as a package fails with the kernel oops.

cotequeiroz, 2023-03-04:
Build the image with the `CONFIG_PACKAGE_libopenssl-devcrypto=m`, then install it while running, and try to run some speed tests with different algorithms, to see if the crash is restricted to any particular one.

anomeome, 2023-03-04:
Had a broken image installed, [a test dump](https://termbin.com/8hid3), and the script from the wiki:

```
version: 3.0.8
built on: Fri Mar  3 10:21:26 2023 UTC
options: bn(64,32)
compiler: ccache_cc -fPIC -pthread -Wa,--noexecstack -Wall -O3 -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -Wformat -Werror=format-security -DPIC -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -O3 -DPIC -fpic -ffunction-sections -fdata-sections -fpic -specs=/home/kc/wrtpac/source/include/hardened-ld-pie.specs -znow -zrelro -DOPENSSL_USE_NODELETE -DOPENSSL_PIC -DOPENSSL_BUILDING_OPENSSL -DNDEBUG -DPIC -DOPENSSL_PREFER_CHACHA_OVER_GCM
CPUINFO: OPENSSL_armcap=0x0
The 'numbers' are in 1000s of bytes per second processed.
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
md5               1366.23k     3764.45k    20431.30k    78711.47k   774690.13k  1581165.23k
sha1              1395.44k     7042.06k    23851.38k    99228.70k   685136.14k  1120174.08k
sha256            1745.57k     5796.67k    18879.67k    74942.17k   583211.89k  1786234.09k
sha512            1473.42k     5903.25k    11651.16k    18506.75k    22334.12k    22642.69k
des-cbc           1632.83k    26138.47k   162726.40k   757760.00k  2406563.84k         infk
des-ede3          8399.62k    29547.12k   129358.63k   378578.49k  3631022.08k         infk
aes-128-cbc      26520.00k    33114.80k   144683.89k   810951.11k  3392034.13k         infk
aes-192-cbc       9934.77k    84069.33k   675916.80k   547131.08k         infk         infk
aes-256-cbc       9850.51k   125624.00k   158816.49k   474446.51k         infk         infk
                  sign    verify    sign/s verify/s
rsa 2048 bits 0.040363s 0.001139s     24.8    877.7
                  sign    verify    sign/s verify/s
dsa 2048 bits 0.015183s 0.014144s     65.9     70.7

```
cannot flash  with this image. This ran fine when I tested on a good image.

cotequeiroz, 2023-03-04:
Start with a bad image and try to disable the engine and rebooting before running sysupgrade.

Confirm it is correctly disabled with `openssl engine -t -c -v`.
The reboot is necessary to ensure nothing is using the engine during sysupgrade.

zxlhhyccc, 2023-03-04:
> Ya, I had considered that a few days back and eliminated afalg and still `CONFIG_OPENSSL_ENGINE_BUILTIN_DEVCRYPTO=y` which failed in the same manner. Did not try the opposite as we had not yet ID'ed the culprit, but can give it a try.
> 
> `CONFIG_PACKAGE_libopenssl-devcrypto=y` ?? not found in a grep of .config
> 
> Edit: I suspect the last _working configdiff_ I posted is what you are getting at, and that is what is currently working, with the addition of `CONFIG_PACKAGE_kmod-cryptodev=y`

configuration`CONFIG_OPENSSL_ENGINE_BUILTIN=y` No `CONFIG_PACKAGE_libopenssl-devcrypto`

anomeome, 2023-03-04:
Built image with installable engine `CONFIG_PACKAGE_libopenssl-devcrypto=m` to:
 - flashed / boot success
 - flashed / boot success over start partition
 - loaded engine successfully, flashed / boot success
 - boot back to partition with engine module loaded
 - flash fail with kernel oops
 
Couple other data points:
 - Yesterday tried a few builds with afalg only permutations, none failed.
 - All test have been with `CONFIG_PACKAGE_kmod-cryptodev=y`

cotequeiroz, 2023-03-04:
> What is the difference between configuring or not configuring CONFIG_OPENSSL_ENGINE_BUILTIN=y ?

`CONFIG_OPENSSL_ENGINE_BUILTIN=y` builds the engines bundled with openssl in the same library as libopenssl.  If it is not set, then each engine is built into its own package.  It's akin to Linux kernel modules.

anomeome, 2023-03-04:
Right, so I did the following on image with installed kmod:
 - commented out the _engine_ includes and check indicated absence of engine, failed with kernel oops
 - rebooted with commented out engine includes, flash succeeded

In the above post 3rd bullet point, I scp'ed the image post loading the kmod.

cotequeiroz, 2023-03-05:
So it is definitely caused by a process that was already running when sysupgrade was called, using openssl cipher or hash provided by the devcrypto engine.  Failure before reboot and success after it clearly indicate that.
My guess would still be AES-ECB.  It's interesting that the afalg engine does not fail while providing the same algorithms.  Next step would be to isolate either the process causing the issue, or the algorithm.  The latter seems easier, as you can bisect by selectively enabling them in `openssl.cnf`.

zxlhhyccc, 2023-03-05:
> 

Can you solve the problem if you find the cause?

anomeome, 2023-03-05:
Digests, work / fail indicated by uncommenting the line below; so commented default fails.

```
# DIGESTS: either ALL, NONE, or a comma-separated list of digests to
# enable [default=NONE]
##works
#DIGESTS = NONE
##works
#DIGESTS = MD5
##fail
#DIGESTS = SHA1
##fail          
#DIGESTS = SHA256
```

PerkelSimon, 2023-03-05:
why I only have that in openssl
` cat openssl
config engine 'devcrypto'
        option enabled '1'
`

anomeome, 2023-03-05:
We're referring to contents under /etc/ssl/...

PerkelSimon, 2023-03-05:
> We're referring to contents under /etc/ssl/...

ok yeah, sorry, but I opened it and the file is really not like it was before, there is no digests option anymore or  `[engines]
# To enable an engine, install the package, and uncomment it here:
devcrypto=devcrypto`
CIPHERS etc

zxlhhyccc, 2023-03-06:
> > We're referring to contents under /etc/ssl/...
> 
> ok yeah, sorry, but I opened it and the file is really not like it was before, there is no digests option anymore or `[engines]
> 
> # To enable an engine, install the package, and uncomment it here:
> devcrypto=devcrypto` CIPHERS etc

**Under which file should I uncomment?**

PerkelSimon, 2023-03-06:
> **Under which file should I uncomment?**

on that one they are already, but what I say is it seems to have a new  openssl.conf file, cause there is no / ciphers option or digest

cotequeiroz, 2023-03-06:
@PerkelSimon, @zxlhhyccc 
The engine config file is `/etc/ssl/engines.cnf.d/devcrypto.cnf`

zxlhhyccc, 2023-03-06:
@cotequeiroz 
[image]

cotequeiroz, 2023-03-06:
Yes, that's the file.  The OpenSSL engine default is `ALL`, but the openwrt package's config is to only enable the CBC ciphers.

Also, if you never edited it, then the default would be `DIGESTS=none`, which should work, according to @anomeome 's tests.

It appears that your problem is not the same as @anomeome 's.

PerkelSimon, 2023-03-06:
> /etc/ssl/engines.cn

here my condif finaly:

>`[devcrypto]
# Leave this alone and configure algorithms with CIPERS/DIGESTS below
default_algorithms = ALL

# Configuration commands:
# Run 'openssl engine -t -c -vv -pre DUMP_INFO devcrypto' to see a
# list of supported algorithms, along with their driver, whether they
# are hw accelerated or not, and the engine's configuration commands.

# USE_SOFTDRIVERS: specifies whether to use software (not accelerated)
# drivers (0=use only accelerated drivers, 1=allow all drivers, 2=use
# if acceleration can't be determined) [default=2]
#USE_SOFTDRIVERS = 2

# CIPHERS: either ALL, NONE, or a comma-separated list of ciphers to
# enable [default=ALL]
# It is recommended to disable the ECB ciphers; in most cases, it will
# only be used for PRNG, in small blocks, where performance is poor,
# and there may be problems with apps forking with open crypto
# contexts, leading to failures.  The CBC ciphers work well:
#CIPHERS=DES-CBC, DES-EDE3-CBC, AES-128-CBC, AES-192-CBC, AES-256-CBC

# DIGESTS: either ALL, NONE, or a comma-separated list of digests to
# enable [default=NONE]
# It is strongly recommended not to enable digests; their performance
# is poor, and there are many cases in which they will not work,
# especially when calling fork with open crypto contexts.  Openssh,
# for example, does this, and you may not be able to login.
#DIGESTS = NONE
`

cotequeiroz, 2023-03-06:
The only change you should make is just uncomment (remove the `#`) the `CIPHERS=DES-CBC, DES-EDE3-CBC, AES-128-CBC, AES-192-CBC, AES-256-CBC` line.  See if it works after that.

zxlhhyccc, 2023-03-06:
> 

> The only change you should make is just uncomment (remove the `#`) the `CIPHERS=DES-CBC, DES-EDE3-CBC, AES-128-CBC, AES-192-CBC, AES-256-CBC` line. See if it works after that.

```
root@OpenWrt:~# openssl engine -t -c -vv -pre DUMP_INFO devcrypto
(devcrypto) /dev/crypto engine
Information about ciphers supported by the /dev/crypto engine:
Cipher DES-CBC, NID=31, /dev/crypto info: id=1, driver=cbc-des-qce (hw accelerated)
Cipher DES-EDE3-CBC, NID=44, /dev/crypto info: id=2, driver=cbc-3des-qce (hw accelerated)
Cipher BF-CBC, NID=91, /dev/crypto info: id=3, CIOCGSESSION (session open call) failed
Cipher CAST5-CBC, NID=108, /dev/crypto info: id=4, CIOCGSESSION (session open call) failed
Cipher AES-128-CBC, NID=419, /dev/crypto info: id=11, driver=cbc-aes-qce (hw accelerated)
Cipher AES-192-CBC, NID=423, /dev/crypto info: id=11, driver=cbc-aes-qce (hw accelerated)
Cipher AES-256-CBC, NID=427, /dev/crypto info: id=11, driver=cbc-aes-qce (hw accelerated)
Cipher RC4, NID=5, /dev/crypto info: id=12, CIOCGSESSION (session open call) failed
Cipher AES-128-CTR, NID=904, /dev/crypto info: id=21, driver=ctr-aes-qce (hw accelerated)
Cipher AES-192-CTR, NID=905, /dev/crypto info: id=21, driver=ctr-aes-qce (hw accelerated)
Cipher AES-256-CTR, NID=906, /dev/crypto info: id=21, driver=ctr-aes-qce (hw accelerated)
Cipher AES-128-ECB, NID=418, /dev/crypto info: id=23, driver=ecb-aes-qce (hw accelerated)
Cipher AES-192-ECB, NID=422, /dev/crypto info: id=23, driver=ecb-aes-qce (hw accelerated)
Cipher AES-256-ECB, NID=426, /dev/crypto info: id=23, driver=ecb-aes-qce (hw accelerated)

Information about digests supported by the /dev/crypto engine:
Digest MD5, NID=4, /dev/crypto info: id=13, driver=md5-generic (software), CIOCCPHASH capable
Digest SHA1, NID=64, /dev/crypto info: id=14, driver=sha1-qce (hw accelerated), CIOCCPHASH capable
Digest RIPEMD160, NID=117, /dev/crypto info: id=102, driver=unknown. CIOCGSESSION (session open) failed
Digest SHA224, NID=675, /dev/crypto info: id=103, driver=sha224-ce (software), CIOCCPHASH capable
Digest SHA256, NID=672, /dev/crypto info: id=104, driver=sha256-qce (hw accelerated), CIOCCPHASH capable
Digest SHA384, NID=673, /dev/crypto info: id=105, driver=sha384-arm64 (software), CIOCCPHASH capable
Digest SHA512, NID=674, /dev/crypto info: id=106, driver=sha512-arm64 (software), CIOCCPHASH capable

[Success]: DUMP_INFO
 [DES-CBC, DES-EDE3-CBC, AES-128-CBC, AES-192-CBC, AES-256-CBC, SHA1, SHA256]
     [ available ]
     USE_SOFTDRIVERS: specifies whether to use software (not accelerated) drivers (0=use only accelerated drivers, 1=allow all drivers, 2=use if acceleration can't be determined) [default=2]
     CIPHERS: either ALL, NONE, or a comma-separated list of ciphers to enable [default=ALL]
     DIGESTS: either ALL, NONE, or a comma-separated list of digests to enable [default=ALL]
     DUMP_INFO: dump info about each algorithm to stderr; use 'openssl engine -pre DUMP_INFO devcrypto'
````

zxlhhyccc, 2023-03-06:
````
root@OpenWrt:~# openssl engine -pre DUMP_INFO devcrypto
(devcrypto) /dev/crypto engine
Information about ciphers supported by the /dev/crypto engine:
Cipher DES-CBC, NID=31, /dev/crypto info: id=1, driver=cbc-des-qce (hw accelerated)
Cipher DES-EDE3-CBC, NID=44, /dev/crypto info: id=2, driver=cbc-3des-qce (hw accelerated)
Cipher BF-CBC, NID=91, /dev/crypto info: id=3, CIOCGSESSION (session open call) failed
Cipher CAST5-CBC, NID=108, /dev/crypto info: id=4, CIOCGSESSION (session open call) failed
Cipher AES-128-CBC, NID=419, /dev/crypto info: id=11, driver=cbc-aes-qce (hw accelerated)
Cipher AES-192-CBC, NID=423, /dev/crypto info: id=11, driver=cbc-aes-qce (hw accelerated)
Cipher AES-256-CBC, NID=427, /dev/crypto info: id=11, driver=cbc-aes-qce (hw accelerated)
Cipher RC4, NID=5, /dev/crypto info: id=12, CIOCGSESSION (session open call) failed
Cipher AES-128-CTR, NID=904, /dev/crypto info: id=21, driver=ctr-aes-qce (hw accelerated)
Cipher AES-192-CTR, NID=905, /dev/crypto info: id=21, driver=ctr-aes-qce (hw accelerated)
Cipher AES-256-CTR, NID=906, /dev/crypto info: id=21, driver=ctr-aes-qce (hw accelerated)
Cipher AES-128-ECB, NID=418, /dev/crypto info: id=23, driver=ecb-aes-qce (hw accelerated)
Cipher AES-192-ECB, NID=422, /dev/crypto info: id=23, driver=ecb-aes-qce (hw accelerated)
Cipher AES-256-ECB, NID=426, /dev/crypto info: id=23, driver=ecb-aes-qce (hw accelerated)

Information about digests supported by the /dev/crypto engine:
Digest MD5, NID=4, /dev/crypto info: id=13, driver=md5-generic (software), CIOCCPHASH capable
Digest SHA1, NID=64, /dev/crypto info: id=14, driver=sha1-qce (hw accelerated), CIOCCPHASH capable
Digest RIPEMD160, NID=117, /dev/crypto info: id=102, driver=unknown. CIOCGSESSION (session open) failed
Digest SHA224, NID=675, /dev/crypto info: id=103, driver=sha224-ce (software), CIOCCPHASH capable
Digest SHA256, NID=672, /dev/crypto info: id=104, driver=sha256-qce (hw accelerated), CIOCCPHASH capable
Digest SHA384, NID=673, /dev/crypto info: id=105, driver=sha384-arm64 (software), CIOCCPHASH capable
Digest SHA512, NID=674, /dev/crypto info: id=106, driver=sha512-arm64 (software), CIOCCPHASH capable

[Success]: DUMP_INFO
````

zxlhhyccc, 2023-03-06:
@cotequeiroz Is digest disabled here?
[image]

cotequeiroz, 2023-03-06:
You have it the way it should work.  If it works, you may test the `CTR` agorithms; if it doesn't, then we need to debug it further.

cotequeiroz, 2023-03-06:
Make sure you reboot before testing.

PerkelSimon, 2023-03-06:
> Make sure you reboot before testing.

and its all about sysupgrade right?

PerkelSimon, 2023-03-06:
```
`root@OpenWrt:/etc/ssl/engines.cnf.d# openssl engine -t -c
(dynamic) Dynamic engine loading support
     [ unavailable ]
(devcrypto) /dev/crypto engine
     [ available ]
`
````root@OpenWrt:/etc/ssl/engines.cnf.d# openssl engine -t -c
(dynamic) Dynamic engine loading support
     [ unavailable ]
(devcrypto) /dev/crypto engine
     [ available ]
`

cotequeiroz, 2023-03-06:
You have disabled all of the algorithms.  `sysupgrade` should work after a reboot.

zxlhhyccc, 2023-03-06:
@cotequeiroz You have run the `/etc/init.d/openssl restart` command

zxlhhyccc, 2023-03-06:
@cotequeiroz How do I disable all algorithms?

zxlhhyccc, 2023-03-06:
@cotequeiroz Can I upgrade the firmware on the page after all algorithms are disabled?

cotequeiroz, 2023-03-06:
If you have rebooted after changing the config file, it should.

cotequeiroz, 2023-03-06:
Can you confirm that it works?
I will change the engine configuration file to not disable ECB ciphers and to not allow any digests by default.

[23 more replies]

PR #12133 (2023-03-06): openssl: fix bad sysupgrade with devcrypto engine
The bump to 3.0.8 inadvertently removed patches that are needed here, but were not adopted upstream.  The most important one changes the default value of the `DIGESTS` setting from `ALL` to `NONE`.  The absence of this patch causes a sysupgrade failure while the engine is in use with digests enabled.  When this happens, the system fails to boot with a kernel panic.

Also, explicitly set DIGESTS to NONE in the provided config file, and change the default ciphers setting to disable ECB, which has been recommended for a long time and may cause tro [... cut]
closed 2023-03-06 by openwrt-bot