openwrt/openwrt #12086: bug mac80211

Oozil, 2023-02-26:
### Describe the bug

There is a bug in all versions of [image], which is that tx is always 6 megabytes

### OpenWrt version

master

### OpenWrt target/subtarget

ath79

### Device

archer c7 v2

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

schuettecarsten, 2023-02-26:
[image]
Hmmm. ath79, tp-link archer c7 v5, ...

Spudz76, 2023-02-26:
I have all sorts of various rates.  One laptop is 6Mbit but it is probably downshifting for powersave.
[image]

castillofrancodamian, 2023-03-03:
> Tengo todo tipo de tarifas variadas. Una computadora portátil es de 6 Mbit, pero probablemente esté reduciendo la marcha para ahorrar energía. [image]

That would be expected on RX, not TX, as is your case.