openwrt/openwrt #12088: [22.03.3] ipq806x: WLAN interface goes down on Netgear R7800 (crash report in the log)

Tarask1n, 2023-02-26:
### Describe the bug

After upgrading on 22.03.3 release I faced an issue. After 2-3 days of normal work, the WLAN interface goes down unexpectedly. A restart of the interface does not help, only a reboot restores normal functioning. I have two R7800 routers and it is reproduced on both.

```[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.161 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 11.2.0 r20028-43d71ad93e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Tue Jan 3 00:24:21 2023
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Netgear Nighthawk X4S R7800
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000042000000-0x000000005fffffff]
[    0.000000] On node 0 totalpages: 122880
[    0.000000]   Normal zone: 1080 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s31180 r8192 d22068 u61440
[    0.000000] pcpu-alloc: s31180 r8192 d22068 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121800
[    0.000000] Kernel command line: 
[    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 473120K/491520K available (6998K kernel code, 618K rwdata, 1024K rodata, 1024K init, 242K bss, 18400K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000010] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000029] Switching to timer-based delay loop, resolution 160ns
[    0.000349] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000381] pid_max: default: 32768 minimum: 301
[    0.000567] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000590] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001678] CPU: Testing write buffer coherency: ok
[    0.002007] qcom_scm: convention: smc legacy
[    0.003083] Setting up static identity map for 0x42300000 - 0x42300060
[    0.003284] rcu: Hierarchical SRCU implementation.
[    0.003609] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.003890] smp: Bringing up secondary CPUs ...
[    0.005941] smp: Brought up 1 node, 2 CPUs
[    0.005962] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005977] CPU: All CPU(s) started in SVC mode.
[    0.016522] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.016691] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016727] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.016878] pinctrl core: initialized pinctrl subsystem
[    0.018619] NET: Registered protocol family 16
[    0.018950] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.020257] thermal_sys: Registered thermal governor 'step_wise'
[    0.022426] cpuidle: using governor ladder
[    0.022501] cpuidle: using governor menu
[    0.070336] usbcore: registered new interface driver usbfs
[    0.070422] usbcore: registered new interface driver hub
[    0.070500] usbcore: registered new device driver usb
[    0.070567] pps_core: LinuxPPS API ver. 1 registered
[    0.070584] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.070624] PTP clock support registered
[    0.072971] clocksource: Switched to clocksource dg_timer
[    0.074088] NET: Registered protocol family 2
[... 611 lines cut ...]
### Device

Netgear Nighthawk X4S R7800

### Image kind

Official downloaded image

### Steps to reproduce

1. Flash 22.03.3 release
2. Use WiFi for 2-3 days intensively

### Actual behaviour

WLAN interface goes down
`CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.10.161 #0`

### Expected behaviour

WLAN should work normally

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

swg0101, 2023-03-07:
Also having the same issue here -- sometimes the WiFi interface would go down and nothing else would respond, and sometimes the kernel crashes, and the whole router reboots. It generally happens around once a day or so, although sometimes it happens within 10 mins of a crash.

Tarask1n, 2023-03-25:
might be related to the wifi driver (ath10k-ct)
https://github.com/greearb/ath10k-ct/issues/131
https://github.com/greearb/ath10k-ct/issues/188

NilsRo, 2023-04-20:
Using ath10k drivers and see the same result. The whole router stops working e.g. no crond entries anymore.

mjw99, 2024-11-05:
I am also seeing this on 23.05.5 with a R7800