openwrt/openwrt #12094: 2.4 GHz wifi disabled but SSID still visible

plouds, 2023-02-27:
### Describe the bug

A beginner here, will try my best.
A router manages home LAN + guest network. Dumb AP hAP ac2 broadcasts open 2.4 GHz guest wifi alongside 5 GHz network as part of home LAN.
Guest wifi remains visible after it's been disabled in LuCI.

### OpenWrt version

r22000-6bc675c0be

### OpenWrt target/subtarget

ipq40xx/mikrotik

### Device

MikroTik hAP ac2

### Image kind

Official downloaded image

### Steps to reproduce

Router, wired-only Netgear R7800 running stable 22.03.3, r20028-43d71ad93e, manages home LAN & guest network via VLAN. Dumb AP has 5 GHz WPA2 attached to home LAN + unmanaged guest interface with the 2.4 GHz wifi attached.
Dumb AP image created using online firmware selector.

### Actual behaviour

2.4 GHz guest wifi is enabled and a client (android phone) connects, receives an IP address from router, accesses internet.
Once guest wifi is disabled, client disconnects.
However, guest wifi SSID remains visible. A client trying to connect generates an "AP-STA-CONNECTED" record in log, followed by "AP-STA-DISCONNECTED" record 18 seconds later. Client never receives an IP address.
To my lay eye it seems that wifi continues to work but guest network stops short of asking router for IP address.
Restart of 2.4 GHz wireless device does not sort the issue.

### Additional info

Please, advise what additional information I should provide to help pinpoint the issue.

### Diffconfig

```text
-ash: ./scripts/diffconfig.sh: not found
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-02-27:
logread and dmesg, and try to maintain a minute between enabling/disabling connecting/disconnecting
Normal behaviout after "disable" is tapped in web interface is that wifi disacciates clients for few seconds and switches off the radio of wifi (do you have multiple AP-s on the same wifi radio?)

plouds, 2023-02-28:
OpenWrt SNAPSHOT, r22000-6bc675c0be

root@pokoj:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.91 (builder@buildhost) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 12.2.0 r22000-6bc675c0be) 12.2.0, GNU ld (GNU Binutils) 2.39) #0 SMP Sun Feb 5 21:39:17 2023
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: MikroTik hAP ac2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000080000000-0x0000000087dfffff]
[    0.000000]   HighMem  [mem 0x0000000087e00000-0x0000000087ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000087dfffff]
[    0.000000]   node   0: [mem 0x0000000087e00000-0x0000000087ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x0000000087ffffff]
[    0.000000] percpu: Embedded 12 pages/cpu s16588 r8192 d24372 u49152
[    0.000000] pcpu-alloc: s16588 r8192 d24372 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32484
[    0.000000] Kernel command line:
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 116372K/131072K available (6819K kernel code, 611K rwdata, 1708K rodata, 1024K init, 246K bss, 14700K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000002] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000025] Switching to timer-based delay loop, resolution 20ns
[    0.000346] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000377] pid_max: default: 32768 minimum: 301
[    0.000638] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000671] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001851] CPU: Testing write buffer coherency: ok
[    0.002301] qcom_scm: convention: smc legacy
[    0.003551] Setting up static identity map for 0x80300000 - 0x8030003c
[    0.003737] rcu: Hierarchical SRCU implementation.
[    0.004016] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.004398] smp: Bringing up secondary CPUs ...
[    0.008095] smp: Brought up 1 node, 4 CPUs
[    0.008139] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.008156] CPU: All CPU(s) started in SVC mode.
[    0.013121] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.013303] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.013339] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.013662] pinctrl core: initialized pinctrl subsystem
[    0.015482] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.015973] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.017141] thermal_sys: Registered thermal governor 'step_wise'
[    0.017658] cpuidle: using governor ladder
[    0.017726] cpuidle: using governor menu
[    0.025335] gpio-414 (enable USB power): hogged as output/high
[    0.043575] cryptd: max_cpu_qlen set to 1000
[    0.048172] usbcore: registered new interface driver usbfs
[    0.048265] usbcore: registered new interface driver hub
[    0.048332] usbcore: registered new device driver usb
[    0.048396] pps_core: LinuxPPS API ver. 1 registered
[    0.048410] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.048441] PTP clock support registered
[    0.050624] clocksource: Switched to clocksource arch_sys_counter
[    0.051589] NET: Registered PF_INET protocol family
[    0.051835] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.052636] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[... 326 lines cut ...]
[311785.814243] br0: port 7(phy0-ap0) entered forwarding state
[312023.667883] device phy0-ap0 left promiscuous mode
[312023.668161] br0: port 7(phy0-ap0) entered disabled state
[378299.474334] qca8k-ipq4019 c000000.switch lan3: Link is Up - 100Mbps/Full - flow control off
[378299.475009] br0: port 3(lan3) entered blocking state
[378299.482697] br0: port 3(lan3) entered forwarding state
[395422.943965] qca8k-ipq4019 c000000.switch lan3: Link is Down
[395422.945130] br0: port 3(lan3) entered disabled state

This is the bit of dmesg around enabling and disabling the sole AP on the 2.4 GHz radio:

[... cut ...]
[311785.559565] ath10k_ahb a000000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[311785.566881] br0: port 7(phy0-ap0) entered blocking state
[311785.574460] br0: port 7(phy0-ap0) entered disabled state
[311785.580357] device phy0-ap0 entered promiscuous mode
[311785.808373] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[311785.808655] br0: port 7(phy0-ap0) entered blocking state
[311785.814243] br0: port 7(phy0-ap0) entered forwarding state
[312023.667883] device phy0-ap0 left promiscuous mode
[312023.668161] br0: port 7(phy0-ap0) entered disabled state

plouds, 2023-02-28:
root@pokoj:~# logread | grep phy0-ap0
27. 02. 2023 00:36:35 notice hostapd[]: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
27. 02. 2023 00:36:37 info kernel[]: [311785.566881] br0: port 7(phy0-ap0) entered blocking state
27. 02. 2023 00:36:37 info kernel[]: [311785.574460] br0: port 7(phy0-ap0) entered disabled state
27. 02. 2023 00:36:37 info kernel[]: [311785.580357] device phy0-ap0 entered promiscuous mode
27. 02. 2023 00:36:37 notice hostapd[]: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
27. 02. 2023 00:36:37 info kernel[]: [311785.808373] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
27. 02. 2023 00:36:37 info kernel[]: [311785.808655] br0: port 7(phy0-ap0) entered blocking state
27. 02. 2023 00:36:37 info kernel[]: [311785.814243] br0: port 7(phy0-ap0) entered forwarding state
27. 02. 2023 00:36:37 notice hostapd[]: phy0-ap0: interface state COUNTRY_UPDATE->ENABLED
27. 02. 2023 00:36:37 notice hostapd[]: phy0-ap0: AP-ENABLED
27. 02. 2023 00:36:38 notice netifd[]: Network device 'phy0-ap0' link is up
27. 02. 2023 00:38:16 notice hostapd[]: phy0-ap0: AP-STA-CONNECTED 86:de:1d:cf:f4:52 auth_alg=open
27. 02. 2023 00:38:55 notice hostapd[]: phy0-ap0: AP-STA-DISCONNECTED 86:de:1d:cf:f4:52
27. 02. 2023 00:40:35 info kernel[]: [312023.667883] device phy0-ap0 left promiscuous mode
27. 02. 2023 00:40:35 info kernel[]: [312023.668161] br0: port 7(phy0-ap0) entered disabled state
27. 02. 2023 13:48:32 notice hostapd[]: phy0-ap0: AP-STA-CONNECTED e2:92:88:41:1b:d0 auth_alg=open
27. 02. 2023 13:48:50 notice hostapd[]: phy0-ap0: AP-STA-DISCONNECTED e2:92:88:41:1b:d0
27. 02. 2023 13:50:16 notice hostapd[]: phy0-ap0: AP-STA-CONNECTED 94:be:46:b6:65:76 auth_alg=open
27. 02. 2023 13:50:52 notice hostapd[]: phy0-ap0: AP-STA-DISCONNECTED 94:be:46:b6:65:76
27. 02. 2023 13:50:56 notice hostapd[]: phy0-ap0: AP-STA-CONNECTED 94:be:46:b6:65:76 auth_alg=open
27. 02. 2023 13:51:33 notice hostapd[]: phy0-ap0: AP-STA-DISCONNECTED 94:be:46:b6:65:76
27. 02. 2023 13:51:37 notice hostapd[]: phy0-ap0: AP-STA-CONNECTED 94:be:46:b6:65:76 auth_alg=open
27. 02. 2023 13:52:15 notice hostapd[]: phy0-ap0: AP-STA-DISCONNECTED 94:be:46:b6:65:76
27. 02. 2023 13:52:19 notice hostapd[]: phy0-ap0: AP-STA-CONNECTED 94:be:46:b6:65:76 auth_alg=open
27. 02. 2023 13:52:57 notice hostapd[]: phy0-ap0: AP-STA-DISCONNECTED 94:be:46:b6:65:76
27. 02. 2023 13:53:11 notice hostapd[]: phy0-ap0: AP-STA-CONNECTED 94:be:46:b6:65:76 auth_alg=open
27. 02. 2023 13:53:48 notice hostapd[]: phy0-ap0: AP-STA-DISCONNECTED 94:be:46:b6:65:76
27. 02. 2023 17:28:26 notice hostapd[]: phy0-ap0: AP-STA-CONNECTED 86:de:1d:cf:f4:52 auth_alg=open
27. 02. 2023 17:28:44 notice hostapd[]: phy0-ap0: AP-STA-DISCONNECTED 86:de:1d:cf:f4:52
27. 02. 2023 17:30:41 notice hostapd[]: phy0-ap0: AP-STA-CONNECTED 86:de:1d:cf:f4:52 auth_alg=open
27. 02. 2023 17:30:59 notice hostapd[]: phy0-ap0: AP-STA-DISCONNECTED 86:de:1d:cf:f4:52

plouds, 2023-02-28:
And I may have been wrong about the wireless client disconnecting. I have another AP, same device (hAP ac2), same setup, in another room. Different snapshot version, r22014-628870146d.
After I disable the guest wifi and don’t touch the client, it remains connected. For a while it shows connected, later "connected, no internet", but I suppose that’s just a delay in detecting the loss of access.

Corresponding logread:

28. 02. 2023 10:34:49 notice hostapd[]: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
28. 02. 2023 10:34:50 warning kernel[]: [434075.231291] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
28. 02. 2023 10:34:50 warning kernel[]: [434075.231371] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
28. 02. 2023 10:34:50 info kernel[]: [434075.266810] ath10k_ahb a000000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
28. 02. 2023 10:34:50 info kernel[]: [434075.267762] ath10k_ahb a000000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
28. 02. 2023 10:34:51 info kernel[]: [434075.551680] ath10k_ahb a000000.wifi: rts threshold -1
28. 02. 2023 10:34:51 warning kernel[]: [434075.551911] ath10k_ahb a000000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
28. 02. 2023 10:34:51 info kernel[]: [434075.559561] br0: port 6(phy0-ap0) entered blocking state
28. 02. 2023 10:34:51 info kernel[]: [434075.566864] br0: port 6(phy0-ap0) entered disabled state
28. 02. 2023 10:34:51 info kernel[]: [434075.572711] device phy0-ap0 entered promiscuous mode
28. 02. 2023 10:34:51 notice hostapd[]: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
28. 02. 2023 10:34:51 info kernel[]: [434075.804156] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
28. 02. 2023 10:34:51 info kernel[]: [434075.804507] br0: port 6(phy0-ap0) entered blocking state
28. 02. 2023 10:34:51 info kernel[]: [434075.809995] br0: port 6(phy0-ap0) entered forwarding state
28. 02. 2023 10:34:51 notice hostapd[]: phy0-ap0: interface state COUNTRY_UPDATE->ENABLED
28. 02. 2023 10:34:51 notice hostapd[]: phy0-ap0: AP-ENABLED
28. 02. 2023 10:34:51 notice hostapd[]: Configuration file: /var/run/hostapd-phy1.conf (phy phy1-ap0) --> new PHY
28. 02. 2023 10:34:53 warning kernel[]: [434077.603427] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
28. 02. 2023 10:34:53 warning kernel[]: [434077.603559] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
28. 02. 2023 10:34:53 info kernel[]: [434077.639362] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
28. 02. 2023 10:34:53 info kernel[]: [434077.640312] ath10k_ahb a800000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
28. 02. 2023 10:34:53 info kernel[]: [434077.923543] ath10k_ahb a800000.wifi: rts threshold -1
28. 02. 2023 10:34:53 warning kernel[]: [434077.923780] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
28. 02. 2023 10:34:53 info kernel[]: [434077.932017] br0: port 7(phy1-ap0) entered blocking state
28. 02. 2023 10:34:53 info kernel[]: [434077.938817] br0: port 7(phy1-ap0) entered disabled state
28. 02. 2023 10:34:53 info kernel[]: [434077.944704] device phy1-ap0 entered promiscuous mode
28. 02. 2023 10:34:53 notice hostapd[]: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
28. 02. 2023 10:34:53 notice hostapd[]: phy1-ap0: interface state COUNTRY_UPDATE->HT_SCAN
28. 02. 2023 10:34:53 notice netifd[]: Wireless device 'radio0' is now up
28. 02. 2023 10:34:53 notice netifd[]: Network device 'phy0-ap0' link is up
28. 02. 2023 10:34:53 notice hostapd[]: phy1-ap0: interface state HT_SCAN->DFS
28. 02. 2023 10:34:53 notice hostapd[]: phy1-ap0: DFS-CAC-START freq=5520 chan=104 sec_chan=-1, width=1, seg0=106, seg1=0, cac_time=60s
28. 02. 2023 10:34:54 notice netifd[]: Wireless device 'radio1' is now up
28. 02. 2023 10:35:17 notice hostapd[]: phy0-ap0: AP-STA-CONNECTED 86:de:1d:cf:f4:52 auth_alg=open
28. 02. 2023 10:35:57 info kernel[]: [434141.897559] ath10k_ahb a800000.wifi: mac flush null vif, drop 0 queues 0xffff
28. 02. 2023 10:35:57 notice hostapd[]: phy1-ap0: DFS-CAC-COMPLETED success=1 freq=5520 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5530 cf2=0
28. 02. 2023 10:35:57 info kernel[]: [434142.168046] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
28. 02. 2023 10:35:57 info kernel[]: [434142.168653] br0: port 7(phy1-ap0) entered blocking state
28. 02. 2023 10:35:57 info kernel[]: [434142.174076] br0: port 7(phy1-ap0) entered forwarding state
28. 02. 2023 10:35:57 notice netifd[]: Network device 'phy1-ap0' link is up
28. 02. 2023 10:35:57 notice hostapd[]: phy1-ap0: interface state DFS->ENABLED
28. 02. 2023 10:35:57 notice hostapd[]: phy1-ap0: AP-ENABLED
28. 02. 2023 10:41:10 info kernel[]: [434455.019932] device phy0-ap0 left promiscuous mode
28. 02. 2023 10:41:10 info kernel[]: [434455.020211] br0: port 6(phy0-ap0) entered disabled state
28. 02. 2023 10:41:10 notice netifd[]: radio0 (13173): WARNING: Variable 'data' does not exist or is not an array/object
28. 02. 2023 10:41:10 notice netifd[]: radio0 (13173): Bug: PHY is undefined for device 'radio0'
28. 02. 2023 10:41:10 notice netifd[]: Wireless device 'radio0' is now down

plouds, 2023-02-28:
Network restart does not sort it. Client remains connected throughout, still showing its IP address, with no internet access.

root@pracovna:~# /etc/init.d/network restart
'radio0' is disabled
root@pracovna:~# logread
...
28. 02. 2023 10:52:12 notice netifd[]: Interface 'guest' is now down
28. 02. 2023 10:52:12 notice netifd[]: Interface 'guest' is disabled
28. 02. 2023 10:52:12 notice netifd[]: 8021q 'br0.3' link is down
28. 02. 2023 10:52:12 notice netifd[]: Interface 'guest' has link connectivity loss
28. 02. 2023 10:52:12 notice netifd[]: Interface 'lan' is now down
28. 02. 2023 10:52:12 info kernel[]: [435116.988960] br0: port 7(phy1-ap0) entered disabled state
28. 02. 2023 10:52:12 info kernel[]: [435116.989138] br0: port 5(wan) entered disabled state
28. 02. 2023 10:52:12 info kernel[]: [435116.995255] device lan1 left promiscuous mode
28. 02. 2023 10:52:12 info kernel[]: [435116.998772] br0: port 1(lan1) entered disabled state
28. 02. 2023 10:52:12 notice hostapd[]: Remove interface 'phy1-ap0'
28. 02. 2023 10:52:12 notice hostapd[]: phy1-ap0: interface state ENABLED->DISABLED
28. 02. 2023 10:52:12 notice hostapd[]: phy1-ap0: AP-DISABLED
28. 02. 2023 10:52:12 notice hostapd[]: phy1-ap0: CTRL-EVENT-TERMINATING
28. 02. 2023 10:52:12 err hostapd[]: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
28. 02. 2023 10:52:12 info kernel[]: [435117.062853] device lan2 left promiscuous mode
28. 02. 2023 10:52:12 info kernel[]: [435117.063211] br0: port 2(lan2) entered disabled state
28. 02. 2023 10:52:12 notice hostapd[]: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
28. 02. 2023 10:52:12 info kernel[]: [435117.103522] device phy1-ap0 left promiscuous mode
28. 02. 2023 10:52:12 info kernel[]: [435117.103955] br0: port 7(phy1-ap0) entered disabled state
28. 02. 2023 10:52:12 info kernel[]: [435117.161494] device lan3 left promiscuous mode
28. 02. 2023 10:52:12 info kernel[]: [435117.162109] br0: port 3(lan3) entered disabled state
28. 02. 2023 10:52:12 info kernel[]: [435117.162292] ath10k_ahb a800000.wifi: mac flush null vif, drop 0 queues 0xffff
28. 02. 2023 10:52:12 warning kernel[]: [435117.293799] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 0
28. 02. 2023 10:52:12 warning kernel[]: [435117.294648] ath10k_ahb a800000.wifi: peer-unmap-event: unknown peer id 0
28. 02. 2023 10:52:12 info kernel[]: [435117.331696] device lan4 left promiscuous mode
28. 02. 2023 10:52:12 info kernel[]: [435117.332325] br0: port 4(lan4) entered disabled state
28. 02. 2023 10:52:12 info kernel[]: [435117.429287] device wan left promiscuous mode
28. 02. 2023 10:52:12 info kernel[]: [435117.429686] br0: port 5(wan) entered disabled state
28. 02. 2023 10:52:13 info kernel[]: [435117.537818] qca8k-ipq4019 c000000.switch wan: Link is Down
28. 02. 2023 10:52:13 info kernel[]: [435117.543440] ipqess-edma c080000.ethernet eth0: Link is Down
28. 02. 2023 10:52:13 notice netifd[]: Interface 'lan' is disabled
28. 02. 2023 10:52:13 notice netifd[]: 8021q 'br0.1' link is down
28. 02. 2023 10:52:13 notice netifd[]: Interface 'lan' has link connectivity loss
28. 02. 2023 10:52:13 notice netifd[]: bridge 'br0' link is down
28. 02. 2023 10:52:13 notice netifd[]: Network device 'phy1-ap0' link is down
28. 02. 2023 10:52:13 notice netifd[]: Network device 'wan' link is down
28. 02. 2023 10:52:13 notice netifd[]: Network device 'eth0' link is down
28. 02. 2023 10:52:13 notice netifd[]: Wireless device 'radio1' is now down
28. 02. 2023 10:52:13 notice netifd[]: Interface 'loopback' is now down
28. 02. 2023 10:52:13 notice netifd[]: Interface 'loopback' is disabled
28. 02. 2023 10:52:13 notice netifd[]: Network device 'lo' link is down
28. 02. 2023 10:52:13 notice netifd[]: Interface 'loopback' has link connectivity loss
28. 02. 2023 10:52:14 notice []: Added device handler type: bonding
28. 02. 2023 10:52:14 notice []: Added device handler type: 8021ad
28. 02. 2023 10:52:14 notice []: Added device handler type: 8021q
28. 02. 2023 10:52:14 notice []: Added device handler type: macvlan
28. 02. 2023 10:52:14 notice []: Added device handler type: veth
28. 02. 2023 10:52:14 notice []: Added device handler type: bridge
28. 02. 2023 10:52:14 notice []: Added device handler type: Network device
28. 02. 2023 10:52:14 notice []: Added device handler type: tunnel
28. 02. 2023 10:52:15 info kernel[]: [435120.494788] ipqess-edma c080000.ethernet eth0: configuring for fixed/internal link mode
28. 02. 2023 10:52:15 info kernel[]: [435120.494915] device eth0 left promiscuous mode
28. 02. 2023 10:52:15 info kernel[]: [435120.502351] ipqess-edma c080000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
28. 02. 2023 10:52:15 info kernel[]: [435120.505594] device eth0 entered promiscuous mode
28. 02. 2023 10:52:15 info kernel[]: [435120.514944] qca8k-ipq4019 c000000.switch lan1: configuring for phy/psgmii link mode
28. 02. 2023 10:52:15 info kernel[]: [435120.521538] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
28. 02. 2023 10:52:15 info kernel[]: [435120.527972] br0: port 1(lan1) entered blocking state
28. 02. 2023 10:52:16 info kernel[]: [435120.533818] br0: port 1(lan1) entered disabled state
28. 02. 2023 10:52:16 info kernel[]: [435120.539372] device lan1 entered promiscuous mode
28. 02. 2023 10:52:16 info kernel[]: [435120.549320] qca8k-ipq4019 c000000.switch lan2: configuring for phy/psgmii link mode
28. 02. 2023 10:52:16 info kernel[]: [435120.551647] br0: port 2(lan2) entered blocking state
[... cut ...]
28. 02. 2023 10:53:21 info kernel[]: [435186.362056] ath10k_ahb a800000.wifi: mac flush null vif, drop 0 queues 0xffff
28. 02. 2023 10:53:22 info kernel[]: [435186.632411] br0: port 6(phy1-ap0) entered blocking state
28. 02. 2023 10:53:22 info kernel[]: [435186.632553] br0: port 6(phy1-ap0) entered forwarding state
28. 02. 2023 10:53:22 notice netifd[]: Network device 'phy1-ap0' link is up
28. 02. 2023 10:53:22 notice hostapd[]: phy1-ap0: interface state DFS->ENABLED
28. 02. 2023 10:53:22 notice hostapd[]: phy1-ap0: AP-ENABLED

plouds, 2023-02-28:
Unable to replicate now. I tried "ip link set dev phy0-ap0 down" which produced the below response in syslog

28. 02. 2023 11:52:25 info kernel[]: [438730.438338] ath10k_ahb a000000.wifi: mac flush null vif, drop 0 queues 0xffff
28. 02. 2023 11:52:26 warning kernel[]: [438730.485607] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 0
28. 02. 2023 11:52:26 notice hostapd[]: phy0-ap0: INTERFACE-DISABLED 
28. 02. 2023 11:52:29 err hostapd[]: Failed to set beacon parameters

and seemed to take the SSID off air.

I re-enabled the guest wifi, connected to it, connected to the internet for a couple of minutes and afterwards it disabled correctly. Will test some more.

plouds, 2023-03-01:
A few tests later it seems that the bug - if ever it was a bug and not my own misconfiguration - does not appear in a later snapshot. In that affected snapshot version it is resolved by running "ip link set dev [device] down" from the command line

closed 2023-03-01 by plouds

plouds, 2023-03-02:
Actually, problem still exists, in the same shape and form, with the r22171-fdd1af9a44 snapshot of 1 March. But issue does not manifest on another AP of the exact same model (different MAC, different batch). Gotta be hardware thing.
My apologies for raising a non-existent issue. OpenWrt-wise.

brada4, 2023-03-02:
What packages did you cut put when configuring online?

plouds, 2023-03-02:
I'm not making things exactly simple as I also play around with extroot on a USB device. So the image to be flashed is stripped of what I hope I can get away with:

-wpad-basic-mbedtls -logd -dnsmasq -odhcp6c -odhcpd-ipv6only -ppp -ppp-mod-pppoe -kmod-leds-gpio -ath10k-firmware-qca4019-ct -kmod-ath10k-ct-smallbuffers -firewall4 -nftables

Added are:

block-mount e2fsprogs blkid f2fs-tools f2fsck kmod-fs-f2fs kmod-usb-storage kmod-usb-storage-uas

Once extroot is created these are installed:

luci ip-tiny htop rsync irqbalance wpad-mbedtls syslog-ng ath10k-firmware-qca4019-ct kmod-ath10k-ct-smallbuffers

(I believe LuCI reinstalls firewall but it is still a dumb AP, firewall not used)

brada4, 2023-03-02:
16MB flash should be sufficient for all default packages plus few extra needed for extroot. Though on other hand interesting which package missing disallows bringing interface down while permitting to bring it up.
Does the problem apply also to 5GHz radio?

plouds, 2023-03-03:
5 GHz radio does as it's told. When disabled, it comes off air.