openwrt/openwrt #12095: Packet steering init.d script tries to enable packet steering on wireless devices

anonymous-one, 2023-02-28:
### Describe the bug

Hello.

I am unsure if this is normal behavior, but `/etc/init.d/packet_steering` will attempt to enable packet steering on wireless interfaces which dumps out `sh: write error: No such file or directory` for each of the wireless interfaces. I have modified `/usr/libexec/network/packet-steering.sh` (which is called by the initi.d script) to output some basic logging:

```
root@OPENWRT-SALON:~# /etc/init.d/packet_steering start ;
/sys/class/net/br-lan
/sys/class/net/lan1
/sys/class/net/lan2
/sys/class/net/lan3
/sys/class/net/lo
/sys/class/net/miireg
/sys/class/net/phy1-ap0
sh: write error: No such file or directory
/sys/class/net/phy1-ap1
sh: write error: No such file or directory
/sys/class/net/phy2-ap0
sh: write error: No such file or directory
/sys/class/net/phy2-ap1
sh: write error: No such file or directory
/sys/class/net/phy2-ap2
sh: write error: No such file or directory
/sys/class/net/wan
/sys/class/net/wlan0
sh: write error: No such file or directory
```

My addition was only the directory output, without my changes the script will dump `sh: write error: No such file or directory` for each of the wireless interfaces.

### OpenWrt version

r22151-1d82a47b49

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-01:
You mean there is no check for queue files existence or that they exist but dont accept writes?

anonymous-one, 2023-03-01:
> You mean there is no check for queue files existence or that they exist but dont accept writes?

Hrm... Slightly strange behavior... It looks as though the file DOES exist, but when written to it fails with no such file or directory.

I am slightly out of my league here and not very well versed in sysfs internals, but I have made a small strace capture that should assist:

```
openat(AT_FDCWD, "/sys/class/net/wlan0/queues/tx-0/xps_cpus", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE, 0666) = 3
fcntl(1, F_DUPFD_CLOEXEC, 10)           = 11
fcntl(11, F_SETFD, FD_CLOEXEC)          = 0
dup3(3, 1, 0)                           = 1
close(3)                                = 0
write(1, "f\n", 2)                      = -1 ENOENT (No such file or directory)
write(2, "sh: write error: No such file or"..., 43sh: write error: No such file or directory
) = 43
```

For further verification:

```
root@OPENWRT-SALON:/etc/init.d# ls -la /sys/class/net/wlan0/queues/tx-0/xps_cpus
-rw-r--r--    1 root     root          4096 Mar  1 12:15 /sys/class/net/wlan0/queues/tx-0/xps_cpus
root@OPENWRT-SALON:/etc/init.d#
```

brada4, 2023-03-01:
If you echo same value as can be read from the "file"? 
As you see it is externally indistinguishable and nothing tells it shall not be written.

anonymous-one, 2023-03-01:
Well, here is the problem:

```
root@OPENWRT-KOTLOWNIA:/etc/init.d# ls -la /sys/class/net/wlan0/queues/tx-0/xps_cpus
-rw-r--r--    1 root     root          4096 Mar  1 20:33 /sys/class/net/wlan0/queues/tx-0/xps_cpus
root@OPENWRT-KOTLOWNIA:/etc/init.d# echo f > /sys/class/net/wlan0/queues/tx-0/xps_cpus
ash: write error: No such file or directory
root@OPENWRT-KOTLOWNIA:/etc/init.d#

```

brada4, 2023-03-02:
Seems driver limitation, not addressable in script.

closed 2023-03-02 by anonymous-one

pesa1234, 2023-03-08:
I get same error on my mvebu device WRT1900ACS, I don't know how to solve... Just disable packet steering right?

brada4, 2023-03-08:
It is a driver limitation/bug , really no need to disable anything, it will go away with time.
Official API is to write CPU mask to that file (or zero to handle packets on interrupting CPU)
Check the respective files of eth* adapters - if they dont multiqueue or steer it does not have any value and can be disabled.
While effect overlaps this is not same as ethtool -L/-l or irqbalance.

LGA1150, 2023-06-25:
Transmit packet steering won't work on single TX queue netdevs.
Previous kernel versions allow you to write to `xps_cpus` (it won't work of course). Now it just return `-ENOENT`. This is intentional behavior.

```c
static ssize_t xps_cpus_show(struct netdev_queue *queue, char *buf)
{
        struct net_device *dev = queue->dev;
        unsigned int index;
        int len, tc;

        if (!netif_is_multiqueue(dev))
                return -ENOENT;
        ...
```

Ansuel, 2024-04-18:
@anonymous-one in main this got fixed. On openwrt-23.05 i pushed a fix that should address this. Thanks for the report.

commit afb5fdd7 (2024-04-18): netifd: packet-steering: silence error on applying queue mask

Some queues can't be tweaked and return -ENOENT if it's not multiqueue.

Silence any error from echo to produce a more clean bootlog.

Fixes: #12095
Suggested-by: Andris PE <neandris@gmail.com>