openwrt/openwrt #12096: Linksys EA6350v4 MT7621DAT USB port does not work

mibas74, 2023-02-28:
### Describe the bug

After boot/restart USB port does not work.
There is no possibility to use any USB attached device.
Shortly after boot USB port seems to be deactivated and power supply line on the USB port is disconnected.
I've tried to use externally powered USB hub, but USB attached device is still being disconnected by usb driver.
Please find enclosed dmesg logs of boot without any USB devices attached to the port and with USB pendrive attached.
[EA6350USB_no_device_attached.txt](https://github.com/openwrt/openwrt/files/10849595/EA6350USB_no_device_attached.txt)
[EA6350USB_pendrive_attached.txt](https://github.com/openwrt/openwrt/files/10849599/EA6350USB_pendrive_attached.txt)

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys EA6350 v4

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dellasorte, 2023-05-16:
I tried active USB hub with external power supply and it worked. I plugged in external SATA to use as network storage and 4G Modem (RNDIS) and router has recognized both. But other problem appeared then. Device reboots randomly, so as for me it looks like the problem is in kernel power module that manages USB. Hope this info will be useful

ttytyper, 2024-05-19:
Same here, no USB power. Running OpenWrt 23.05.3 (r23809-234f1a2efa).

There's a forum post here https://forum.openwrt.org/t/linksys-ea6300-usb-port-not-working/135248 regarding missing USB power on EA6300, which seems to indicate that USB power is controlled by GPIO. If I understand it correctly. In that case specifically GPIO 10. Maybe EA6350v4 uses a similar approach?

dima8389, 2025-05-31:
Disable led script in system startup.
Issue will be resolved.
LEDs will work as expected.