openwrt/openwrt #12100: Archer C7 v2 Atheros AR8327N only 100Mbit

nicox, 2023-03-01:
### Describe the bug

The Firmware 22.03.0, 22.03.3 and the nightly build does not support Gigabit Connections on any Pport of the chip

### OpenWrt version

r22170-5e48c534f7

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v2

### Image kind

Official downloaded image

### Steps to reproduce

Connect the Router to a Gigabit switch.

### Actual behaviour

The link is only with 100Mbit, but the the Switch supports Gigabit

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

nicox, 2023-03-01:
Very interesting, i will retest with multiple devices tomorrow. i did this with an old firmware in the past but not with the new one.
Regards,
Nico

brada4, 2023-03-02:
Try cable shipped with router.