openwrt/openwrt #12103: wiki: authentication failure using github login

lynxis, 2023-03-01:
### Describe the bug

* Go to https://openwrt.org
* Press login on github
* After a long timeout it shows `An error occured during the request to the oauth provider: Could not connect to ssl://github.com:443 Connection timed out (110)`

### OpenWrt version

wiki

### OpenWrt target/subtarget

wiki

### Device

wiki

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

andyboeh, 2023-03-01:
I do not get this error, but the (expected) "self registration disabled" message:
[image]

ynezz, 2023-03-02:
> ssl://github.com:443 Connection timed out (110)

[image]

Works for me as well.