openwrt/openwrt #12112: MSS clamping does not play nice with policy-based routing (mwan3, pbr)

patrakov, 2023-03-03:
### Describe the bug

OpenWrt applies MSS clamping in the `firewall` and `firewall4` packages, by hooking into the `FORWARD `chain of the `mangle` table, or its `nftables` equivalent. This is done on the assumption that locally generated SYN packets will never need their MSS fixed up - but I found a situation where this assumption is wrong, and packets with MSS=1460 are wrongly sent through the VPN.

### OpenWrt version

21.02.1 and 22.03.3

### OpenWrt target/subtarget

ath79/generic and mediatek/mt7622

### Device

TP-Link Archer C7 V2 and Linksys E8450

### Image kind

Official downloaded image

### Steps to reproduce

Create a usual DHCP WAN connection, and a PPTP VPN. Set the metrics, as required by mwan3 which we will install at the next step: 80 on WAN, 90 on PPTP VPN. Assign both to the WAN firewall zone. At this point we have a default routing table with two default routes with different metrics, with the traffic going over WAN, because its default route has a lower metric. In the firewall settings, enable MSS clamping for the WAN zone.

Install mwan3 and curl. Install luci-app-mwan3 if you want.

Configure mwan3. The default policies that prefer wan over wanb, and wanb over wan, should be sufficient, when "wanb" is globally replaced with "vpn" in the config where it refers to the interface name. The default policy should be wan_wanb, thus preferring the WAN. Then create a rule in mwan3: if the destination IP is 139.59.209.225 (or whatever the current IPv4 address of openwrt.org is), then use the wanb_wan policy, thus preferring the VPN. Push the rule to the top, so that it is evaluated first. Delete the load-balancing rule. Save and apply.

The result at this point should be that everything goes through the WAN, except 139.59.209.225, which goes through the VPN. The real use case is to have an ipset instead of the single IP, but for testing, a single IP will suffice.

Then, on the router itself, while observing the resulting packets, run:

```
curl -4 https://openwrt.org/
```

### Actual behaviour

The initial SYN packet indicates MSS=1460, despite the rule that fixes the MTU. If the VPN blackholes "Fragmentation Needed" packets, then the page won't load.

### Expected behaviour

The initial SYN packet should have MSS clamped to 1360.

### Additional info

Applying the MSS clamp in the POSTROUTING chain, in addition to FORWARD (or maybe instead of FORWARD?), would fix the issue. OUTPUT is not sufficient.

Let's see why the wrong MSS exists in the first place, though.

When determining the MSS for locally-created SYN packets, apparently, the kernel only looks into the default routing table. Indeed, this decision is made before running through iptables/netfilter, and therefore, the kernel cannot know that it will select a non-default routing table based on the not-yet-existing firewall mark. Therefore, as the default routing table clearly says "go through WAN, MTU=1500", it selects MSS=1460.

The stock MSS-clamping rule is applied only on forward, but the packet is locally-originated, not forwarded, so it doesn't apply.

End result: packet with MSS=1460 (which is wrong) goes out.

The POSTROUTING chain of the mangle table would still catch it.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-03-03:
mark

brada4, 2023-03-04:
It is configurable per zone, by default enabled on WAN zone, so you need to enable MSS clamping also for your VPN zone, which will add 4 rules for that zone.

patrakov, 2023-03-04:
The VPN is a part of the WAN zone, and the checkbox is already enabled. The complaint is that there is a situation (involving policy-based routing) where the MSS still ends up being wrong despite the checkbox.

brada4, 2023-03-04:
Your estimation regarding POSTROUTING table is almost there, other direction should be altered at INPUT table (at least UI-wise checkbox applies to interface)

patrakov, 2023-03-04:
I don't know - maybe you are right. For me, POSTROUTING seems to be sufficient. Look, in your scenario (external connection arriving via VPN - and yes I have tested this) a SYN packet is the first thing that arrives via the VPN. Then the kernel tries to reply with SYN+ACK, and that's where the MSS needs to be adjusted. As this SYN+ACK is an outgoing packet, it is handled by OUTPUT and POSTROUTING. But in my original scenario, the first packet is a locally-originated SYN, so INPUT is definitely irrelevant.

patrakov, 2023-03-04:
Just to make it clear, adding a rule to POSTROUTING has been tested and confirmed to fix the following two scenarios:

1. Router-initiated TCP connections to hosts where policy routing says that they should go through the VPN despite the main routing table saying otherwise.
2. Externally initiated TCP connections to the router via the VPN with the policy sending the SYN+ACK reply via the VPN despite the main routing table saying otherwise.

brada4, 2023-03-04:
Agree that rule covering SYN and SYN+ACK in one direction functionally covers the need. Be it POSTROUTING or INPUT.

brada4, 2023-03-04:
ping @stintel @jow-

ptpt52, 2023-03-04:
mss change in mangle_forward chain
but with mwan3 policy-based routing, it may change the outgoing interface during POSTROUTING
e.g.
```
forward indicate the packets gona out via dev wan1, and mss changed in the mangle_forward rules
but then the packet go out via wan2 after the policy-based routing
```

ptpt52, 2023-03-04:
maybe we should move the MTU firewall rules to mangle_postrouting chain. so it resolve the issue.

jow-, 2023-03-04:
Can you suggest a specific diff?

ptpt52, 2023-03-04:
```diff
diff --git a/root/usr/share/firewall4/templates/ruleset.uc b/root/usr/share/firewall4/templates/ruleset.uc
index d6eedfd..91d5123 100644
--- a/root/usr/share/firewall4/templates/ruleset.uc
+++ b/root/usr/share/firewall4/templates/ruleset.uc
@@ -413,6 +413,14 @@ table inet fw4 {
 {% for (let rule in fw4.rules("mangle_postrouting")): %}
                {%+ include("rule.uc", { fw4, rule }) %}
 {% endfor %}
+{% for (let zone in fw4.zones()): %}
+{%  if (zone.mtu_fix): %}
+{%   for (let rule in zone.match_rules): %}
+               {%+ include("zone-mssfix.uc", { fw4, zone, rule, egress: false }) %}
+               {%+ include("zone-mssfix.uc", { fw4, zone, rule, egress: true }) %}
+{%   endfor %}
+{%  endif %}
+{% endfor %}
 {% fw4.includes('chain-append', 'mangle_postrouting') %}
        }
 
@@ -440,14 +448,6 @@ table inet fw4 {
 {% for (let rule in fw4.rules("mangle_forward")): %}
                {%+ include("rule.uc", { fw4, rule }) %}
 {% endfor %}
-{% for (let zone in fw4.zones()): %}
-{%  if (zone.mtu_fix): %}
-{%   for (let rule in zone.match_rules): %}
-               {%+ include("zone-mssfix.uc", { fw4, zone, rule, egress: false }) %}
-               {%+ include("zone-mssfix.uc", { fw4, zone, rule, egress: true }) %}
-{%   endfor %}
-{%  endif %}
-{% endfor %}
 {% fw4.includes('chain-append', 'mangle_forward') %}
        }
 {% fw4.includes('table-append') %}
```
@jow- 
what about this changes/diff?

patrakov, 2023-03-04:
We also need to do the same change in the legacy firewall3, for those who install it (e.g. docker currently recommends it instead of firewall4)

ptpt52, 2023-03-04:
wait, V2 patch:
```diff
diff --git a/root/usr/share/firewall4/templates/ruleset.uc b/root/usr/share/firewall4/templates/ruleset.uc
index d6eedfd..57b9094 100644
--- a/root/usr/share/firewall4/templates/ruleset.uc
+++ b/root/usr/share/firewall4/templates/ruleset.uc
@@ -413,6 +413,13 @@ table inet fw4 {
 {% for (let rule in fw4.rules("mangle_postrouting")): %}
                {%+ include("rule.uc", { fw4, rule }) %}
 {% endfor %}
+{% for (let zone in fw4.zones()): %}
+{%  if (zone.mtu_fix): %}
+{%   for (let rule in zone.match_rules): %}
+               {%+ include("zone-mssfix.uc", { fw4, zone, rule, egress: true }) %}
+{%   endfor %}
+{%  endif %}
+{% endfor %}
 {% fw4.includes('chain-append', 'mangle_postrouting') %}
        }
 
@@ -444,7 +451,6 @@ table inet fw4 {
 {%  if (zone.mtu_fix): %}
 {%   for (let rule in zone.match_rules): %}
                {%+ include("zone-mssfix.uc", { fw4, zone, rule, egress: false }) %}
-               {%+ include("zone-mssfix.uc", { fw4, zone, rule, egress: true }) %}
 {%   endfor %}
 {%  endif %}
 {% endfor %}
```

brada4, 2023-03-04:
```
*mangle
:FORWARD ACCEPT
:POSTROUTING ACCEPT
-A FORWARD -o eth0.2 -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone wan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
-A FORWARD -i eth0.2 -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone wan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
COMMIT
```
```
*mangle
:FORWARD ACCEPT
:POSTROUTING ACCEPT
-A POSTROUTING -o eth0.2 -p tcp -m tcp --tcp-flags FIN,SYN,RST SYN -m comment --comment "!fw3: Zone wan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
# -A FORWARD -i eth0.2 -p tcp -m tcp --tcp-flags SYN,RST SYN -m comment --comment "!fw3: Zone wan MTU fixing" -j TCPMSS --clamp-mss-to-pmtu
COMMIT
```

Both ideas are implemented
1/ handle only output and really closest to hitting the wire
2/ fixup SYN/SYN+ACK in same rule

ptpt52, 2023-03-05:
wait. @patrakov  what is the issue?

mwan3 mark the packets in PREROUTING hook, then MSS changes happen in FORWARD hook, I think there is not issue.

patrakov, 2023-03-05:
The FORWARD hook does not apply to locally generated packets, that's why.

brada4, 2023-03-05:
```diff
diff --git a/zones.c b/zones.c
index 51a8fdf..bb50fd2 100644
--- a/zones.c
+++ b/zones.c
@@ -577,7 +577,7 @@ print_interface_rule(struct fw3_ipt_handle *handle, struct fw3_state *state,
                                fw3_ipt_rule_comment(r, "Zone %s MTU fix logging", zone->name);
                                fw3_ipt_rule_target(r, "LOG");
                                fw3_ipt_rule_addarg(r, false, "--log-prefix", buf);
-                               fw3_ipt_rule_replace(r, "FORWARD");
+                               fw3_ipt_rule_replace(r, "POSTROUTING");
                        }
 
                        r = fw3_ipt_rule_create(handle, &tcp, NULL, dev, NULL, sub);
@@ -586,15 +586,15 @@ print_interface_rule(struct fw3_ipt_handle *handle, struct fw3_state *state,
                        fw3_ipt_rule_comment(r, "Zone %s MTU fixing", zone->name);
                        fw3_ipt_rule_target(r, "TCPMSS");
                        fw3_ipt_rule_addarg(r, false, "--clamp-mss-to-pmtu", NULL);
-                       fw3_ipt_rule_replace(r, "FORWARD");
+                       fw3_ipt_rule_replace(r, "POSTROUTING");
 
-                       r = fw3_ipt_rule_create(handle, &tcp, dev, NULL, sub, NULL);
+                       /* r = fw3_ipt_rule_create(handle, &tcp, dev, NULL, sub, NULL);
                        fw3_ipt_rule_addarg(r, false, "--tcp-flags", "SYN,RST");
                        fw3_ipt_rule_addarg(r, false, "SYN", NULL);
                        fw3_ipt_rule_comment(r, "Zone %s MTU fixing", zone->name);
                        fw3_ipt_rule_target(r, "TCPMSS");
                        fw3_ipt_rule_addarg(r, false, "--clamp-mss-to-pmtu", NULL);
-                       fw3_ipt_rule_replace(r, "FORWARD");
+                       fw3_ipt_rule_replace(r, "FORWARD"); */
                }
        }
        else if (handle->table == FW3_TABLE_RAW)
```

brada4, 2023-03-05:
@ptpt52 hey, the zone-mssfix.uc has syn/ * , it should be syn/syn,fin,rst to also fixup to SYN/ACK sent out (think "us" running a web server)

patrakov, 2023-03-05:
@brada4 could you please explain why the last rule has to be commented out? Does this still work with a case when a port is forwarded to an internal server? In other words, aren't we undoing the fix for #8122 ?

brada4, 2023-03-05:
fw4 addendum to @ptpt52 to also handle syn+ack leaving local servers.
```diff
diff --git a/root/usr/share/firewall4/templates/zone-mssfix.uc b/root/usr/share/firewall4/templates/zone-mssfix.uc
index b76cfb6..17b6e92 100644
--- a/root/usr/share/firewall4/templates/zone-mssfix.uc
+++ b/root/usr/share/firewall4/templates/zone-mssfix.uc
@@ -1,7 +1,7 @@
 {%+ if (rule.family): -%}
        meta nfproto {{ fw4.nfproto(rule.family) }} {%+ endif -%}
 {%+ include("zone-match.uc", { egress, rule }) -%}
-tcp flags syn tcp option maxseg size set rt mtu {%+ if (zone.log & 2): -%}
+tcp flags syn / syn,fin,rst tcp option maxseg size set rt mtu {%+ if (zone.log & 2): -%}
        log prefix "MSSFIX {{ zone.name }} out: " {%+ endif -%}
 comment "!fw4: Zone {{ zone.name }} {{
        fw4.nfproto(rule.family, true)
```

brada4, 2023-03-05:
We now patch packets in correct position, so no need to hack incoming traffic under tcpip stack so to make it respond properly.

quick ref https://www.digitalocean.com/community/tutorials/a-deep-dive-into-iptables-and-netfilter-architecture#chain-traversal-order

(yes, not the common sense - pre/post routing are closer to the wire than in/output)

brada4, 2023-03-05:
Peeking at #8122 - we dont control routers down the path that enforce MTU 512 or 64 and drop DF packets without sending frag-needed. In the extreme case e can take pessimism from google/cloudglare/azure/dnsmasq, like 1200-1300 MSS

patrakov, 2023-03-06:
I would rather still see the removal of the rule that handles traffic forwarded to the LAN machines separately. Or, better, not at all.

brada4, 2023-03-06:
Another repeater - docker traffic does not hit either of those old rules...
Your proposal is complete fw3 way, fw4 does not yet fixup SYN+ACK, once it does - one rule is faster than two.

brada4, 2023-03-07:
(i started with incomplete tests)
v21 - egress filter is catching SYN and SYN+ACK, but misplaced and does not grip docker traffic
v22 - egress filter is placed correctly but does not patch SYN+ACK packets
(solutions, tested by setting "wan" port with small MTU and passing "normal" connections both ways to local nginx/wget docker and NAT network, with mirror ports in and tcpdump in container)
v21 - move egress filter to postrouting table that really handles all traffic specified by `-o` and remove ingress filter [patch](https://github.com/openwrt/openwrt/issues/12112#issuecomment-1455082751)
v22 - change egress filter to handle both sides of 3-way [patch](https://github.com/openwrt/openwrt/issues/12112#issuecomment-1455131634) and remove ingress filter [patch](https://github.com/openwrt/openwrt/issues/12112#issuecomment-1454778428)

ptpt52, 2023-03-07:
@brada4 
I believe that the `tcp flags syn` match includes matching for both SYN packets and SYN/ACK packets.

brada4, 2023-03-07:
```
iptables-translate -A ZZZ -p tcp -m tcp --syn
nft 'add rule ip filter ZZZ tcp flags syn / fin,syn,rst,ack counter'
``

ptpt52, 2023-03-07:
@brada4 
it looks like `fin,syn,rst,ack` is somewhat like mask.

brada4, 2023-03-07:
I tried to be consistent with historical iptables default. It is bitops in kernel like not and or etc.

brada4, 2023-03-21:
@ptpt52 I looked at the final patch and it looks like incorporating all suggestions?

patrakov, 2023-03-21:
Sorry to say this - I still haven't tested the patch. Will hopefully do so in 48 hours.

brada4, 2023-04-13:
Does not seem to apply to 22.03.4 as tagged :-(

brada4, 2023-04-26:
@jow- (@asvio)
Correct firewall4 diff (fixing small oversight on @ptpt52 's part)
1a) relocate mssfix machinery from mangle-forward to mangle-postrouting chain
1b) leave only outward rule because of (2)
2) capture 1st and 2nd pieces of TCP 3-way in one rule
Tested once and found "worksforme" with rule relocated and MASQ and docker traffic being clapmped to artificially reduced MTU.

```diff
diff --git a/root/usr/share/firewall4/templates/ruleset.uc b/root/usr/share/firewall4/templates/ruleset.uc
index 7bd9309..0cfd48b 100644
--- a/root/usr/share/firewall4/templates/ruleset.uc
+++ b/root/usr/share/firewall4/templates/ruleset.uc
@@ -417,6 +417,13 @@ table inet fw4 {
 {% for (let rule in fw4.rules("mangle_postrouting")): %}
                {%+ include("rule.uc", { fw4, rule }) %}
 {% endfor %}
+{% for (let zone in fw4.zones()): %}
+{%  if (zone.mtu_fix): %}
+{%   for (let rule in zone.match_rules): %}
+                {%+ include("zone-mssfix.uc", { fw4, zone, rule, egress: true }) %}
+{%   endfor %}
+{%  endif %}
+{% endfor %}
 {% fw4.includes('chain-append', 'mangle_postrouting') %}
        }
 
@@ -444,14 +451,6 @@ table inet fw4 {
 {% for (let rule in fw4.rules("mangle_forward")): %}
                {%+ include("rule.uc", { fw4, rule }) %}
 {% endfor %}
-{% for (let zone in fw4.zones()): %}
-{%  if (zone.mtu_fix): %}
-{%   for (let rule in zone.match_rules): %}
-               {%+ include("zone-mssfix.uc", { fw4, zone, rule, egress: false }) %}
-               {%+ include("zone-mssfix.uc", { fw4, zone, rule, egress: true }) %}
-{%   endfor %}
-{%  endif %}
-{% endfor %}
 {% fw4.includes('chain-append', 'mangle_forward') %}
        }
 {% fw4.includes('table-append') %}
diff --git a/root/usr/share/firewall4/templates/zone-mssfix.uc b/root/usr/share/firewall4/templates/zone-mssfix.uc
index b76cfb6..17b6e92 100644
--- a/root/usr/share/firewall4/templates/zone-mssfix.uc
+++ b/root/usr/share/firewall4/templates/zone-mssfix.uc
@@ -1,7 +1,7 @@
 {%+ if (rule.family): -%}
        meta nfproto {{ fw4.nfproto(rule.family) }} {%+ endif -%}
 {%+ include("zone-match.uc", { egress, rule }) -%}
-tcp flags syn tcp option maxseg size set rt mtu {%+ if (zone.log & 2): -%}
+tcp flags syn / syn,fin,rst tcp option maxseg size set rt mtu {%+ if (zone.log & 2): -%}
        log prefix "MSSFIX {{ zone.name }} out: " {%+ endif -%}
 comment "!fw4: Zone {{ zone.name }} {{
        fw4.nfproto(rule.family, true)
```

ptpt52, 2023-04-27:
@brada4 
1. `tcp flags syn` include `syn`, `syn,ack`, that is all.
2. `mangle_forward ` need mssfix to fix the download dir packets, and fix upload dir packets in `mangle_postrouting` only.
3. `mangle_postrouting` handle the forward packets  and the local_out packets, that is why we fix mss here.

brada4, 2023-04-27:
i dont want to fix up packets that do not open actual connection
forward does not help local traffic vs postrouting and as per OP does not adapt varied MSS on PBR interfaces.

brada4, 2023-04-27:
Some more low-level considerations.
- Not PBR, but one can degrade MTU per route using net/host routes besides default/bigger ones, this is known to ?tables only post-routing.
- `man iptables-extensions` states that `-p tcp -j TCPMSS --...` is completely legit and will extract packets for fixup on its own. Go figure if it is faster that way.

brada4, 2023-06-20:
@jow- can this be implemented, seems ages since it last worked as intended, i.e PPPoE as only means to connect internet getting awkward fixes from "forums"

jow-, 2023-06-20:
@brada4 - please open a PR against https://github.com/openwrt/firewall4

brada4, 2023-06-21:
I will do some RTFM, v22 is few lines off - is single PR ok or I need some magic to add 7 lines away backport?

jow-, 2023-06-21:
A PR against master/main of firewall4.git is fine, I'll take care of the backport myself. I just lack the time to turn the diff suggestions here into a proper commit (and to figure out which one is the correct one and if it needs further amendments).

brada4, 2023-09-10:
run `fw4 print | less`
check that wan interfaces are in $wan_devices list as by default.
Go to `/etc/nftables.d/`
Create file named say my-mssfix.nft
```
chain my_mssfix {                                                    
    type filter hook postrouting priority mangle; policy accept;     
    oifname $wan_devices tcp flags syn / syn,fin,rst tcp option maxseg size set rt mtu
}
```
These includes are not checked by fw4 check or nft -c, be very careful to not kill your own connection by botching ruleset.

closed 2023-11-03 by openwrt-bot

Ovear, 2024-01-03:
Sorry for bother this closed issue. 

Does this patch include in firewall3 for Openwrt 21.02? Or will it be applied to firewall3 in the future if not?

Thanks

brada4, 2024-01-03:
Forks added it, never reached openwrt
https://github.com/openwrt/openwrt/issues/12112#issuecomment-1455082751