openwrt/openwrt #12121: realtek/rtl838x: GS1900-24 v1 ethernet slow

casipw, 2023-03-04:
### Describe the bug

It's really cool that OpenWrt now supports ZyXEL GS1900 switches. Unfortunately ethernet performance seems to be bad here. `iperf3` measures ~47 MBit/s in the one way and ~65 MBit/s in the other way. My machine is connected to the switch with a 1m Cat 6a ethernet cable. Both links have negotiated gigabit ethernet. I have tried different cables and also observed the slow speed in the wild.

### OpenWrt version

r22199-1f53fa2b83

### OpenWrt target/subtarget

realtek/rtl838x

### Device

ZyXEL GS1900-24 v1

### Image kind

Self-built image

### Steps to reproduce

On a fresh OpenWrt install:

```
root@OpenWrt:~# opkg update
root@OpenWrt:~# opkg install iperf3
root@OpenWrt:~# cat /sys/class/net/eth0/speed 
1000
root@OpenWrt:~# iperf3 -c 192.168.1.2
Connecting to host 192.168.1.2, port 5201
[  5] local 192.168.1.1 port 44990 connected to 192.168.1.2 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.20   sec  6.25 MBytes  43.6 Mbits/sec    0    191 KBytes       
[  5]   1.20-2.01   sec  4.62 MBytes  47.8 Mbits/sec    0    235 KBytes       
[  5]   2.01-3.14   sec  6.25 MBytes  46.6 Mbits/sec    0    235 KBytes       
[  5]   3.14-4.05   sec  5.00 MBytes  45.8 Mbits/sec    0    235 KBytes       
[  5]   4.05-5.11   sec  6.25 MBytes  49.5 Mbits/sec    0    235 KBytes       
[  5]   5.11-6.01   sec  5.00 MBytes  46.7 Mbits/sec    0    235 KBytes       
[  5]   6.01-7.13   sec  6.25 MBytes  46.8 Mbits/sec    0    235 KBytes       
[  5]   7.13-8.01   sec  5.00 MBytes  47.6 Mbits/sec    0    235 KBytes       
[  5]   8.01-9.06   sec  6.25 MBytes  50.2 Mbits/sec    0    235 KBytes       
[  5]   9.06-10.10  sec  6.25 MBytes  50.3 Mbits/sec    0    235 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.10  sec  57.1 MBytes  47.5 Mbits/sec    0             sender
[  5]   0.00-10.11  sec  57.1 MBytes  47.4 Mbits/sec                  receiver

iperf Done.
root@OpenWrt:~# iperf3 -s
-----------------------------------------------------------
Server listening on 5201 (test #1)
-----------------------------------------------------------
Accepted connection from 192.168.1.2, port 39994
[  5] local 192.168.1.1 port 5201 connected to 192.168.1.2 port 40006
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  7.97 MBytes  66.6 Mbits/sec                  
[  5]   1.00-2.00   sec  8.05 MBytes  67.5 Mbits/sec                  
[  5]   2.00-3.00   sec  7.07 MBytes  59.5 Mbits/sec                  
[  5]   3.00-4.00   sec  7.59 MBytes  63.6 Mbits/sec                  
[  5]   4.00-5.00   sec  7.68 MBytes  64.4 Mbits/sec                  
[  5]   5.00-6.00   sec  8.13 MBytes  68.0 Mbits/sec                  
[  5]   6.00-7.00   sec  7.85 MBytes  66.0 Mbits/sec                  
[  5]   7.00-8.00   sec  7.33 MBytes  61.5 Mbits/sec                  
[  5]   8.00-9.00   sec  7.48 MBytes  62.6 Mbits/sec                  
[  5]   9.00-10.00  sec  7.79 MBytes  65.4 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  76.9 MBytes  64.5 Mbits/sec                  receiver
```

On my machine:

```
$ cat /sys/class/net/eth0/speed 
1000
$ iperf3 -s
-----------------------------------------------------------
Server listening on 5201 (test #1)
-----------------------------------------------------------
Accepted connection from 192.168.1.1, port 44984
[  5] local 192.168.1.2 port 5201 connected to 192.168.1.1 port 44990
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  5.12 MBytes  43.0 Mbits/sec                  
[  5]   1.00-2.00   sec  5.62 MBytes  47.2 Mbits/sec                  
[  5]   2.00-3.00   sec  5.49 MBytes  46.0 Mbits/sec                  
[  5]   3.00-4.00   sec  5.53 MBytes  46.4 Mbits/sec                  
[  5]   4.00-5.00   sec  5.83 MBytes  48.9 Mbits/sec                  
[  5]   5.00-6.00   sec  5.65 MBytes  47.4 Mbits/sec                  
[  5]   6.00-7.00   sec  5.62 MBytes  47.2 Mbits/sec                  
[  5]   7.00-8.00   sec  5.62 MBytes  47.2 Mbits/sec                  
[  5]   8.00-9.00   sec  6.00 MBytes  50.3 Mbits/sec                  
[  5]   9.00-10.00  sec  5.95 MBytes  49.9 Mbits/sec                  
[  5]  10.00-10.11  sec   696 KBytes  50.1 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.11  sec  57.1 MBytes  47.4 Mbits/sec                  receiver
-----------------------------------------------------------
Server listening on 5201 (test #2)
-----------------------------------------------------------
^Ciperf3: interrupt - the server has terminated
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.2 port 40006 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  8.45 MBytes  70.9 Mbits/sec   16   36.8 KBytes       
[  5]   1.00-2.00   sec  7.95 MBytes  66.7 Mbits/sec   16   32.5 KBytes       
[  5]   2.00-3.00   sec  7.02 MBytes  58.9 Mbits/sec   30   39.6 KBytes       
[  5]   3.00-4.00   sec  7.64 MBytes  64.1 Mbits/sec   24   36.8 KBytes       
[  5]   4.00-5.00   sec  7.58 MBytes  63.6 Mbits/sec   12   55.1 KBytes       
[  5]   5.00-6.00   sec  8.33 MBytes  69.9 Mbits/sec    9   43.8 KBytes       
[  5]   6.00-7.00   sec  7.83 MBytes  65.7 Mbits/sec   10   39.6 KBytes       
[  5]   7.00-8.00   sec  7.27 MBytes  61.0 Mbits/sec   11   36.8 KBytes       
[  5]   8.00-9.00   sec  7.46 MBytes  62.6 Mbits/sec   18   33.9 KBytes       
[  5]   9.00-10.00  sec  7.89 MBytes  66.2 Mbits/sec    8   33.9 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  77.4 MBytes  65.0 Mbits/sec  154             sender
[  5]   0.00-10.00  sec  76.9 MBytes  64.5 Mbits/sec                  receiver

iperf Done.
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-04:
Please check speed between 2 switch ports.

musashino205, 2023-03-05:
That is not originally an assumed usage on RTL83xx/RTL93xx SoCs and it does not have the ability to demonstrate sufficient speed on communication through CPU. This is not a bug but design...

neheb, 2023-03-16:
it's possible to speed up if someone takes the time to clean the driver up. bu as @musashino205 said, this is not an assumed usage. These are switches.

Hurricos, 2023-03-16:
> That is not originally an assumed usage on RTL83xx/RTL93xx SoCs and it does not have the ability to demonstrate sufficient speed on communication through CPU. This is not a bug but design...

+1. Please close this.

Neustradamus, 2024-12-22:
Good news, there is a new merged PR linked to Zyxel GS1900:
- https://github.com/openwrt/openwrt/pull/16442