openwrt/openwrt #12124: lto1: internal compiler error: compiler does not support ZSTD LTO compression

fda77, 2023-03-04:
### Describe the bug

Please reopen and see https://github.com/openwrt/openwrt/issues/10832
There are 2 commits to fix
```
lto1: internal compiler error: compiler does not support ZSTD LTO compression
```

> The 2 commits af3417cf7c1d0f701c9b02777b2e9c054795513e and ac21375cbc2a3888a72302f2ce55d24f3ef84611 from https://github.com/dhewg/openwrt/commits/toolchain fixed it for me. Thx @dhewg !

### OpenWrt version

git

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

anomeome, 2023-03-05:
See [PR11970](https://github.com/openwrt/openwrt/pull/11970) and note the ccache chatter

fda77, 2023-03-05:
Thanks @anomeome so this issue could be closed.
The patches there workes for me. I just did a new git-clone and applied patches.

But I DID NOT CLEAR ccache! It is not needed.
Im using ccache 4.7.4 with fedora and using the cache also for other things beside openwrt
And i have no idea why everywhere everyone writes about cache cleaning

fda77, 2023-03-05:
-> #11970

closed 2023-03-05 by fda77

anomeome, 2023-03-05:
In this case a very real issue, long fight to resolve [ending here](https://github.com/openwrt/openwrt/pull/10862#issuecomment-1424843519).

fda77, 2023-03-05:
Thx, but ccache is not an issue for me.
Openwrt does never clear it for me as i use the default location:
```
grep CCA .config
CONFIG_CCACHE=y
CONFIG_CCACHE_DIR="/home/user/.cache/ccache"
```

This way all my checkout share the ccache cache.

Freetz had earlier also the silly idea to have a ccache PER CHECKOUT dir - i have no idea why someone should do this!

```
ccache -s
Cacheable calls:   3704181 / 5764316 (64.26%)
  Hits:            2344628 / 3704181 (63.30%)
    Direct:        1961582 / 2344628 (83.66%)
    Preprocessed:   383046 / 2344628 (16.34%)
  Misses:          1359553 / 3704181 (36.70%)
Uncacheable calls: 2058002 / 5764316 (35.70%)
Errors:               2133 / 5764316 ( 0.04%)
Local storage:
  Cache size (GB):    3.82 /    5.00 (76.34%)
  Cleanups:            376

```

dhewg, 2023-03-05:
ccache is the root if the problem. You merely avoided the issue with the zstd enabling patches. But that doesn't make the ccache issue go away

fda77, 2023-03-05:
ccache causes gcc+binutils to be build without zstd? And not the makefiles?

neheb, 2023-03-06:
My theory is that there's a GCC bug in 8-10 that causes ccache to misbehave. ccache built with GCC 12 seems to work fine.

Anyway, wipe .ccache to fix all issues.

fda77, 2023-03-06:
My theory: Since many year many peoply cry to "clean ccache" all the time. I never did it. Imho these people have no arguments but only theories.

dhewg, 2023-03-06:
anomeome linked to the messages where we nailed the issue.
If you don't even bother to read and understand we cannot help you.
And that tone doesn't help at all, go troll somewhere else.