openwrt/openwrt #12130: USB2 is not working on brcm47xx / ASUS WL-500GP

iddq, 2023-03-06:
### Describe the bug

USB2 is not working.

It was ok on 17.x but not ok on 18.x and later.

maybe all brcm47xx are affeted.

### OpenWrt version

r8077-7cbbab7246

### OpenWrt target/subtarget

brcm47xx/legacy

### Device

Asus WL500GP V1

### Image kind

Self-built image

### Steps to reproduce

```
root@OpenWrt:/tmp# lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/2p, 12M
root@OpenWrt:/tmp# lsusb
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
root@OpenWrt:/tmp# lsmod | grep usb
nls_base                5184  1 usbcore
usb_common              1566  1 usbcore
usbcore               133982  4 ohci_platform,ohci_hcd,ehci_platform,ehci_hcd
root@OpenWrt:/tmp# lsmod | grep hci
ehci_hcd               34871  1 ehci_platform
ehci_platform           3440  0 
ohci_hcd               24095  1 ohci_platform
ohci_platform           2960  0 
usbcore               133982  4 ohci_platform,ohci_hcd,ehci_platform,ehci_hcd
root@OpenWrt:/tmp#
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-06:
Try kmod-usb2-pci , at least spec says VIA USB controller is PCI.

iddq, 2023-03-07:
Thanks for the help!

brada4, 2023-03-07:
dont close, it has to be packaged by default as it is welded piece of mainboard.