openwrt/openwrt #12136: Since the WED was enabled inside the DTS, Redmi ax6000, the stock layout fell into a bootloop

PussAzuki, 2023-03-07:
### Describe the bug

As the title says,  I recently happened to upgrade to openssl3 and wed support firmware, and then found that it would not boot, I thought it was a problem with openssl3, so I waited until after today's openssl fix was submitted, I recompiled the firmware, however still infinite bootloop.

At this point I thought that since there was no report on github, the official daily firmware should boot fine, but the opposite happened and it still fell into bootloop.

At this point, I realized that this is not a problem that I alone encountered.

### OpenWrt version

r22235-a03076cc39

### OpenWrt target/subtarget

mediatek/flogic

### Device

xiaomi_redmi_ax6000

### Image kind

Official downloaded image

### Steps to reproduce

Flash as described on the official toh page

### Actual behaviour

bootloop

### Expected behaviour

boot to system

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

chbgdn, 2023-03-07:
Builds after adding WED into device tree not working only on stock layout variant. After moving to OpenWRT U-Boot, r22244-635d177ac9 successfully booted.

ping @nbd168

PussAzuki, 2023-03-07:
I really think you need to tell users about this...

chbgdn, 2023-03-07:
I think developers didn't know about this issue, because they using ubootmod version, which is working well. Bug affect only stock u-boot devices, and will be fixed as soon, as devs will know about it.

soxrok2212, 2023-03-09:
I can confirm this issue. Colleague attempted to flash OpenWrt from stock fw and bricked (it’s on its way to me to unbrick).

ptpt52, 2023-03-12:
How to fix it? Is there a workaround?

limianscfox, 2023-03-12:
Has this problem been solved in the latest stock layout version? Does it work?

chbgdn, 2023-03-12:
> Has this problem been solved in the latest stock layout version? Does it work?

Not yet
To use latest version on stock layout you can try revert [this](https://github.com/openwrt/openwrt/commit/9f6d25551e71b4fdb9c198606eb2389f85ae2edd) commit

limianscfox, 2023-03-13:
> > Has this problem been solved in the latest stock layout version? Does it work?
> 
> Not yet To use latest version on stock layout you can try revert [this](https://github.com/openwrt/openwrt/commit/9f6d25551e71b4fdb9c198606eb2389f85ae2edd) commit

I don't have the ability to compile by myself, I can only wait for bug fixes.

smerschjohann, 2023-03-14:
I think this is the same issue that I created right now: https://github.com/openwrt/openwrt/issues/12182

I unbricked mine with an older image using UART. I had to modify the load address to 0x46000000, it did not work with the 0x48000000 address described on the wiki page.

```

[0;33mSelect (enter for default):[0m 

[4;36mInput U-Boot's IP address:[0m 192.168.0.1
[4;36mInput TFTP server's IP address:[0m 192.168.0.150
[4;36mInput IP netmask:[0m 255.255.255.0
[4;36mInput file name:[0m initramfs-kernel.bin

Using ethernet@15100000 device
TFTP from server 192.168.0.150; our IP address is 192.168.0.1
Filename 'initramfs-kernel.bin'.
Load address: 0x46000000
Loading: *#################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 ########
	 4.5 MiB/s
done
```

nbd168, 2023-03-16:
Please try this commit from my staging tree: https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commitdiff;h=0fc9eefd495915303a90648318e9428a5cdba402

PussAzuki, 2023-03-16:
I'll give it a try, but can I ask how to easily download the patch from git.openwrt.org?

PussAzuki, 2023-03-16:
> Please try this commit from my staging tree: https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commitdiff;h=0fc9eefd495915303a90648318e9428a5cdba402

Nice, it works, at least now.

PussAzuki, 2023-03-16:
But I DO NOT see the mt7986 wo firmware to be loaded...

PussAzuki, 2023-03-16:
According to what is said [here](https://github.com/openwrt/mt76/issues/746), I should have successfully enabled WED

[image]

That should work, right? @nbd168

zhuang1234, 2023-03-20:
怪不得我新编译的文件怎么刷都无法正常启动，用openwrt官方最新版也不行。stock 版本的最大好处就是方便使用小米官方工具救砖。希望这个问题可以早点解决合入到官方吧

chika0801, 2023-03-20:
> > Please try this commit from my staging tree: https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commitdiff;h=0fc9eefd495915303a90648318e9428a5cdba402
> 
> Nice, it works, at least now.

What command should I enter to use this patch?

PussAzuki, 2023-03-20:
Execute in the root of your openwrt source code:
`curl -L 'https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commitdiff;h=0fc9eefd495915303a90648318e9428a5cdba402' -o mtk.patch`
`git apply mtk.patch`

And you should see these patches under target/linux/mediatek/patches-5.15

nbd168, 2023-03-20:
Thanks for testing, I pushed the fix to master. Should be working now without patches.

closed 2023-03-20 by nbd168