openwrt/openwrt #12138: (Failed to send request: Operation not permitted) on ipv4 only uplink when lan has ipv6

dandudikof, 2023-03-07:
### Describe the bug

(21.02.5 on e4200/ea4500)

Getting a (Failed to send request: Operation not permitted) on wget or opkg updates.

On a router that is disabled for ipv6 addressing on its internet port.
If forget to disable ip6v on any other interface/vlan. 

while pings and dns works.

Deleting/disabling the last ipv6 address, solves the problem.

But i assume that might not solve it for someone who wants to have 
ipv6 on some interfaces while having it disabled on the uplink.

### OpenWrt version

r16688-fa9a932fdb

### OpenWrt target/subtarget

kirkwood/generic

### Device

Linksys EA4500 (Viper)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

gregpalaci, 2024-09-16:
[image]

https://techoverflow.net/2020/12/24/how-to-fix-openwrt-opkg-failed-to-send-request-operation-not-permitted/

V3ntus, 2025-01-12:
Commenting to say that @jswhisperer helped out here. I had it pointing to a DNS server on my LAN (192.168.2.6 for example), but I suppose it was giving a bad DNS answer for OpenWRT repos. Setting the custom DNS to `1.1.1.` instead worked.