openwrt/openwrt #12142: 2.4 Wifi not working in client mode after upgrading to Openwrt 22.03.x

yHpgi, 2023-03-09:
### Describe the bug

Hi, I want to report a bug.
I am using Xiaomi Miwifi Mini.

in the latest release of openwrt (v22 series), the 2.4Ghz wifi work better than v21 series in AP mode. but not when I want to use the device in client mode.
it's can't even scan available wifi network around my router. 
I already tried all of the version of openwrt to check what version work best. I found that chaos-calmer is the best in Wifi 2.4 (in client mode) but it's really outdated isn't it?
Now I'm using my own compiled openwrt from LEDE 17.01.0 because it's has similar signal strength as Chaos-calmer i think.
I hope somebody can fix this.
Sorry for my bad English, and thanks in advance.

Regards.
yHpgi.

### OpenWrt version

20.03.0, r19685-512e76967f

### OpenWrt target/subtarget

ramips MT7620

### Device

xiaomi-miwifi-mini

### Image kind

Official downloaded image

### Steps to reproduce

-

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-09:
Try from command line:
Set region.
`iw reg set "HK"`
Then
`iw dev wlan0 scan | tee 5g`
`iw dev wlan1 scan | tee 2g`

yHpgi, 2023-03-09:
> Try from command line: Set region. `iw reg set "HK"` Then `iw dev wlan0 scan | tee 5g` `iw dev wlan1 scan | tee 2g`
> 
> Note that in accordance to description 5GHz network card has disconnected 2GHz antenna.

In OpenWrt 22 Series (I am using 22.03.2 for this test) it's return nothing.

```
root@OpenWrt:~# iw reg set "HK"
root@OpenWrt:~# iw dev wlan0 scan | tee 5g
root@OpenWrt:~# iw dev wlan1 scan | tee 2g
root@OpenWrt:~#
```

And it's whats the return in LEDE 17.01.0
* 5GHZ wifi is not available around here because people still using 2.4Ghz in my location, so I can't test the 5Ghz in client mode. is it working or not.

```
root@LEDE:~# iw reg set "HK"
root@LEDE:~# iw dev wlan0 scan | tee 5g
root@LEDE:~# iw dev wlan1 scan | tee 2g
BSS 10:27:f5:c6:8b:52(on wlan1)
        TSF: 64663747123 usec (0d, 17:57:43)
        freq: 2422
        beacon interval: 100 TUs
        capability: ESS (0x0401)
        signal: -67.00 dBm
        last seen: 810 ms ago
        Information elements from Probe Response frame:
        SSID: TPLINK-WR840N
        HT capabilities:
                Capabilities: 0x106e
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        No RX STBC
                        Max AMSDU length: 3839 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT RX MCS rate indexes supported: 0-15, 32
                HT TX MCS rate indexes are undefined
        HT operation:
                 * primary channel: 3
                 * secondary channel offset: above
                 * STA channel width: any
BSS d4:a1:48:3b:c8:fa(on wlan1)
        TSF: 633427729797 usec (7d, 07:57:07)
        freq: 2432
        beacon interval: 100 TUs
        capability: ESS (0x1411)
        signal: -99.00 dBm
        last seen: 670 ms ago
        Information elements from Probe Response frame:
        SSID: AMPEYAN Office
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)
        HT capabilities:
                Capabilities: 0x19bc
                        HT20
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
[... 101 lines cut ...]
        signal: -79.00 dBm
        last seen: 40 ms ago
        Information elements from Probe Response frame:
        SSID: @wifi.id
        HT capabilities:
                Capabilities: 0x19ad
                        RX LDPC
                        HT20
                        SM Power Save disabled
                        RX HT20 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 8 usec (0x06)
                HT RX MCS rate indexes supported: 0-15
                HT TX MCS rate indexes are undefined
        HT operation:
                 * primary channel: 13
                 * secondary channel offset: no secondary
                 * STA channel width: 20 MHz
root@LEDE:~#

```

brada4, 2023-03-09:
That is material proof of "does not work", thank you.

Obviously should be region where you live to get CH12-13 and anything 5GHz.

2 files have the channel report recorded by `tee` command, for purpose here you did enough.

For 5GHz - maybe you can start tether AP on your mobile phone and confirm working yes/no ?

yHpgi, 2023-03-09:
Thankyou for the response.
Yes, I've tried using my phone for hotspot and forcing it to run at 5ghz, but for some reason it just changes it from Wifi 4 to Wifi 5 and it's still running on the 2.4Ghz frequency.
I also tried changing the region of my phone, but that doesn't make any difference.
So, I'm sorry. for now I can not provide information about the performance of 5Ghz wifi.

brada4, 2023-03-09:
Thank you for trying, its already enough information.