openwrt/openwrt #12143: Bananapi BPI R3 periodic disconnect all

jtxnn678, 2023-03-09:
### Describe the bug

Bananapi BPI R3 is running as main router (DHCP server) with adblock, samba4 (nvme ssd), transmission daemon (disabled)

It was running smoothly last few months with longest uptime of 30+ days. Just recently it acted weird, it would disconnect all network clients and stopped respond to ping, ssh, luci login...etc. like every 2 days.
hard restart is only thing that can bring it back online again.

I didn't touched anything or modified any config since its last good run, it just suddenly act like this recently.

**Things I've tried but didn't solve:**

-     updated to the latest official snapshot.

-     Tried disable adblock, transmission.

UPDATE:
Disconnection just happened again after running for about 12hrs from fresh reboot (latest snapshot).

_REMARK: The disconnection has nothing to do with the overheat or temp related, as my records showed that BPI-R3 only reached 50c average for 48hrs with metal case cover fully closed._

The ethernet (which my PC connected directly to BPI-R3) crashed.

**Complete log since the ERROR:**
[crash.log](https://github.com/openwrt/openwrt/files/10928516/crash.log)

### OpenWrt version

r22248-bf055fcdca

### OpenWrt target/subtarget

mediatek filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Actual behaviour

- All client network devices lost connection to DHCP server (Openwrt Bananapi BPI-R3)
- ssh, ping, telnet all got no response from the router.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

tip2tail, 2023-03-10:
@jtxnn678 I have the same issue. I seen that you reverted to a previous commit where this worked without issues, can you provide me with the hash for that... Or even better the compiled image so I can flash my router too 

M

jtxnn678, 2023-03-10:
@tip2tail If you having exactly same issue as mine, can you please append to this post so that everyone can see this is not a isolated case. 

As far as I remember the commit that was running for up to 30 days straight without any problem is 9260027535.

You can use '**git checkout 9260027535**' command if you are able to compile the firmware yourself. Or I can compile it for you, just let me know where I can send the file.

tip2tail, 2023-03-11:
@jtxnn678 I am a technical person, developer etc, but the world of the bananapi is a little new to me still.

I am trying to figure out how to persist the log files and get the off the device to post here.

In the meantime if you were able to compile the fw and send to "t2tmark" at gmail dot com thanks

tip2tail, 2023-03-11:
Log trace as requested:

```
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.631015] ------------[ cut here ]------------
Sat Mar 11 09:41:43 2023 kern.warn kernel: [13274.635614] WARNING: CPU: 2 PID: 4335 at ___ieee80211_stop_tx_ba_session+0x348/0x3ac [mac80211]
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.644331] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_
ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tabl
es nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 slhc sfp nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcrc32c crc_ccitt com
pat crypto_safexcel pwm_fan hwmon i2c_gpio i2c_algo_bit sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_commo
n
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.713339] CPU: 2 PID: 4335 Comm: kworker/u8:2 Tainted: G        W         5.15.98 #0
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.721233] Hardware name: Bananapi BPI-R3 (DT)
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.725746] Workqueue: phy1 ieee80211_ba_session_work [mac80211]
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.731751] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.738691] pc : ___ieee80211_stop_tx_ba_session+0x348/0x3ac [mac80211]
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.745300] lr : ___ieee80211_stop_tx_ba_session+0x210/0x3ac [mac80211]
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.751909] sp : ffffffc00e8bbca0
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.755207] x29: ffffffc00e8bbca0 x28: ffffffc008bb0010 x27: ffffff80065fc000
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.762322] x26: 0000000000000001 x25: ffffff8003728880 x24: ffffffc008bb0000
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.769437] x23: ffffffc0009f2a2c x22: ffffff80051460e8 x21: 0000000000000001
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.776553] x20: ffffff800102eb00 x19: ffffff80065fc000 x18: ffffffc008aea2e0
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.783669] x17: 0000000000005900 x16: ffffffc008f65000 x15: 0000000000000498
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.790784] x14: 0000000000000188 x13: ffffffc00e8bb7a8 x12: ffffffc008b422e0
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.797899] x11: 0000000000000040 x10: ffffffc008ae7840 x9 : ffffffc008ae7838
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.805014] x8 : 0000000000000002 x7 : 0000000000000000 x6 : 00000027eb794bcf
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.812129] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.819244] x2 : 0000000000000001 x1 : 0000000000000002 x0 : 00000000ffffff92
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.826360] Call trace:
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.828791]  ___ieee80211_stop_tx_ba_session+0x348/0x3ac [mac80211]
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.835054]  ieee80211_ba_session_work+0x36c/0x390 [mac80211]
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.840796]  process_one_work+0x210/0x3b0
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.844792]  worker_thread+0x170/0x4d0
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.848525]  kthread+0x11c/0x130
Sat Mar 11 09:41:43 2023 kern.debug kernel: [13274.851740]  ret_from_fork+0x10/0x20
Sat Mar 11 09:41:43 2023 kern.warn kernel: [13274.855301] ---[ end trace 9190f7b6f7e483e4 ]---
```

jtxnn678, 2023-03-11:
@tip2tail you can put a script in Local Startup to save system log file to an another file somewhere else so it will be persistent.

In my case, I save it to a file on attached nvme SSD.
`(logread -f >> /mnt/nvme0n1p1/openwrt.log) &`

I am still trying to figure out what caused this connection issue (aside from firmware bug), because I also have Redmi AX6000 (almost identical hardware as BPI-R3) running stable on latest commit.

Can you pls answer following, so I can have some idea:
1. Have you upgraded the firmware before the issue occure?
2.  Did you choose to save settings during upgrade?

I can compile to previous commit (at least no disconnection issue), do you need me to include NVME driver built in kernel?

frank-w, 2023-03-12:
do you have enabled any kind of hardware-acceleration (flowoffload,wed)? if yes, can you try without this?

jtxnn678, 2023-03-12:
> do you have enabled any kind of hardware-acceleration (flowoffload,wed)? if yes, can you try without this?

Prior to the periodic disconnection issue, I've tried enable/disable both Soft/Hardware flow offload before to test SQM and bufferbload, it never worked or has any effect/impact with that settings. So I turned it off by default.

I think I may found some clue (see below), but still need longer time for conclusion.

- I re-flash to latest commit (on that day) r22253-89390a9e50

- Discard all existing settings, and reconfig only whats neccessary (Wifi, Firewall, Samba4 share, adblock)

After 2 days 11 hours, no issue and everything is stable, even the high memory usage is gone.
[image]

tip2tail, 2023-03-12:
Many thanks @jtxnn678 I have been using the previous commit you sent me and it has not caused me any issues since.

jtxnn678, 2023-03-12:
> Many thanks @jtxnn678 I have been using the previous commit you sent me and it has not caused me any issues since.

Welcome.
**Can you also try upgrade to the latest commit without save settings?**

_I updated to the latest snapshot on March 09, discarded all existing settings then reconfigured it from fresh start. **No more disconnection encountered so far.**_

jtxnn678, 2023-03-15:
**CLOSED due to unable to reproduce the reported issue after an fresh upgrade (discard all existing all settings)**

melroy89, 2023-03-22:
This still was a kernel module/driver crash, which is never good. 

I think despite you are now unable to reproduce it, somebody might want to look into this a bit deeper.

jtxnn678, 2023-03-23:
@melroy89 Thank you.
Its been running 13 days straight without the said issue or kernel crash, since I reflashed firmware and discarded all existing settings.

However, I am still observing & counting the uptime. Anyone has encountered the same issue please feel free to update this topic, I will reopen it.

jtxnn678, 2023-03-23:
> This still was a kernel module/driver crash, which is never good.
> 
> I think despite you are now unable to reproduce it, somebody might want to look into this a bit deeper.

What a coincidence! Just when you commented today, it occurred again just 20min ago.

```
Thu Mar 23 22:30:51 2023 kern.warn kernel: [1164017.939720] ------------[ cut here ]------------
Thu Mar 23 22:30:51 2023 kern.info kernel: [1164017.944517] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 3 timed out
Thu Mar 23 22:30:51 2023 kern.warn kernel: [1164017.951656] WARNING: CPU: 2 PID: 0 at dev_watchdog+0x330/0x33c
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164017.957655] Modules linked in: iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables sch_cake sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ifb pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 slhc sfp nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcrc32c crc_ccitt compat crypto_safexcel pwm_fan
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164017.957807]  hwmon i2c_gpio i2c_algo_bit nvme nvme_core autofs4 sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.064271] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.15.98 #0
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.070433] Hardware name: Bananapi BPI-R3 (DT)
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.075118] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.082232] pc : dev_watchdog+0x330/0x33c
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.086403] lr : dev_watchdog+0x330/0x33c
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.090570] sp : ffffffc008c0bdb0
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.094042] x29: ffffffc008c0bdb0 x28: 0000000000000140 x27: 00000000ffffffff
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.101332] x26: 0000000000000000 x25: 0000000000000002 x24: ffffff80009224c0
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.108620] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008ad6000
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.115909] x20: ffffff8000922000 x19: 0000000000000003 x18: ffffffc008aea2e0
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.123198] x17: ffffffc0770ed000 x16: ffffffc008c0c000 x15: 0000000000000798
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.130487] x14: 0000000000000288 x13: ffffffc008c0bad8 x12: ffffffc008b422e0
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.137776] x11: 712074696d736e61 x10: ffffffc008b422e0 x9 : 0000000000000000
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.145065] x8 : ffffffc008aea290 x7 : ffffffc008aea2e0 x6 : 0000000000000001
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.152353] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.159642] x2 : 0000000000000040 x1 : 0000000000000004 x0 : 000000000000003f
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.166931] Call trace:
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.169536]  dev_watchdog+0x330/0x33c
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.173358]  call_timer_fn.constprop.0+0x20/0x80
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.178133]  __run_timers.part.0+0x208/0x284
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.182560]  run_timer_softirq+0x38/0x70
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.186639]  _stext+0x10c/0x28c
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.189940]  __irq_exit_rcu+0xdc/0xfc
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.193762]  irq_exit+0xc/0x1c
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.196976]  handle_domain_irq+0x60/0x8c
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.201059]  gic_handle_irq+0x50/0x120
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.204969]  call_on_irq_stack+0x28/0x44
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.209050]  do_interrupt_handler+0x4c/0x54
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.213389]  el1_interrupt+0x2c/0x4c
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.217124]  el1h_64_irq_handler+0x14/0x20
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.221378]  el1h_64_irq+0x74/0x78
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.224937]  arch_cpu_idle+0x14/0x20
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.228672]  do_idle+0xc0/0x150
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.231976]  cpu_startup_entry+0x20/0x60
Thu Mar 23 22:30:51 2023 kern.debug kernel: [1164018.236056]  secondary_start_kernel+0x130/0x140
[... cut ...]
Thu Mar 23 22:30:52 2023 kern.err kernel: [1164018.919988] mtk_soc_eth 15100000.ethernet: warm reset failed
Thu Mar 23 22:30:52 2023 kern.info kernel: [1164018.938997] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
Thu Mar 23 22:30:52 2023 kern.info kernel: [1164018.947550] mtk_soc_eth 15100000.ethernet eth1: configuring for inband/2500base-x link mode
Thu Mar 23 22:30:52 2023 kern.info kernel: [1164018.947561] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
Thu Mar 23 22:30:57 2023 daemon.notice hostapd: phy1-ap0: AP-STA-

```

jtxnn678, 2023-03-23:
@tip2tail  Any update with your BPI-R3?

frank-w, 2023-03-23:
Imho this should be separated...

Trace from @tip2tail is different and points to mac80211 (wifi): https://github.com/openwrt/openwrt/issues/12143#issuecomment-1464886933

@jtxnn678 trace points to mtk eth driver (eth0 transmit timeout)

jtxnn678, 2023-03-24:
> Imho this should be separated...
> 
> Trace from @tip2tail is different and points to mac80211 (wifi): [#12143 (comment)](https://github.com/openwrt/openwrt/issues/12143#issuecomment-1464886933)
> 
> @jtxnn678 trace points to mtk eth driver (eth0 transmit timeout)

Yeah, it was always **NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 3 timed out** for me.

frank-w, 2023-03-24:
You could try to drop this patch

https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=target/linux/generic/backport-5.15/730-04-v6.3-net-ethernet-mtk_eth_soc-implement-multi-queue-suppo.patch;h=79f2d0a38ab0d2c23be9c16fd343170575ed8f82;hb=HEAD

This is known to bring issues to different mtk boards

jtxnn678, 2023-03-24:
@frank-w Is this patch already included in the latest commit?

frank-w, 2023-03-24:
The patch yes,it is in some time...see my link.

But you should try to drop/revert it.

I'm not familar with openwrts patch system. It is maybe more work as just deleting the patch file as later patches may depend on it.

jtxnn678, 2023-03-24:
> I'm not familar with openwrts patch system. It is maybe more work as just deleting the patch file as later patches may depend on it.

Same, I am also not familiar on how patch works. I will try to install latest commit see if the issue still exists.

jtxnn678, 2023-03-28:
The latest commit (or commit after Feb, 2023) really has some issues.

Flashed to latest commit OpenWrt SNAPSHOT r22387-47de2c6862, running for 2 days then lost connection again. This time, the Ping, ssh, and even Luci is still accessible, the clients retained their assigned IP but just lost upstream connection (Internet).

```
Tue Mar 28 08:48:10 2023 kern.warn kernel: [258902.740244] ------------[ cut here ]------------
Tue Mar 28 08:48:10 2023 kern.info kernel: [258902.744949] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 3 timed out
Tue Mar 28 08:48:10 2023 kern.warn kernel: [258902.752008] WARNING: CPU: 0 PID: 0 at dev_watchdog+0x330/0x33c
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.757915] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 slhc sfp nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcrc32c crc_ccitt compat crypto_safexcel fuse pwm_fan hwmon i2c_gpio i2c_algo_bit nvme nvme_core autofs4 sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.829449] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.15.102 #0
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.835610] Hardware name: Bananapi BPI-R3 (DT)
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.840208] pstate: 40000005 (nZcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.847236] pc : dev_watchdog+0x330/0x33c
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.851316] lr : dev_watchdog+0x330/0x33c
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.855395] sp : ffffffc008003db0
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.858780] x29: ffffffc008003db0 x28: 0000000000000140 x27: 00000000ffffffff
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.865983] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff80009214c0
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.873184] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008ad6000
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.880386] x20: ffffff8000921000 x19: 0000000000000003 x18: ffffffc008aea2e0
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.887588] x17: ffffffc0770cb000 x16: ffffffc008004000 x15: 0000000000000681
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.894789] x14: 000000000000022b x13: ffffffc008003ad8 x12: ffffffc008b422e0
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.901991] x11: 712074696d736e61 x10: ffffffc008b422e0 x9 : 0000000000000000
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.909193] x8 : ffffffc008aea290 x7 : ffffffc008aea2e0 x6 : 0000000000000001
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.916394] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.923594] x2 : ffffff807fb89080 x1 : ffffffc0770cb000 x0 : 000000000000003f
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.930798] Call trace:
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.933316]  dev_watchdog+0x330/0x33c
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.937048]  call_timer_fn.constprop.0+0x20/0x80
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.941736]  __run_timers.part.0+0x208/0x284
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.946075]  run_timer_softirq+0x38/0x70
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.950068]  _stext+0x10c/0x28c
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.953280]  __irq_exit_rcu+0xdc/0xfc
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.957015]  irq_exit+0xc/0x1c
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.960142]  handle_domain_irq+0x60/0x8c
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.964137]  gic_handle_irq+0x50/0x120
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.967958]  call_on_irq_stack+0x28/0x44
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.971952]  do_interrupt_handler+0x4c/0x54
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.976204]  el1_interrupt+0x2c/0x4c
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.979853]  el1h_64_irq_handler+0x14/0x20
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.984019]  el1h_64_irq+0x74/0x78
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.987491]  arch_cpu_idle+0x14/0x20
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.991138]  do_idle+0xc0/0x150
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.994355]  cpu_startup_entry+0x20/0x60
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258902.998347]  rest_init+0xc4/0xd0
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258903.001648]  arch_call_rest_init+0xc/0x14
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258903.005729]  start_kernel+0x5b4/0x5d4
Tue Mar 28 08:48:10 2023 kern.debug kernel: [258903.009461]  __primary_switched+0xa0/0xa8
Tue Mar 28 08:48:10 2023 kern.warn kernel: [258903.013542] ---[ end trace 333227c911159d38 ]---
Tue Mar 28 08:48:10 2023 kern.err kernel: [258903.018249] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
Tue Mar 28 08:48:10 2023 kern.info kernel: [258903.157617] mtk_soc_eth 15100000.ethernet eth0: Link is Down
Tue Mar 28 08:48:11 2023 kern.err kernel: [258903.690323] mtk_soc_eth 15100000.ethernet: warm reset failed
```

melroy89, 2023-04-01:
Now you know where the error occurs (dev_watchdog+0x330/0x33c, etc..), you could be able to use gdb with the `list` command, to pin-point the problem on code level. See also: https://wiki.ubuntu.com/Kernel/KernelDebuggingTricks#Using_GDB_to_find_the_location_where_your_kernel_panicked_or_oopsed.

Also for the record, I do see a very similar stack trace in this PR: https://github.com/openwrt/openwrt/pull/11990#issuecomment-1449988314

jtxnn678, 2023-04-01:
@melroy89 Thank you for point that out, though I am knowledgeable in C, C++ but I lacking experience in low level & kernel debugging, I will need sometime to read and learn.

I've read the link, and pulled updates from git master. It seems the 'hack' patch has been included. I will compile the latest commit and give it a try. Will report back later...

[image]

lvqier, 2023-04-14:
I experiencing same issue on Redmi AX6000, here's the kernel panic log from dmesg:

> [   29.513290] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[   29.520279] br-lan: port 3(phy0-ap0) entered blocking state
[   29.526109] br-lan: port 3(phy0-ap0) entered forwarding state
[  361.718463] ------------[ cut here ]------------
[  361.723340] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 4 timed out
[  361.730571] WARNING: CPU: 2 PID: 0 at dev_watchdog+0x330/0x33c
[  361.736657] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 chacha_neon cfg80211 slhc poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha hwmon crc_ccitt compat crypto_safexcel ip6_udp_tunnel udp_tunnel sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_ws2812b leds_gpio gpio_button_hotplug
[  361.809487] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.15.94 #0
[  361.815736] Hardware name: Xiaomi Redmi Router AX6000 (OpenWrt U-Boot layout) (DT)
[  361.823541] pstate: 40000005 (nZcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  361.830741] pc : dev_watchdog+0x330/0x33c
[  361.834998] lr : dev_watchdog+0x330/0x33c
[  361.839252] sp : ffffffc008c0bdb0
[  361.842810] x29: ffffffc008c0bdb0 x28: 0000000000000140 x27: 00000000ffffffff
[  361.850186] x26: 0000000000000000 x25: 0000000000000002 x24: ffffff80002f14c0
[  361.857561] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008ad6000
[  361.864936] x20: ffffff80002f1000 x19: 0000000000000004 x18: ffffffc008aea2e0
[  361.872312] x17: ffffffc0173f5000 x16: ffffffc008c0c000 x15: 0000000000000306
[  361.879687] x14: 0000000000000102 x13: ffffffc008c0bad8 x12: ffffffc008b422e0
[  361.887063] x11: 712074696d736e61 x10: ffffffc008b422e0 x9 : 0000000000000000
[  361.894438] x8 : ffffffc008aea290 x7 : ffffffc008aea2e0 x6 : 0000000000000001
[  361.901814] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[  361.909189] x2 : ffffff801feb3080 x1 : ffffffc0173f5000 x0 : 000000000000003f
[  361.916564] Call trace:
[  361.919256]  dev_watchdog+0x330/0x33c
[  361.923164]  call_timer_fn.constprop.0+0x20/0x80
[  361.928027]  __run_timers.part.0+0x208/0x284
[  361.932540]  run_timer_softirq+0x38/0x70
[  361.936707]  _stext+0x10c/0x28c
[  361.940093]  __irq_exit_rcu+0xdc/0xfc
[  361.944002]  irq_exit+0xc/0x1c
[  361.947304]  handle_domain_irq+0x60/0x8c
[  361.951473]  gic_handle_irq+0x50/0x120
[  361.955468]  call_on_irq_stack+0x28/0x44
[  361.959635]  do_interrupt_handler+0x4c/0x54
[  361.964062]  el1_interrupt+0x2c/0x4c
[  361.967884]  el1h_64_irq_handler+0x14/0x20
[  361.972224]  el1h_64_irq+0x74/0x78
[  361.975870]  arch_cpu_idle+0x14/0x20
[  361.979691]  do_idle+0xc0/0x150
[  361.983081]  cpu_startup_entry+0x20/0x60
[  361.987248]  secondary_start_kernel+0x130/0x140
[  361.992021]  __secondary_switched+0x50/0x54
[  361.996449] ---[ end trace 24ac3e152a450e88 ]---

I have met the same issue twice. The first time was on Mar 16, and the second time was on April 13. When the issue occurs, every ethernet port freezes while WIFI keeps working.
The issue keeps present all day both on Mar 16 and April 13. No luck with soft reboot or hard reset by unplugging power, the wired network keeps dropping into frozen minutes after boot. The most strange thing is that the issue gone away on the next day, looks like it never happens.

jtxnn678, 2023-04-14:
I have RedMi AX6000 as well, but it is used as dump AP which only provides WiFi access. Maybe because of absence of connected Ethernet device, I haven't had the network drop issue as BPI-R3, consider both of them had almost identical hardware.

melroy89, 2023-04-14:
I think there is a bug between the LAN and WAN ports causing the mtk_soc_eth transmit queue timed out. 
It's either a software bug in MediaTek MT7986A (ETH chip) or MediaTek MT7531AE (Switch).

Both the Banana BPI R3 as well as the previously mentioned RedMi AX6000 has the same MediaTek MT7531AE switch chip.

So that should be a clue..

frank-w, 2023-04-17:
@dangowrt @nbd168 any idea here? I got report for this too for mainline-kernel..i guess it is related to the multi queue patch (https://github.com/torvalds/linux/commit/f63959c7eec3151c30a2ee0d351827b62e742dcb) but i have no idea how to trace it

melroy89, 2023-04-17:
> @dangowrt @nbd168 any idea here? I got report for this too for mainline-kernel..i guess it is related to the multi queue patch ([torvalds/linux@f63959c](https://github.com/torvalds/linux/commit/f63959c7eec3151c30a2ee0d351827b62e742dcb)) but i have no idea how to trace it

There are only a limited dev_dbg lines. But you could try to enable kernel debugging for this file. Like so: https://stackoverflow.com/a/63682160/518879

Alternatively, you can recompile the kernel module with additional printf statements for debugging.

Or.. Try to revert the patch. And recompile. And validate if the issues are gone or not.

frank-w, 2023-04-17:
I do not even know how to trigger it...i have not seen it by myself...seems like it happens after some time running...i guess i have not seen it yet as i often reboot for different reasons...revert is not possible as there are commits fixing parts of it. Is my assumption right that problem is in tx_poll? The trace comes from watchdog which makes it hard to find the root cause

lvqier, 2023-04-29:
The issue happened on my Rexmi AX6000 again today. I have another device Redmi AC2100 connected as AP to lan2 of Redmi AX6000. It seems that after disconnecting the Redmi AC2100, the issue disappears. My Redmi AC2100 is running with OpenWrt 22.03.4. I have also checked the state of Redmi AC2100, ethernet ports is freezed as well (but no kernel panic log). I can confirm that the issue gone away after rebooting Redmi AC2100. I think there's something wrong with OpenWrt on my Redmi AC2100, it triggers this issue on Redmi AX6000.

jtxnn678, 2023-04-29:
> The issue happened on my Rexmi AX6000 again today.
@lvqier Please include your Openwrt version so everybody can see what versions are still effected. Thank you.

**UPDATE:**
The last commit  (r2258x) I've tried for 12 days, and I didn't get disconnection. Now I am running commit r22649. Hope it is fixed.

lvqier, 2023-05-01:
> Please include your Openwrt version so everybody can see what versions are still effected.

It is OpenWrt SNAPSHOT r22652-9a26669510 running on my Redmi AX6000. Sorry for the inconvenience.

jtxnn678, 2023-05-01:
> It is OpenWrt SNAPSHOT r22652-9a26669510 running on my Redmi AX6000.

You are ahead of me. I also have AX6000 running as dump AP, I didn't experiencing disconnection issue with it. However, it only serves Wifi and without any Ethernet device attached.

[image]
[image]

Christopher876, 2023-06-14:
This problem seems to still happen. I'm using a Banana Pi R3 and it exhibits this bad behavior every 12-48 hours. LAN and WAN stops working when it happens. As stated by the OP, only a hard reboot brings it back

```
Wed Jun 14 19:17:09 2023 kern.info kernel: [38347.150721] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 3 timed out
Wed Jun 14 19:17:09 2023 kern.warn kernel: [38347.157695] WARNING: CPU: 1 PID: 0 at dev_watchdog+0x330/0x33c
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.163519] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 chacha_neon cfg80211 slhc sfp poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcurve25519_generic libcrc32c libchacha crc_ccitt compat crypto_safexcel pwm_fan i2c_gpio i2c_algo_bit ip6_udp_tunnel udp_tunnel sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.241908] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 5.15.114 #0
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.247984] Hardware name: Bananapi BPI-R3 (DT)
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.252497] pstate: 40400005 (nZcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.259440] pc : dev_watchdog+0x330/0x33c
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.263436] lr : dev_watchdog+0x330/0x33c
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.267429] sp : ffffffc00800bdb0
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.270726] x29: ffffffc00800bdb0 x28: 0000000000000140 x27: 00000000ffffffff
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.277842] x26: 0000000000000000 x25: 0000000000000001 x24: ffffff8000cf24c0
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.284958] x23: 0000000000000000 x22: 0000000000000001 x21: ffffffc008af6000
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.292073] x20: ffffff8000cf2000 x19: 0000000000000003 x18: ffffffc008b0a338
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.299188] x17: ffffffc0770bb000 x16: ffffffc008008000 x15: 0000000000000411
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.306303] x14: 000000000000015b x13: ffffffc00800bad8 x12: ffffffc008b62338
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.313419] x11: 712074696d736e61 x10: ffffffc008b62338 x9 : 0000000000000000
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.320533] x8 : ffffffc008b0a2e8 x7 : ffffffc008b0a338 x6 : 0000000000000001
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.327649] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.334763] x2 : ffffff807fb99080 x1 : ffffffc0770bb000 x0 : 000000000000003f
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.341878] Call trace:
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.344311]  dev_watchdog+0x330/0x33c
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.347957]  call_timer_fn.constprop.0+0x20/0x80
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.352559]  __run_timers.part.0+0x208/0x284
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.356813]  run_timer_softirq+0x38/0x70
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.360719]  _stext+0x10c/0x28c
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.363846]  __irq_exit_rcu+0xdc/0xfc
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.367496]  irq_exit+0xc/0x1c
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.370536]  handle_domain_irq+0x60/0x8c
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.374443]  gic_handle_irq+0x50/0x120
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.378180]  call_on_irq_stack+0x20/0x34
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.382087]  do_interrupt_handler+0x4c/0x54
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.386252]  el1_interrupt+0x2c/0x4c
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.389814]  el1h_64_irq_handler+0x14/0x20
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.393892]  el1h_64_irq+0x78/0x7c
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.397278]  arch_cpu_idle+0x14/0x20
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.400838]  do_idle+0xc0/0x150
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.403967]  cpu_startup_entry+0x24/0x60
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.407875]  secondary_start_kernel+0x130/0x140
Wed Jun 14 19:17:09 2023 kern.debug kernel: [38347.412389]  __secondary_switched+0x50/0x54
Wed Jun 14 19:17:09 2023 kern.warn kernel: [38347.416560] ---[ end trace bee3f9bad88ff936 ]---
Wed Jun 14 19:17:09 2023 kern.err kernel: [38347.421186] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
Wed Jun 14 19:17:10 2023 kern.info kernel: [38347.559518] mtk_soc_eth 15100000.ethernet eth0: Link is Down
Wed Jun 14 19:17:10 2023 kern.err kernel: [38348.096219] mtk_soc_eth 15100000.ethernet: warm reset failed
[... cut ...]
Wed Jun 14 19:17:10 2023 kern.info kernel: [38348.123544] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
Wed Jun 14 19:17:27 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) a8:46:9d:33:7d:71
Wed Jun 14 19:17:27 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 10.0.0.118 a8:46:9d:33:7d:71
Wed Jun 14 19:18:10 2023 kern.warn kernel: [38408.214999] page_pool_release_retry() stalled pool shutdown 64 inflight 60 sec
```

I'm on version `OpenWrt SNAPSHOT r23248-21f713d5ab / LuCI Master git-23.118.79121-6fb185f`, so way newer than the versions people are using here and it still happens

jtxnn678, 2023-06-15:
@Christopher876 Did you have NVME installed?
My BPI-R3 was running as main router which serves DHCP, and NVME storage as Samba server. Occasionally update the firmware, the issue is a hit or miss, some versions are more stable than other, can't say which one, its pretty frustrating.

Now, I bought a used HP EliteDesk Mini PC (i5-10500T, 16GB DDR4) and use it for the Samba server. BPI-R3 is only act as main router with DHCP server and WiFi (removed NVME). I got less disconnection with it, but it still exists.

Its strange that my RedMi AX6000 never had such issue regardless the snapshot versions, but it only function as dumb AP with 1 Ethernet connected to BPI-R3.

Just try different commit, and stick to the most reliable one and do not update once you found it.

NovaPixell, 2023-06-17:
I have this issue on Zyxel EX-5601. Did you figure out how to fix it?

jtxnn678, 2023-06-17:
> When I first got the board on this same version, it didn't show this behavior for 6 days and then it did it the first time.

My longest uptime was 21 days before got disconnected, its really hard to judge for how long it is needed to be considered as "stable".  :-)

Christopher876, 2023-06-17:
For me, I would consider it stable when it can have the same uptime as my previous router not running OpenWRT. I expect it to work well "forever" honestly. That router had an uptime of 2 years+.

I have tried the version you suggested, it doesn't seem to be any better for me. I did oddly enough have a disconnect about 6 hours after install.

For now I have to run two separate routers because this issue has become a problem for my job as well now.

anonymous-one, 2023-06-17:
i apologize for the slightly hijack, but:

dang, what a shame this mediatek suffers from this bug. i am running 2 aps that use the ipq807x chipset (one 72 and one 74)... and i get a strange 'once every 30-45 days' bug as well... wont got into details as its not relevant to this but all i will say is it requires a complete reboot to clear... i was debating on swapping them out for something that is built around this mediatek chipset instead. prices have gone down quite a bit and i was about to pull the trigger on 2 x redmi ax6 off aliexpress. guess i will hold off until this is resolved.

fingers crossed someone here is able to figure this out.

anonymous-one, 2023-06-17:
btw, if i may suggest for those experiencing this running a bpi r3:

a few ifs but:

if you are only concerned about a single (or two i guess...) ethernet port being affected, eg: you are using one or two of the eth ports... a possible (very dirty) workaround might be to populate the 2 sfp ports and use those instead of the onboard eth ports...

this looks related to the ethernet port(s) connected to the mtk chipset (guessing by the mtk_soc_eth)... so maybe using the sfp ports you will essentially bypass them and avoid this bug.

Christopher876, 2023-06-17:
@anonymous-one eth0 is in that reset. I don't think that using the sfp ports will bypass this
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP qlen 1000
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-wan state DOWN qlen 1000
4: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-wan state UP qlen 1000
5: lan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
6: lan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
7: lan3@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
8: lan4@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
9: sfp2@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
12: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
```

dangowrt, 2023-06-17:
Two things come to mind:
 * Looks like there is a bug in reset-dependency handling. @LorenzoBianconi may know more.
 * The fact that it doesn't happen on the Xiaomi device could very simply be related to the R3 being a development board without heatsinks, and esp. if inside a case it may easily overheat... If you haven't done that yet, add some heatsinks or even a fan, and then see if this resolves the issue.

anonymous-one, 2023-06-17:
speaking as someone who has only done a bit of window shopping (eg: i dont own this...)...

what a shame they dont ship it with this pre-installed... ummmm.... whyyyy?

[image]

> Two things come to mind:
>
> * Looks like there is a bug in reset-dependency handling. @LorenzoBianconi may know more.
> * The fact that it doesn't happen on the Xiaomi device could very simply be [...]

jtxnn678, 2023-06-18:
> if inside a case it may easily overheat...
Its not overheat related, I almost have all the chips covered with copper heatsink as soon as I received the device, it was sit on top of a laptop cooling pad, and now I've installed a exhaust fan inside the case too.
[https://files.digitalof.me//api/shares/YxMDY4N/files/246c13e3-91b0-4a75-a4c4-eeb428929f12](url)

melroy89, 2023-06-18:
> > if inside a case it may easily overheat...
> 
> Its not overheat related, I almost have all the chips covered with copper heatsink as soon as I received the device, it was sit on top of a laptop cooling pad, and now I've installed a exhaust fan inside the case too.

I agree this is NOT an overhead issue people. This is a kernel module crash.

Christopher876, 2023-06-23:
So I have been experimenting, disabling Adblock seems to have made things more stable? My Banana Pi R3 has been running for 7+ days now. 

Turning on Adblock makes it exhibit the crash like I was experiencing before and also seeing the 150 max dns connections issue before the crash.

I am not sure why this is but I've just moved adblocking to my raspberry pi and pointed my stuff to that and everything seems to be happy. But it appears that adblocking is relating to the crash and something that it does with DNS (perhaps causing something to hang?).

melroy89, 2023-06-24:
> Adblock

So you mean https://openwrt.org/packages/pkgdata/adblock ?

Christopher876, 2023-06-24:
> > Adblock
> 
> 
> 
> So you mean https://openwrt.org/packages/pkgdata/adblock ?

Yes that one and Adguard Home caused the problem in my case

HS1542, 2023-06-26:
I am experiencing similar crashes when a power cycle is the only way of bringing the bpi-r3 back online.
It has nothing to do with other services as I only use it as a simple AP that bridges 3 different SSIDs to three VLANs through the eth1 (sfp1), since this is the only port supporting VLANs? 
And I have not installed anything other than vanilla v23.05.0-rc1

However, the frequency of crashes might be impacted somehow by the client devices.
I say this because we usually experience a crash every 1-2 weeks, but whenever we have guests bringing new clients, the frequency of crashs increase. Last weekend we had crashes every 10minutes-3hours.

anonymous-one, 2023-06-26:
Just a quick FYI... it is highly unlikely that anything in the user mode (eg: Adblock) has anything to do with this...

Unless adblock is making some weird calls to the driver (...its not) OR the increase (ever so slightly) in traffic or cpu usage (again, ever so slight) is the cause.

...again, a bit of a shame. I am still expericing my ipq8072 / ipq8074 bug... I have two bpi r3s in my amazon basket as we speak but am going to hold off on pulling the trigger.

Christopher876, 2023-06-26:
> Just a quick FYI... it is highly unlikely that anything in the user mode (eg: Adblock) has anything to do with this...

If transmission is frozen for a period of time, that can cause the kernel to restart the driver, which it's very likely to be frozen due to the message about the max dns connections if there is another dns connection that is about to be made. At least this is what happened in my case with Adblock.

I'm not entirely sure how it even gets up to 150 simultaneous connections at once since I've been doing packet captures and there's never a bunch of them all at once. Unless there are dns sessions not getting cleaned up.

Here's a comment on how it can happen. It's comment 2 that has a link and quote of the relevant scenarios. https://bbs.archlinux.org/viewtopic.php?id=261832

Christopher876, 2023-06-27:
> i may be way off, but, in order:
> 
> 
> 
> a) network stack crashes (eg: mtk_soc_eth)
> 
> b) bpi r3 is able to receive traffic, but not send it
> 
> c) your dns connections start stacking because your dns daemon is able to receive traffic, but not send it
> 
> d) 150+ connections message pops up due to all your pending connections
> 
> e) due to some timeouts / retries in the kernel, the kernel message pops up after the 150+ message... even though the network stack crashed before hand
> 
> 
> 
> just a guess, but a strong possibility thats what is happening.

Probably, seems likely. There is likely a bug somewhere in the reconfiguration of the driver.

I don't know much about network drivers in the kernel though to start putting logs in and figuring out where it is (Could be a "fun" weekend project?). And worse since I can't just trigger the bug in a minute or 2 to test.

Though, if anyone has any ideas on how to trigger it quickly, that would be useful

anonymous-one, 2023-06-27:
i'm on my last try to resolve my IPQ8074/IPQ8072 bug. without going into too much detail, between 1-4 weeks of uptime i get an issue where all my wifi clients will start to lag big time. pings of 1000-10000+ms, including right out ping losses. usually this can't be resolved until i do a full out reboot.

as mentioned, pretty much tried everything and am down to the last straw. i have 3 of these devices (2 x IPQ8072, 1 x IPQ8074).

if it pops up again over the next 30 days i'll be ordering a bpi r3 will start to migrate everything over slowly.

my topology is fairly standard, wan comes in on the ground floor + each floor gets its own access point (so 1 gateway + 2 dumb APs in total).

i mention all this because if / once i get the bpi r3 ill jump in the ring and see what i can do to help resolve this. that being said, i'll be going the sfp route right off the bat so my setup might differ a bit from most in this thread.

again, best of luck figuring this out. will keep an eye + post if / when my bpi r3 is on its way here.

> > i may be way off, but, in order:
> > a) network stack crashes (eg: mtk_soc_eth)
> > b) bpi r3 is able to receive traffic, but not send it
> > c) your dns connections start stacking because your dns daemon is able to receive traffic, but not send it
> > d) 150+ connections message pops up due to all your pending connections
> > e) due to some timeouts / retries in the kernel, the kernel message pops up after the 150+ message... even though the network stack crashed before hand
> > just a guess, but a strong possibility thats what is happening.
> 
> Probably, seems likely. There is likely a bug somewhere in the reconfiguration of the driver.
> 
> I don't know much about network drivers in the kernel though to start putting logs in and figuring out where it is (Could be a "fun" weekend project?). And worse since I can't just trigger the bug in a minute or 2 to test.
> 
> Though, if anyone has any ideas on how to trigger it quickly, I would try it. Kind of miss Adblock at the router level (even if this wasn't the root cause, does help not having it in case of crashes).

wulfy23, 2023-07-01:
hmmmmm... fwiw came across this today ( usb3 rtl gigabit )... grain of salt but may point somewhere higher up the stack / arm acpi? / random other guess here?

```
[969225.296035] NETDEV WATCHDOG: eth775c13 (r8152): transmit queue 0 timed out
[969225.296089] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:525 dev_watchdog+0x240/0x250

[969225.297240] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C         6.1.32 #0
[969225.297252] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[969225.297258] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[969225.297270] pc : dev_watchdog+0x240/0x250
[969225.297281] lr : dev_watchdog+0x240/0x250
[969225.297291] sp : ffffffc008003e20
[969225.297295] x29: ffffffc008003e20 x28: ffffffc008c34000 x27: ffffffc008003f00
[969225.297312] x26: 00000000000000e0 x25: ffffffc008c34648 x24: 00000000000000a0
[969225.297329] x23: 0000000000000000 x22: ffffffc008b16000 x21: 0000000000000000
[969225.297345] x20: ffffff8043baa000 x19: ffffff8043baa4c0 x18: 0000000000000000
[969225.297360] x17: ffffffc07707b000 x16: ffffffc008000000 x15: ffffffffffffffff
[969225.297375] x14: ffffffc008c101ea x13: 74756f2064656d69 x12: 7420302065756575
[969225.297390] x11: 712074696d736e61 x10: 7274203a29323531 x9 : ffffffc008b2a8a8
[969225.297406] x8 : ffffffc008003c50 x7 : ffffffc008b870d0 x6 : 0000000000000001
[969225.297421] x5 : 0000000000000000 x4 : 0000000000000040 x3 : 0000000000000004
[969225.297436] x2 : 0000000000000000 x1 : 0000000000000005 x0 : 000000000000003e
[969225.297451] Call trace:
[969225.297456]  dev_watchdog+0x240/0x250
[969225.297467]  call_timer_fn.constprop.0+0x24/0x80
[969225.297485]  __run_timers.part.0+0x238/0x284
[969225.297499]  run_timer_softirq+0x3c/0x7c
[969225.297513]  _stext+0x10c/0x290
[969225.297522]  ____do_softirq+0x10/0x20
[969225.297532]  call_on_irq_stack+0x24/0x58
[969225.297542]  do_softirq_own_stack+0x1c/0x30
[969225.297551]  __irq_exit_rcu+0xc4/0xf0
[969225.297563]  irq_exit_rcu+0x10/0x20
[969225.297572]  el1_interrupt+0x38/0x5c
[969225.297581]  el1h_64_irq_handler+0x18/0x2c
[969225.297589]  el1h_64_irq+0x68/0x6c
[969225.297597]  arch_cpu_idle+0x18/0x2c
[969225.297606]  default_idle_call+0x30/0x74
[969225.297619]  do_idle+0x230/0x294
[969225.297627]  cpu_startup_entry+0x24/0x30
[969225.297635]  kernel_init+0x0/0x130
[969225.297644]  arch_post_acpi_subsys_init+0x0/0x18
[969225.297652]  start_kernel+0x568/0x58c
[969225.297659]  __primary_switched+0xb4/0xbc
[969225.297671] ---[ end trace 0000000000000000 ]---
[969225.297690] r8152 2-2:1.0 eth775c13: Tx timeout
[969225.300798] r8152 2-2:1.0 eth775c13: Tx status -2
[969225.300892] r8152 2-2:1.0 eth775c13: Tx status -2
[969225.457225] usb 2-2: reset SuperSpeed USB device number 3 using xhci_hcd
```

not seen on 5.x ever... and timelines match up with above discussion...

NovaPixell, 2023-07-16:
Same issue here, on Zyxel EX5601-T0, #13122 
Hasn't a fix been identified?

NerdyProjects, 2023-07-17:
I wonder if my problem on WAX206 (MT7622) is related? https://forum.openwrt.org/t/transmit-queue-timed-out-on-netgear-wax206-and-fritzbox-7530-nearly-at-same-moment/165972/2

The affected device also had adblock (+2x VPN + vlan + some wifis) running. After reading this, I am not 100% sure if I had ethernet connection through the other ports, but at least the Ethernet LEDs were still working.
Replacement device is running a plain snapshot OpenWRT and is fine so far.

As my device was fine for three months, I also expect a bad traffic/usage pattern to expose the bug as it was reproduced about 5 times in 1 hour this morning.

NovaPixell, 2023-07-17:
@jtxnn678 have you tryed to disable Hardware/Software flow offloading? As I said [here](https://github.com/openwrt/openwrt/issues/13122#issuecomment-1638793757).

Christopher876, 2023-07-17:
> @jtxnn678 have you tryed to disable Hardware/Software flow offloading? As I said [here](https://github.com/openwrt/openwrt/issues/13122#issuecomment-1638793757).

How do you do so? Does it require recompiling the kernel? Or is there a simple command I am missing?

I actually just want to use mine as an AP now since I can’t return it anymore. Gave up on it after the issue happened again and switched to an x86 opnsense box. Getting rid of all services just made it a little more stable rather than fixing the issue.

NovaPixell, 2023-07-17:
> How do you do so? Does it require recompiling the kernel? Or is there a simple command I am missing?
>
> I actually just want to use mine as an AP now since I can’t return it anymore. Gave up on it [...]

Using the LuCi Web GUI, in the Firewall section you can disable these options.

jtxnn678, 2023-07-19:
@NovaPixell 
> @jtxnn678 have you tryed to disable Hardware/Software flow offloading? As I said [here](https://github.com/openwrt/openwrt/issues/13122#issuecomment-1638793757).

Yeah, I tried that but not helping. Right now, I literally use it as main router with DHCP server, and provide Wifi AP only now. But it still give disconnection issue after awhile like 20+ days.

Disable/removed:

- NVME storage
- External USB storage
- adblock, samba.

Its running as vanilla Openwrt with no additional package installed, tried countless updated version but still the issue persists.

[119 more replies]