openwrt/openwrt #12145: mwlwifi crashes kernel

M95D, 2023-03-09:
### Describe the bug

Linksys WRT-1900ACS, OpenWrt SNAPSHOT, r22212
I get random reboots with this error:
```
[23317.209424] 8<--- cut here ---
[23317.212521] Unable to handle kernel NULL pointer dereference at virtual address 000000a8
[23317.220666] pgd = 19b3b006
[23317.223387] [000000a8] *pgd=00000000
[23317.226979] Internal error: Oops: 5 [#1] SMP ARM
[23317.231616] Modules linked in: nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack ipt_REJECT xt_time xt_tcpudp xt_physdev xt_multiport xt_mark xt_mac xt_limit xt_iprange xt_iface xt_comment xt_TRACE xt_TEE xt_TCPMSS xt_PROTO xt_LOG ppp_async nfnetlink_queue nfnetlink nf_reject_ipv4 nf_log_syslog nf_dup_ipv6 nf_dup_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_raw iptable_mangle iptable_filter ip6table_raw ip_tables br_netfilter rtl2832 regmap_i2c dvb_usb_rtl28xxu cxd2841er r820t dvb_usb_v2 dvb_core mc i2c_mux mwlwifi mac80211 cfg80211 compat cryptodev ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv3 nfs
[23317.231783]  rpcsec_gss_krb5 cifs cifs_md4 cifs_arc4 dns_resolver zram des_generic gpio_button_hotplug dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax
[23317.332116] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.15.98 #0
[23317.338150] Hardware name: Marvell Armada 380/385 (Device Tree)
[23317.344093] PC is at 0xbf20c824 [mwlwifi@da1210cd+0x22000]
[23317.349604] LR is at 0xbf20c818 [mwlwifi@da1210cd+0x22000]
[23317.355112] pc : [<bf20c824>]    lr : [<bf20c818>]    psr: 60000113
[23317.361405] sp : c0d01e00  ip : 00000001  fp : c40e6cc0
[23317.366650] r10: c40e6260  r9 : 00000022  r8 : ffffffff
[23317.371895] r7 : 00000000  r6 : c4208000  r5 : c41b1800  r4 : 00000000
[23317.378448] r3 : 00000000  r2 : 0000ffff  r1 : ffffffff  r0 : 000000a8
[23317.385002] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[23317.392168] Control: 10c5387d  Table: 1540804a  DAC: 00000051
[23317.397937] Register r0 information: non-paged memory
[23317.403011] Register r1 information: non-paged memory
[23317.408082] Register r2 information: non-paged memory
[23317.413154] Register r3 information: NULL pointer
[23317.417877] Register r4 information: NULL pointer
[23317.422600] Register r5 information: non-slab/vmalloc memory
[23317.428284] Register r6 information: non-slab/vmalloc memory
[23317.433967] Register r7 information: NULL pointer
[23317.438689] Register r8 information: non-paged memory
[23317.443761] Register r9 information: non-paged memory
[23317.448833] Register r10 information: non-slab/vmalloc memory
[23317.454603] Register r11 information: non-slab/vmalloc memory
[23317.460373] Register r12 information: non-paged memory
[23317.465531] Process swapper/0 (pid: 0, stack limit = 0x1ea34c38)
[23317.471563] Stack: (0xc0d01e00 to 0xc0d02000)
[23317.475939] 1e00: 00000000 c40e0560 c40e1860 10624dd3 c427f858 c40e63e0 e06cd35c c0c695c0
[23317.484153] 1e20: c40e62c8 c40e6cc8 c40e64a0 00000001 c40e6400 ffffffff 00000000 c40e63fc
[23317.492366] 1e40: 00000000 c40e6400 dfbd42c8 c0c69180 00000040 00000100 c0d01e80 c013589c
[23317.500579] 1e60: 00000000 c0d03098 c0d00000 c0d03080 40000006 00000018 00000100 c01012b4
[23317.508792] 1e80: c0d01ec0 00000052 c0d03080 c0c67330 0000000a c0c672d0 c0c69180 c0c69180
[23317.517005] 1ea0: 00231f46 c0d03d00 04200002 ffffe000 e080210c c0d01f00 1d51aac3 c0c687c4
[23317.525217] 1ec0: 00000000 00000041 00000000 e080210c c0d01f00 1d51aac3 c0d01ff4 c0135d94
[23317.533430] 1ee0: c0c687c4 c017c3fc c0d0a2e0 c0d01f20 e0802100 c0c687d0 e080210c c05e1ec0
[23317.541643] 1f00: c01074f8 60000013 ffffffff c0d01f54 6c4c635a c0d00000 1d51aac3 c0100b7c
[23317.549856] 1f20: 1db321f4 00000000 00000001 c01169e0 c0d00000 c0d09f28 00000000 c0d09f58
[23317.558068] 1f40: 6c4c635a 315f6a72 1d51aac3 c0d01ff4 d548803c c0d01f70 c01074f4 c01074f8
[23317.566281] 1f60: 60000013 ffffffff 00000051 c018cb0c c0d00000 c015f900 000000e6 ccc61056
[23317.574494] 1f80: c0d4adf8 c0d6e010 6c4c635a c015fca0 9820a81b c0c01a78 ffffffff ffffffff
[23317.582707] 1fa0: 00000000 c0c00a0c 00000000 c0af41a0 c0d09ec8 c0d0d260 dfbec300 00000000
[23317.590919] 1fc0: 00000000 c0c568d8 518715cb 00000000 00000000 c0c00460 00000051 10c0387d
[23317.599131] 1fe0: 00000210 01157480 414fc091 10c5387d 00000000 00000000 00000000 00000000
[23317.607349] Function entered at [<bf20c824>] from [<c013589c>]
[23317.613207] Function entered at [<c013589c>] from [<c01012b4>]
[23317.619064] Function entered at [<c01012b4>] from [<c0135d94>]
[23317.624920] Function entered at [<c0135d94>] from [<c017c3fc>]
[23317.630777] Function entered at [<c017c3fc>] from [<c05e1ec0>]
[23317.636634] Function entered at [<c05e1ec0>] from [<c0100b7c>]
[23317.642490] Exception stack(0xc0d01f20 to 0xc0d01f68)
[23317.647563] 1f20: 1db321f4 00000000 00000001 c01169e0 c0d00000 c0d09f28 00000000 c0d09f58
[23317.655775] 1f40: 6c4c635a 315f6a72 1d51aac3 c0d01ff4 d548803c c0d01f70 c01074f4 c01074f8
[23317.663986] 1f60: 60000013 ffffffff
[23317.667487] Function entered at [<c0100b7c>] from [<c01074f8>]
[23317.673344] Function entered at [<c01074f8>] from [<c015f900>]
[23317.679201] Function entered at [<c015f900>] from [<c015fca0>]
[23317.685057] Function entered at [<c015fca0>] from [<c0c01a78>]
[23317.690915] Code: eb3c1d8c e28400a8 f57ff05b f590f000 (e1906f9f) 
[23317.697063] ---[ end trace 957d6322223c18ab ]---
[23317.701711] Kernel panic - not syncing: Fatal exception in interrupt
[23317.708093] CPU1: stopping
[23317.710813] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D           5.15.98 #0
[23317.718243] Hardware name: Marvell Armada 380/385 (Device Tree)
[23317.724187] Function entered at [<c010e454>] from [<c010a534>]
[23317.730045] Function entered at [<c010a534>] from [<c05c92c0>]
[23317.735902] Function entered at [<c05c92c0>] from [<c010c908>]
[23317.741758] Function entered at [<c010c908>] from [<c010c954>]
[23317.747614] Function entered at [<c010c954>] from [<c01825e4>]
[23317.753469] Function entered at [<c01825e4>] from [<c017c3f8>]
[23317.759326] Function entered at [<c017c3f8>] from [<c05e1ec0>]
[23317.765183] Function entered at [<c05e1ec0>] from [<c0100b7c>]
[23317.771038] Exception stack(0xc144ff80 to 0xc144ffc8)
[23317.776112] ff80: 017771ae 00000000 00000001 c01169e0 c144e000 c0d09f28 00000001 c0d09f58
[23317.784325] ffa0: 0000406a 414fc091 00000000 00000000 c841e03c c144ffd0 c01074f4 c01074f8
[23317.792536] ffc0: 60000013 ffffffff
[23317.796038] Function entered at [<c0100b7c>] from [<c01074f8>]
[23317.801894] Function entered at [<c01074f8>] from [<c015f900>]
[23317.807750] Function entered at [<c015f900>] from [<c015fca0>]
[23317.813605] Function entered at [<c015fca0>] from [<00101510>]
[23317.819466] Rebooting in 3 seconds..
```

### OpenWrt version

r22212

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1900ACS

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

anomeome, 2023-03-10:
Maybe related to the openssl issue, try moving forward to an image of at least [commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=c75cd5f6028da6ceb1fb3438da93e2305cd720b1)

M95D, 2023-03-11:
How could OpenSSL crash the kernel? I don't understand. Kernel uses internal crypto, not a user-mode library. Even if it could, that means a DoS vulnerability in the kernel.

anomeome, 2023-03-11:
[issue](https://github.com/openwrt/openwrt/issues/12084)

M95D, 2023-05-06:
I no longer use mwlwifi.

closed 2023-05-06 by M95D