openwrt/openwrt #12146: Can't compile Broadcom-wl using master branch

leezhiwei, 2023-03-10:
### Describe the bug

I have an issue while compiling a image for my DIR-865L for development. While using the wl profile using make menuconfig then a make, the following errors occur.
`/home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/driver/wl_iw.c: In function 'dev_wlc_ioctl':
/home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/driver/wl_iw.c:121:14: error: implicit declaration of function 'get_fs'; did you mean 'sget_fc'? [-Werror=implicit-function-declaration]
  121 |         fs = get_fs();
      |              ^~~~~~
      |              sget_fc
/home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/driver/wl_iw.c:121:14: error: incompatible types when assigning to type 'mm_segment_t' from type 'int'
/home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/driver/wl_iw.c:122:9: error: implicit declaration of function 'set_fs'; did you mean 'sget_fc'? [-Werror=implicit-function-declaration]
  122 |         set_fs(KERNEL_DS);
      |         ^~~~~~
      |         sget_fc
/home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/driver/wl_iw.c:122:16: error: 'KERNEL_DS' undeclared (first use in this function); did you mean 'KERNFS_NS'?
  122 |         set_fs(KERNEL_DS);
      |                ^~~~~~~~~
      |                KERNFS_NS
/home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/driver/wl_iw.c:122:16: note: each undeclared identifier is reported only once for each function it appears in
/home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/driver/wl_iw.c: In function 'wl_iw_set_freq':
/home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/driver/wl_iw.c:393:9: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
  393 |         if (fwrq->m > 4000 && fwrq->m < 5000)
      |         ^~
/home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/driver/wl_iw.c:396:17: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
  396 |                 chan = wf_mhz2channel(fwrq->m, sf);
      |                 ^~~~
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:289: /home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/driver/wl_iw.o] Error 1
make[4]: *** [Makefile:1904: /home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/driver] Error 2
make[3]: *** [Makefile:185: /home/lee/openwrt/build_dir/target-mipsel_mips32_musl/linux-bcm47xx_generic/broadcom-wl-5.10.56.27.3/.built] Error 2
make[3]: Leaving directory '/home/lee/openwrt/package/kernel/broadcom-wl'
time: package/kernel/broadcom-wl/compile#2.07#0.76#2.59
    ERROR: package/kernel/broadcom-wl failed to build.
make[2]: *** [package/Makefile:116: package/kernel/broadcom-wl/compile] Error 1
make[2]: Leaving directory '/home/lee/openwrt'
make[1]: *** [package/Makefile:110: /home/lee/openwrt/staging_dir/target-mipsel_mips32_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lee/openwrt'
make: *** [/home/lee/openwrt/include/toplevel.mk:231: world] Error 2
`

### OpenWrt version

SNAPSHOT

### OpenWrt target/subtarget

bcm47xx/mips32

### Device

DIR-865L (generic device)

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mirh, 2024-05-16:
Presumably this was fixed in #12545?
Putting even aside that the latest main [dropped](http://lists.openwrt.org/pipermail/openwrt-devel/2024-January/042162.html) it..

leezhiwei, 2024-05-16:
Yep. Understood. Wow. This issue has been open for.... A while now.