openwrt/openwrt #12148: ramips/mt76x8 winbond w25q256 32M spi-nor-flash kernel dump on HLK-7628N

joerg555, 2023-03-10:
### Describe the bug

- After booting my device i get a kernel dump
- i get "enabling reset hack; may not recover from unexpected reboots" but 4-Byte opcodes are supported

see "Steps to reproduce"

`[    0.383229] printk: bootconsole [early0] disabled

[    0.402672] 10000d00.uart1: ttyS1 at MMIO 0x10000d00 (irq = 29, base_baud = 2500000) is a 16550A

[    0.421735] spi-mt7621 10000b00.spi: sys_freq: 193333333

[    0.454178] ------------[ cut here ]------------

[    0.463410] WARNING: CPU: 0 PID: 1 at drivers/mtd/spi-nor/core.c:2997 0x80362054

[    0.478071] enabling reset hack; may not recover from unexpected reboots

[    0.491337] Modules linked in:

[    0.497396] CPU: 0 PID: 1 Comm: swapper Not tainted 5.10.161 #0

[    0.509118] Stack : 8095806c 800609dc 806c0000 805c6edc 00000000 00000000 00000000 00000000

`
[hlk7688-bootlog.txt](https://github.com/openwrt/openwrt/files/10941622/hlk7688-bootlog.txt)

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt76x8

### Device

HILINK HLK-7628N

### Image kind

Official downloaded image

### Steps to reproduce

I Boot my device with serial console and get a kernel crash dump

### Additional info

There are following problems

1. The crash dump is new in openwrt 22.03, in 21.03 and earlier versions it's OK
2. My device has a winbond w25q256jveq and it supports SPI_NOR_4B_OPCODES. But the reading of the SFPT / BFPT dose not work in all my devices, i have checked all with printk in kernel, and the reading in "spi_nor_read_sfdp_dma_unsafe" in _drivers/mtd/spi-nor/sfdp.c_ allwaxs get 0xff. The kernel message _enabling reset hack; may not recover from unexpected reboots_ show this behavier. i think this must be a problem of the _drivers/spi/spi-mt7621.c driver.
3. I have checked all openwrt versions, only 18.06 without the FPT / BFPT check is ok. 18.06 has the old table with SPI_NOR_4B_OPCODES enabled, and it's working perfekt without "reset hack" and boot problems.
4. Openwrt 19.xx and newer have the "broken-flash-reset;" entry in dts, but it's only because SFPT / BFPT is not working, i have hilink HLK-7628N and HKL-7688A devices thince 2018 and they allways have w25q256jv or other flash without problems.
5. we need a patch to enable the old  SPI_NOR_4B_OPCODES feature with dts entry "w25q256-4b-opcodes;" in dts file, or we have to find the problem in spi-driver.

If you need hardware, i can send it to you, or i can make a patch for ramips-devices.

73 Joerg

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Bartel-C8, 2023-03-17:
Same here, follow! master custom build of openwrt:  OpenWrt SNAPSHOT, r22248-bf055fcdca

Bartel-C8, 2023-03-17:
Extra info, we have multiple HLK-7621A boards, some have the error, some not, with the same flashed image!

Only difference in boot logs:

The one with error above:
```
[    0.000000] Linux version 5.15.98 (bartel@ubuntu) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.2.0 r22248-bf055fcdca) 12.2.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Wed Mar 8 01:19:01 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
```

One without error above:
```
[    0.000000] Linux version 5.15.98 (bartel@ubuntu) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.2.0 r22248-bf055fcdca) 123
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:4
```

------------

The one with error above:
```
[    0.000000] Writing ErrCtl register=00007240
[    0.000000] Readback ErrCtl register=00007240
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 248676K/262144K available (7189K kernel code, 627K rwdata, 1460K rodata, 1272K init, 242K bss, 13468K reserved, 0K cma-reserved, 0K highmem)
```

One without error above:
```
[    0.000000] Writing ErrCtl register=000209d0
[    0.000000] Readback ErrCtl register=000209d0
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 248676K/262144K available (7189K kernel code, 627K rwdata, 1460K rodata, 1272K init, 242K bss, 13468K )
```

After this the error pops up in the boot log (or not)

joerg555, 2023-03-23:
Hi thanks for your anwer !
can you make a 
`dmesg | grep spi`
to see your flash chip ?

Bartel-C8, 2023-03-29:
```
[    1.168398] spi-mt7621 1e000b00.spi: sys_freq: 250000000
[    1.190770] WARNING: CPU: 0 PID: 1 at drivers/mtd/spi-nor/core.c:2853 spi_nor_init+0x17c/0x184
[    1.368373] [<8047cdc8>] spi_nor_init+0x17c/0x184
[    1.377727] [<8047d65c>] spi_nor_scan+0x7a4/0xba0
[    1.387079] [<8047e39c>] spi_nor_probe+0x94/0x310
[    1.466236] [<8049d8a0>] __spi_add_device+0x80/0x160
[    1.476109] [<8049d9e0>] spi_add_device+0x60/0x9c
[    1.485459] [<8049e384>] spi_register_controller+0x7f4/0xb34
[    1.496710] [<804a0f10>] mt7621_spi_probe+0x168/0x220
[    1.628370] spi-nor spi0.0: w25q256 (32768 Kbytes)
[    1.638018] 4 fixed-partitions partitions found on MTD device spi0.0
[    1.650711] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    1.665304] OF: Bad cell count for /palmbus@1e000000/spi@b00/flash@0/partitions
[    1.680155] Creating 4 MTD partitions on "spi0.0":
```

We have the W25Q256JVE on it.

Djfe, 2023-04-25:
Weird this was already fixed in all releases of 22 (I haven't checked 21)
https://github.com/openwrt/openwrt/commit/e6b3e77e6ef635688b810a9bba528d477998270e
https://github.com/openwrt/openwrt/commit/9a7e3d170cb7ea46ca0a34f4af7db1b3d1b68739

Bartel-C8, 2023-04-25:
@Djfe : The issue is present on 21.x, 22.x and even master.

The patches are actually present, I guess, as they expose the real issue, like @joerg555 mentions.

For some reason, the probing of `sfdp ` goes wrong. And that's why it falls back to 3byte opcodes?

Djfe, 2023-04-25:
Different flash chip but is this related?
https://github.com/openwrt/openwrt/commit/ed7c1a504eae6b220e01da330c62c47d299e2c5d

joerg555, 2023-04-25:
I have now a HLK-7621A, HLK-7688A and HLK-7628N.
all my boards with 32M flash "w25q256jveq" have this issue, may be is don't crash, but they all have reset problem.
I have one board (HLK-7628N), with 64M flash, and it is not affected with this patch.

Normaly the kernel driver has to be changed, if SFPT-Info is not available it must be 4-Byte adressing with extended register, but its verry hard to find the way for a offical kernel driver patch.
May be some one know how to ask for official kernel patch.

codehero, 2024-03-14:
This is not easily solved in the general case with a kernel patch.
Both the NOR chip and the MT7628 have to boot in 4B mode.
For the NOR this is just a simple CFI command to change the boot mode.
For the MT7628AN, this is a pullup on the CS1 bootstrap line to tell the microcode to perform 4B SPI commands.
No kernel or u-boot patch can make this pullup happen if it's not in the hardware.

joerg555, 2024-06-19:
You are right with 4B Modem but if you could not chenge the hardware, the Winbond 32M w25q256 can work in 3B mode with with extended register with my patch.

I think this patch will be good for Winbond too.
https://github.com/openwrt/openwrt/commit/ed7c1a504eae6b220e01da330c62c47d299e2c5d