openwrt/openwrt #12158: usbnet kernel fail

tinware, 2023-03-11:
### Describe the bug

**Device:** UniElec u7621-06 256m 16m (changed to 64m)
**System:** OpenWrt SNAPSHOT r22256-14334c222e / LuCI Master git-23.039.28596-41e9b8d
**Kernel:** 5.15.98
**Time the problem occurred:** from v19.07.x to now
**Occurrence of the problem:** during system startup or work
**Where there is no problem:** v19.07.2, v19.07.3

**Installed packages:** 
- kmod-usb-core 
- kmod-usb-net-cdc-ether 
- usb-modeswitch 
- libusb-1.0
- usbutils

**Connected devices:**
- Hub USB 2.0 + additional power
- e3372h-153 HiLink (4 pieces)

**Kernel log:**
```
[    0.000000] Linux version 5.15.98 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.2.0 r22256-14334c222e) 12.2.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Fri Mar 10 01:31:50 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is UniElec U7621-06 (64M flash)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s15856 r8192 d21008 u45056
[    0.000000] pcpu-alloc: s15856 r8192 d21008 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00071812
[    0.000000] Readback ErrCtl register=00071812
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 248320K/262144K available (7423K kernel code, 631K rwdata, 1496K rodata, 1252K init, 251K bss, 13824K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.016058] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.088351] pid_max: default: 32768 minimum: 301
[    0.097736] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.112151] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.130916] rcu: Hierarchical SRCU implementation.
[    0.140777] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.156388] smp: Bringing up secondary CPUs ...
[    0.166183] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.166212] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.166227] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.166275] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.226193] Synchronize counters for CPU 1: done.
[    0.288235] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[... 342 lines cut ...]
```

### OpenWrt version

r22256-14334c222e

### OpenWrt target/subtarget

ramips/mt7621

### Device

UniElec U7621-06 (64M flash)

### Image kind

Official downloaded image

### Steps to reproduce

Turn on and off the device from the power supply

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mrkiko, 2023-03-13:
If you actually reduced the available RAM memory on the system, then it seems your kernel seems not aware of that.
This may lead to problems.

mrkiko, 2023-03-13:
Oh, if you refer to falsh size, then sorry for the noise.