openwrt/openwrt #12159: ssh from emacs shell fails with SIGSEGV to ash, but ssh from xterm is OK

Steve-Newcomb, 2023-03-11:
### Describe the bug

Using Debian Linux emacs shell, I log into a TP-Link A7 V5 router running r22170-5e48c534f7, and I'm immediately logged off because the kernel kills `ash` with a SIGSEGV.  (I'm able to see what's being logged as this happens because I have no problem logging in from a Debian Linux xterm.)  

### OpenWrt version

r22170-5e48c534f7

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer A7 v5

### Image kind

Self-built image

### Steps to reproduce

Here's what happens in the Emacs shell:
```
230311 10:17 /persist/home/srn root@basil{1}# ssh rpc148
BusyBox v1.36.0 (2023-02-27 10:11:01 UTC) built-in shell (ash)
r22170-5e48c534f7
Connection to rpc148 closed.
230311 10:24 /persist/home/srn root@basil{1}# 
```

Here's what the `logread -f` says as this happens:

```
Sat Mar 11 10:24:10 2023 auth.info sshd[3511]: Accepted publickey for root from 192.168.148.205 port 47982 ssh2: RSA SHA256:xxxxx
Sat Mar 11 10:24:10 2023 kern.info kernel: [ 2187.842219] do_page_fault(): sending SIGSEGV to ash for invalid read access from 00000010
Sat Mar 11 10:24:10 2023 kern.info kernel: [ 2187.850942] epc = 0043a76b in busybox[400000+52000]
Sat Mar 11 10:24:10 2023 kern.info kernel: [ 2187.856171] ra  = 0043a5cf in busybox[400000+52000]
Sat Mar 11 10:24:10 2023 auth.info sshd[3511]: Received disconnect from 192.168.148.205 port 47982:11: disconnected by user
Sat Mar 11 10:24:10 2023 auth.info sshd[3511]: Disconnected from user root 192.168.148.205 port 47982
```
As you can see, I'm using an openssh server on the router.  

### Actual behaviour

Here's what happens in the Emacs shell:
```
230311 10:17 /persist/home/srn root@basil{1}# ssh rpc148
BusyBox v1.36.0 (2023-02-27 10:11:01 UTC) built-in shell (ash)
r22170-5e48c534f7
Connection to rpc148 closed.
230311 10:24 /persist/home/srn root@basil{1}# 
```

By logging in from an xterm, I'm able to see what `logread -f` says as this happens:

```
Sat Mar 11 10:24:10 2023 auth.info sshd[3511]: Accepted publickey for root from 192.168.148.205 port 47982 ssh2: RSA SHA256:xxxxx
Sat Mar 11 10:24:10 2023 kern.info kernel: [ 2187.842219] do_page_fault(): sending SIGSEGV to ash for invalid read access from 00000010
Sat Mar 11 10:24:10 2023 kern.info kernel: [ 2187.850942] epc = 0043a76b in busybox[400000+52000]
Sat Mar 11 10:24:10 2023 kern.info kernel: [ 2187.856171] ra  = 0043a5cf in busybox[400000+52000]
Sat Mar 11 10:24:10 2023 auth.info sshd[3511]: Received disconnect from 192.168.148.205 port 47982:11: disconnected by user
Sat Mar 11 10:24:10 2023 auth.info sshd[3511]: Disconnected from user root 192.168.148.205 port 47982
```
As you can see, I'm using an openssh server on the router.  

### Expected behaviour

Here's what happens if I log in from an xterm (it works well):
```
230311 10:40 /persist/home/srn root@basil# ssh rpc148
BusyBox v1.36.0 (2023-02-27 10:11:01 UTC) built-in shell (ash)

r22170-5e48c534f7
root@rpc148:~# 
```
As this happens, logread -f says only something like:

```
Sat Mar 11 10:24:10 2023 auth.info sshd[3511]: Accepted publickey for root from 192.168.148.205 port 47982 ssh2: RSA SHA256:xxxxx
```

### Diffconfig

```text
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_tplink_archer-a7-v5=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_ATH10K_LEDS=y
CONFIG_BATMAN_ADV_BATMAN_V=y
CONFIG_BATMAN_ADV_BLA=y
CONFIG_BATMAN_ADV_DAT=y
CONFIG_BATMAN_ADV_MCAST=y
CONFIG_BATMAN_ADV_NC=y
CONFIG_BIND_ENABLE_DOH=y
CONFIG_BUSYBOX_CONFIG_ADDGROUP=y
CONFIG_BUSYBOX_CONFIG_ADDUSER=y
CONFIG_BUSYBOX_CONFIG_DIFF=y
CONFIG_BUSYBOX_CONFIG_FEATURE_ADDUSER_TO_GROUP=y
CONFIG_BUSYBOX_CONFIG_FEATURE_CHECK_NAMES=y
CONFIG_BUSYBOX_CONFIG_FEATURE_DIFF_DIR=y
CONFIG_BUSYBOX_CONFIG_FEATURE_DIFF_LONG_OPTIONS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_EDITING_SAVEHISTORY=y
CONFIG_BUSYBOX_CONFIG_FEATURE_FIND_DELETE=y
CONFIG_BUSYBOX_CONFIG_FEATURE_FIND_EMPTY=y
CONFIG_BUSYBOX_CONFIG_FEATURE_FIND_EXECUTABLE=y
CONFIG_BUSYBOX_CONFIG_FEATURE_FIND_EXEC_PLUS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_FIND_QUIT=y
CONFIG_BUSYBOX_CONFIG_FIRST_SYSTEM_ID=0
CONFIG_BUSYBOX_CONFIG_HOSTNAME=y
CONFIG_BUSYBOX_CONFIG_IONICE=y
CONFIG_BUSYBOX_CONFIG_LAST_ID=0
CONFIG_BUSYBOX_CONFIG_LAST_SYSTEM_ID=0
CONFIG_KERNEL_BUILD_DOMAIN="coolheads.com"
CONFIG_KERNEL_BUILD_USER="srn"
CONFIG_KERNEL_DETECT_HUNG_TASK=y
CONFIG_KERNEL_SOFTLOCKUP_DETECTOR=y
CONFIG_KERNEL_WQ_WATCHDOG=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LUA_ECO_DEFAULT_OPENSSL=y
# CONFIG_LUA_ECO_MBEDTLS is not set
CONFIG_LUA_ECO_OPENSSL=y
CONFIG_OPENSSH_LIBFIDO2=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_PACKAGE_ath10k-firmware-qca988x=y
# CONFIG_PACKAGE_ath10k-firmware-qca988x-ct is not set
CONFIG_PACKAGE_batctl-full=y
CONFIG_PACKAGE_bind-dig=y
CONFIG_PACKAGE_bind-libs=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_curl=y
# CONFIG_PACKAGE_iw is not set
CONFIG_PACKAGE_iw-full=y
CONFIG_PACKAGE_kmod-ath10k=y
# CONFIG_PACKAGE_kmod-ath10k-ct is not set
CONFIG_PACKAGE_kmod-batman-adv=y
# CONFIG_PACKAGE_kmod-hwmon-core is not set
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libbz2=y
CONFIG_PACKAGE_libcbor=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libevdev=y
CONFIG_PACKAGE_libffi=y
CONFIG_PACKAGE_libfido2=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libpcre=y
CONFIG_PACKAGE_libpopt=y
CONFIG_PACKAGE_libpython3=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libudev-zero=y
CONFIG_PACKAGE_libuv=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_mailsend=y
CONFIG_PACKAGE_nano-full=y
CONFIG_PACKAGE_nstat=y
CONFIG_PACKAGE_openssh-client=y
CONFIG_PACKAGE_openssh-client-utils=y
CONFIG_PACKAGE_openssh-keygen=y
CONFIG_PACKAGE_openssh-server=y
CONFIG_PACKAGE_python3-asyncio=y
CONFIG_PACKAGE_python3-base=y
CONFIG_PACKAGE_python3-light=y
CONFIG_PACKAGE_python3-multiprocessing=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_rsync=y
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_shadow=y
CONFIG_PACKAGE_shadow-chage=y
CONFIG_PACKAGE_shadow-chfn=y
CONFIG_PACKAGE_shadow-chgpasswd=y
CONFIG_PACKAGE_shadow-chpasswd=y
CONFIG_PACKAGE_shadow-chsh=y
CONFIG_PACKAGE_shadow-common=y
CONFIG_PACKAGE_shadow-expiry=y
CONFIG_PACKAGE_shadow-faillog=y
CONFIG_PACKAGE_shadow-gpasswd=y
CONFIG_PACKAGE_shadow-groupadd=y
CONFIG_PACKAGE_shadow-groupdel=y
CONFIG_PACKAGE_shadow-groupmems=y
CONFIG_PACKAGE_shadow-groupmod=y
CONFIG_PACKAGE_shadow-groups=y
CONFIG_PACKAGE_shadow-grpck=y
CONFIG_PACKAGE_shadow-grpconv=y
CONFIG_PACKAGE_shadow-grpunconv=y
CONFIG_PACKAGE_shadow-lastlog=y
CONFIG_PACKAGE_shadow-login=y
CONFIG_PACKAGE_shadow-logoutd=y
CONFIG_PACKAGE_shadow-newgidmap=y
CONFIG_PACKAGE_shadow-newgrp=y
CONFIG_PACKAGE_shadow-newuidmap=y
CONFIG_PACKAGE_shadow-newusers=y
CONFIG_PACKAGE_shadow-nologin=y
CONFIG_PACKAGE_shadow-passwd=y
CONFIG_PACKAGE_shadow-pwck=y
CONFIG_PACKAGE_shadow-pwconv=y
CONFIG_PACKAGE_shadow-pwunconv=y
CONFIG_PACKAGE_shadow-su=y
CONFIG_PACKAGE_shadow-useradd=y
CONFIG_PACKAGE_shadow-userdel=y
CONFIG_PACKAGE_shadow-usermod=y
CONFIG_PACKAGE_shadow-utils=y
CONFIG_PACKAGE_shadow-vipw=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wget-ssl=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mesh-mbedtls=y
CONFIG_PACKAGE_zlib=y
CONFIG_shadow-all=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Steve-Newcomb, 2023-03-20:
Same problem continues in r22307.

akossomfai, 2023-04-12:
Hi, fixed in busybox upstream via https://git.busybox.net/busybox/commit/?id=853cfe927fd656a2688ac2bfc81c69e1004c44df

Steve-Newcomb, 2023-04-14:
I confirm that the fix works for me.  I can now log in via ssh normally from within an Emacs shell buffer.

closed 2023-04-14 by Steve-Newcomb