openwrt/openwrt #12166: R6260 MT7621 slow WiFi speed/coverage

netstickbat, 2023-03-12:
### Describe the bug

I think last release with decent speed/coverage was probably 21.x. Been running snapshots ever since but the issue still persists. As soon as I move further away (~3m) from the router coverage and thus speed drops significantly whereas this used to be just fine before. Quick search shows [this ](https://forum.openwrt.org/t/r6260-low-tx-power/94366) post on the forum related with low TX power.

### OpenWrt version

r22257-166ab6f90e

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear R6260

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-13:
What does `iwinfo` say for programmed tx power? (you can post it maksing/editing away MAC and SSID)

netstickbat, 2023-03-13:
Here it is. Let me know if you need more details.

```
phy0-ap0  ESSID: "x"
          Access Point: AB:CD:EF:01:23:45
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 20 dBm  Link Quality: 22/70
          Signal: -88 dBm  Noise: unknown
          Bit Rate: 13.7 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7603 14C3:7603 [MediaTek MT7603E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy1-ap0  ESSID: "y"
          Access Point: AB:CD:EF:01:23:45
          Mode: Master  Channel: 100 (5.500 GHz)  HT Mode: VHT160
          Center Channel 1: 114 2: unknown
          Tx-Power: 6 dBm  Link Quality: 35/70
          Signal: -75 dBm  Noise: -90 dBm
          Bit Rate: 30.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

phy1-ap1  ESSID: "z"
          Access Point: AB:CD:EF:01:23:45
          Mode: Master  Channel: 100 (5.500 GHz)  HT Mode: VHT160
          Center Channel 1: 114 2: unknown
          Tx-Power: 6 dBm  Link Quality: 34/70
          Signal: -76 dBm  Noise: -90 dBm
          Bit Rate: 216.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

tapper82, 2023-03-13:
This is what I get from my r7800.

 wlan0     ESSID: "tapper's net"                                                 
           Signal: -73 dBm  Noise: -108 dBm                                      
           Bit Rate: 490.0 MBit/s                                                
           Encryption: WPA2 PSK (CCMP)                                           
           Type: nl80211  HW Mode(s): 802.11ac/n                                 
           Hardware: 168C:0046 168C:CAFE [Qualcomm Atheros QCA9984]              
           TX power offset: none                                                 
           Frequency offset: none                                                
           Supports VAPs: yes  PHY name: phy0                                    
 wlan1     ESSID: "tapper's net"                                                 
           Access Point: 78:D2:94:A2:14:9B                                       
           Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT20                   
           Center Channel 1: 1 2: unknown                                        
           Tx-Power: 20 dBm  Link Quality: 36/70                                 
           Signal: -74 dBm  Noise: -102 dBm                                      
           Bit Rate: 72.9 MBit/s                                                 
           Encryption: WPA2 PSK (CCMP)                                           
           Type: nl80211  HW Mode(s): 802.11b/g/n                                
           Hardware: 168C:0046 168C:CAFE [Qualcomm Atheros QCA9984]              
           TX power offset: none                                                 
           Frequency offset: none                                                
           Supports VAPs: yes  PHY name: phy1                                    
 root@TappersAP:~#                                                               

This is from my r6260

 phy0-ap0  ESSID: "tapper's net"                                                 
           Access Point: 38:aa:bb:cc:EA:D3                                       
           Mode: Master  Channel: 11 (2.462 GHz)  HT Mode: HT40                  
           Center Channel 1: 9 2: unknown                                        
           Tx-Power: 20 dBm  Link Quality: 40/70                                 
           Signal: -70 dBm  Noise: unknown                                       
           Bit Rate: 86.7 MBit/s                                                 
           Encryption: WPA2 PSK (CCMP)                                           
           Type: nl80211  HW Mode(s): 802.11b/g/n                                
           Hardware: 14C3:7603 14C3:7603 [MediaTek MT7603E]                      
           TX power offset: none                                                 
           Frequency offset: none                                                
           Supports VAPs: yes  PHY name: phy0                                    
 phy1-ap0  ESSID: "tapper's net"                                                 
           Access Point: 38:aa:bb:cc:dd:ee                                       
           Mode: Master  Channel: 100 (5.500 GHz)  HT Mode: VHT80                
           Center Channel 1: 106 2: unknown                                      
           Tx-Power: 20 dBm  Link Quality: 51/70                                 
           Signal: -59 dBm  Noise: -89 dBm                                       
           Bit Rate: 530.8 MBit/s                                                
           Encryption: WPA2 PSK (CCMP)                                           
           Type: nl80211  HW Mode(s): 802.11ac/n                                 
           Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]                      
           TX power offset: none                                                 
           Frequency offset: none                                                
           Supports VAPs: yes  PHY name: phy1                                    
 root@TappersAP:~#                                                               
R7800 is on stable and r6260 is on master.

tapper82, 2023-03-13:
@netstickbat  Are you talking about the 2.4 or the 5 gig radio?

brada4, 2023-03-14:
```
       Access Point: AB:CD:EF:01:23:45
          Mode: Master  Channel: 100 (5.500 GHz)  HT Mode: VHT160
          Tx-Power: 6 dBm  Link Quality: 35/70
```
Please set region in web UI, the setting is separate per radio, confirm with `iw reg get`

You can adjust channel auto-selection to prefer more powerful channels by setting minimal power in 5GHz band in `/wtc/config/wireless` like
```
config wifi-device 'radio0'
...
        option band '5g'
...
        # real country code
        option country 'XX'
        # automatic selection
        option channel 'auto'
        # prefer realistically strong channels, selects strongest if too high
        option min_tx_power '19'
```

netstickbat, 2023-03-14:
> @netstickbat Are you talking about the 2.4 or the 5 gig radio?

@tapper82 Both. Although 2.4 is definitely slower than before.

> ```
>        Access Point: AB:CD:EF:01:23:45
>           Mode: Master  Channel: 100 (5.500 GHz)  HT Mode: VHT160
>           Tx-Power: 6 dBm  Link Quality: 35/70
> ```
> 
> Please set region in web UI, the setting is separate per radio, confirm with `iw reg get`

@brada4 This was already set. Here's the output of that command:

```
# iw reg get
global
country GB: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
        (5925 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 71000 @ 2160), (N/A, 40), (N/A)
```
 
> You can adjust channel auto-selection to prefer more powerful channels by setting minimal power in 5GHz band in `/wtc/config/wireless` like
> 
> ```
> config wifi-device 'radio0'
> ...
>         option band '5g'
> ...
>         # real country code
>         option country 'XX'
>         # automatic selection
>         option channel 'auto'
>         # prefer realistically strong channels, selects strongest if too high
>         option min_tx_power '19'
> ```
> 
> followed by `service wpad restart`

Done. Although things are the same coverage/speed wise after a `service wpad restart`. Here's the output:

```
# iwinfo
phy0-ap0  ESSID: "x"
          Access Point: AB:CD:EF:01:23:45
          Mode: Master  Channel: 3 (2.422 GHz)  HT Mode: HT20
          Center Channel 1: 3 2: unknown
          Tx-Power: 20 dBm  Link Quality: 23/70
          Signal: -87 dBm  Noise: unknown
          Bit Rate: 13.8 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7603 14C3:7603 [MediaTek MT7603E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy1-ap0  ESSID: "y"
          Access Point: AB:CD:EF:01:23:45
          Mode: Master  Channel: 100 (5.500 GHz)  HT Mode: VHT160
          Center Channel 1: 114 2: unknown
          Tx-Power: 6 dBm  Link Quality: 36/70
          Signal: -74 dBm  Noise: -90 dBm
          Bit Rate: 30.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

phy1-ap1  ESSID: "z"
          Access Point: AB:CD:EF:01:23:45
          Mode: Master  Channel: 100 (5.500 GHz)  HT Mode: VHT160
          Center Channel 1: 114 2: unknown
          Tx-Power: 6 dBm  Link Quality: 36/70
          Signal: -74 dBm  Noise: -90 dBm
          Bit Rate: 58.5 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

# grep -Ev "key|ssid|network" /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option band '2g'
        option htmode 'HT40'
        option channel 'auto'
        option cell_density '0'
        option country 'GB'
        option min_tx_power '19'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option mode 'ap'
        option encryption 'psk2'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option htmode 'VHT160'
        option channel 'auto'
        option country 'GB'
        option cell_density '0'
        option min_tx_power '19'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option mode 'ap'
        option isolate '1'
        option encryption 'psk2'

config wifi-iface 'wifinet2'
        option device 'radio1'
        option mode 'ap'
        option encryption 'psk2'
        option isolate '1'
```

brada4, 2023-03-14:
Seems 5ghz calibration data is wrong indeed. Ch 100 is DFS as per reg domain, so it should be 26dBm or Off . Maybe try VHT80, it has some non-DFS options to choose from?

Regarding 20-40MHz in 2.4GHz band - there is advanced option to force 40MHz in busy area, but normally that does not add transfer speeds, just interferes more with your neighbours.

netstickbat, 2023-03-20:
> Seems 5ghz calibration data is wrong indeed. Ch 100 is DFS as per reg domain, so it should be 26dBm or Off . Maybe try VHT80, it has some non-DFS options to choose from?

Looks the same with VHT80. Would that be the calibration as mentioned on the forum post that I have linked on my original post above? How do I go about finding the right address for the .dtsi files? Would this apply to the 2.4GHz radio as well since I am having issues with that one too?

As far as I understood by the forum post this should be dynamically picked up but a patch had to be reverted due to some issues. Do you happen to know if there are any plans on getting that sorted on OpenWRT?

> Regarding 20-40MHz in 2.4GHz band - there is advanced option to force 40MHz in busy area, but normally that does not add transfer speeds, just interferes more with your neighbours.

I'll try that.

> Also highly recommended to set 2.4 and 5 GHz SSID names equal so that exiting coverage of one other is automatically picked up (no roaming support needed in common case)

Will do. Thanks for your help!

DragonBluep, 2023-05-19:
R6260 does not support 160 MHz BW.