openwrt/openwrt #12173: Cant get 22.03 intalled on TP-LINK  Archer C7 V5

real-t0mg, 2023-03-12:
### Describe the bug

Image verification fails for official image when attempting an install of 22.03 from 19.07. Forcing install results in bricked device. Debricking as per the wiki for the device works with TFTP. 

I've been trying to get 22.03 to work on my tp-link archer C7 v5 for months now, without avail. Compiling from scratch with defaults after selecting correct device from menuconfig renders an ununsuable image larger than the mtd partitions and does not include the Candela Tech firmware for ath10k.

[image]

### OpenWrt version

22.03

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v5

### Image kind

Official downloaded image

### Steps to reproduce

log in as root to device, execute: 

# cd /tmp 
# wget https://downloads.openwrt.org/releases/22.03.3/targets/ath79/generic/openwrt-22.03.3-ath79-generic-tp
link_archer-c7-v5-squashfs-factory.bin
# sysupgrade -T openwrt-22.03.3-ath79-generic-tplink_archer-c7-v5-squashfs-factory.bin

### Actual behaviour

Image metadata not found
Use sysupgrade -F to override this check when downgrading or flashing to vendor firmware
Image check failed.

### Expected behaviour

sysupgrade from factory.bin to work flawless with official provided image. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-12:
You need sysupgrade image for sysupgrade. Factory image is to install via vendor web ui first time.

brada4, 2023-03-13:
Direct links:
https://firmware-selector.openwrt.org/?version=22.03.3&target=ath79%2Fgeneric&id=tplink_archer-c7-v5

closed 2023-03-13 by real-t0mg