openwrt/openwrt #12177: Luci and other components use and display wrong uptime when running inside an lxd container

MaxRower, 2023-03-13:
### Describe the bug

Instead of using /proc/uptime, which shows the correct uptime of the container itself, it seems to use another source not corrected by lxcfs. 
busybox uptime displays the wrong value as well, but that can be corrected by installing procps-ng-uptime. But that doesn't help in other places. 
So the Luci status overview shows the hosts uptime instead of the containers, and the services - dynamic dns status page always shows a wrong last update time and "verify" for the next update time, which normally indicates an error.
collectd is affected as well. collectd uptime within a container running Ubuntu 22.04 seems to show correct values.

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

x86/64

### Device

NUC6CAYB running Ubuntu 22.04

### Image kind

Official downloaded image

### Steps to reproduce

root@container:~# uptime           (installed from procps-ng-uptime)
 00:31:57 up  1:23,  0 users,  load average: 0.14, 0.42, 0.49
root@container:~# busybox uptime
 00:31:59 up 17 days,  6:30,  load average: 0.14, 0.41, 0.48

### Additional info

The container was created by "lxc launch images:openwrt/22.03"

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

MaxRower, 2023-03-13:
All of this because of the problems of the current 22.03 release and master with my WRT32X, it's unusable until hopefully the next major release. So i had to virtualize the router, and my trusty spare Archer C7 v2 serves as an AP (it can't handle wireguard and python like the WRT32X does)

MaxRower, 2023-03-13:
Sorry, didn't want to close the issue.

brada4, 2023-03-14:
It is fixed in upstream of Ubuntu:
https://github.com/lxc/lxd/issues/10545
No problem with busybox or OpenWRT.

MaxRower, 2023-03-14:
> It is fixed in upstream of Ubuntu: [lxc/lxd#10545](https://github.com/lxc/lxd/issues/10545) No problem with busybox or OpenWRT.
> 
> And you can use `coreutils-uptime` to have version visualising /proc/ file.

That fix should be included in lxd 5.11? But on another NUC with ubuntu 22.04 und the latest lxd 5.11 snap, it still does occur:
```
BusyBox v1.35.0 (2022-10-14 22:44:41 UTC) built-in shell (ash)

~ # uptime
 20:40:08 up 6 days,  7:55,  load average: 0.13, 0.10, 0.18
~ # cat /proc/uptime
39253.07 39218.53
```

MaxRower, 2023-03-14:
> And you can use `coreutils-uptime` to have version visualising /proc/ file.

When I install that, uptime does not work anymore.

```
~ # opkg install coreutils-uptime
Installing coreutils-uptime (9.0-2) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/x86_64/packages/coreutils-uptime_9.0-2_x86_64.ipk
Installing coreutils (9.0-2) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/packages/x86_64/packages/coreutils_9.0-2_x86_64.ipk
Configuring coreutils.
Configuring coreutils-uptime.
~ # uptime
uptime: couldn't get boot time: Not supported
```

brada4, 2023-03-14:
On a clean KVM+libvirt vm all variants of uptime work - busybox, coreutils-uptime, procps-ng-uptime.

MaxRower, 2023-03-14:
> On a clean KVM+libvirt vm all variants of uptime work - busybox, coreutils-uptime, procps-ng-uptime.

But you mean a KVM virtual machine, with it's own kernel, do you?

pkgadd, 2023-03-15:
You do realize that running an OpenWrt guest inside a lxc/ lxd container isn't really a supported runtime environment to begin with?

OpenWrt has rather peculiar dependencies on the host kernel and its (sysctl-, netfilter-, network drivers- and more-) preconfiguration, which often differs from that of common desktop linux, and relies on the behaviour of the OpenWrt kernel. This isn't a problem for full system virtualization like kvm/ virtualbox/ Hyper-V/ vmware (where OpenWrt's own kernel is run), but simply cannot work inside a container. Issues -including serious security ones- are to be expected there, just don't do it.

brada4, 2023-03-15:
busybox uses sysinfo(2) to get uptime, and that does not seem to be provided by lxc. Add ton of problems because of missing lxc host kmods, while ones doing NAT are preloaded by virtue of installing lxc, others may not even exist.