openwrt/openwrt #12183: luci-app-attendedsysupgrade: x64 image size adjustment (previously installed packages do not fit default image)

beatstick, 2023-03-15:
I am trying to update an x64 box via attendedsysupgrade and getting the error: out of space, while the app tries to integrate the previously installed packages. I've installed around 700 MB worth of packages, so this does not come as a surprise, as the default x64-ext4-image comes at a very small size (I think 128 MB). 

Many x64 users will have expended their root partitions to accommodate more packages, as the default is rather small and not up to par with recent ssd-drive sizes. 

Solutions: 
1. Maybe there could be an option to proceed with the installation, even if not all packages where installed? The user could manually reinstall everything else that could not fit later with opkg. 

2. A great solution would be for the sysupgrade app to resize the image to the size currently on the users disk, so everything will work as before after the attended sysupgrade.

cooper2k4, 2023-04-15:
Is it possible to add an input field to specify the individual image/rootfs size in the LuCI attendedsysupgrade expert mode (where you specify the additional packages)? This would be a great improvement based on the recent imagebuilder patch (7b7edd25a571568438c886529d3443054e02f55f).

aparcar, 2023-04-15:
Sure do you mind adding it? Should be complicated

aparcar, 2023-04-15:
The parameter is called rootfs_size, it's not yet active on the main server but I'll add it in a bit

cooper2k4, 2023-04-15:
I would add the following line [above the list of packages](https://github.com/openwrt/luci/blob/5ca5692e449d22e5c5480c93b4919f7c09f800ec/applications/luci-app-attendedsysupgrade/htdocs/luci-static/resources/view/attendedsysupgrade/overview.js#L375)

`o = s.option(form.Value, 'rootfs_size', _('Custom rootfs size'));`

* Is it possible to get the default rootfs size for the selected profile? Then the input field could be initialized with that in [line 356](https://github.com/openwrt/luci/blob/5ca5692e449d22e5c5480c93b4919f7c09f800ec/applications/luci-app-attendedsysupgrade/htdocs/luci-static/resources/view/attendedsysupgrade/overview.js#L356).
* Additionally it would be nice to use the [`input type="number"`](https://developer.mozilla.org/en-US/docs/Web/HTML/Element/input/number) for the input field.

beatstick, 2023-10-19:
Sorry, is this implemented, yet? 

I cannot see a gui option, could you please give an example for the auc command?

beatstick, 2024-09-10:
This seems not very complicated on first look. Could I just create the one line above

> o = s.option(form.Value, 'rootfs_size', _('Custom rootfs size'));

on my installation in overview.js or could I end up with a bricked firewall?