openwrt/openwrt #12190: Fortinet FortiGate 50E: no serial output during boot / dmesg / no serial terminal

johnhaywood, 2023-03-16:
### Describe the bug

no openwrt serial output of any kind.
no dmesg messages output to serial console
no shell login output to serial

### OpenWrt version

commit	102dc5a6250608f8324ce20174e8a07523f98bcc

### OpenWrt target/subtarget

DISTRIB_RELEASE='SNAPSHOT' DISTRIB_REVISION='r22278-7dceef5ee5' DISTRIB_TARGET='mvebu/cortexa9' DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16' DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r22278-7dceef5ee5'

### Device

Fortinet FortiGate 50E

### Image kind

Official downloaded image

### Steps to reproduce

bootup device with serial cable connected.
verify serial connection is correct by seeing uboot menu/messages.
ie uboot messages are shown
no other output is shown (ie all missing).

### Actual behaviour

no serial output is shown after uboot menu/messages.

### Expected behaviour

expect to see dmesg output to serial console.
expect to see shell login prompt on boot completion.

### Additional info

new target.
works in other devices of same target
FIX (guess)
no uboot env defined for serial
ie issue shell command;
fw_setenv console ttyS0,115200

note currently no way to test this as all ubootenv commands are also broken (logs as separate bug)
no access to uboot command prompt for this device so can not set uboot envs from their either...

or

is it an issue with the serial console not defined in the dts file?
target/linux/mvebu/files/arch/arm/boot/dts/armada-385-fortinet-fg-50e.dts

works on other mvebu/cortexa9/ targets.
needs serial port definition correctly adding/adapting for this new target?
and/or serial port correctly specifying as output for stout / shell?

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

musashino205, 2023-03-17:
Hi, could you provide the output of ```[I]: System information.``` ---> ```[S]:  Set serial port baudrate.``` on bootmenu?

The default baudrate of 50E is 9600 and OpenWrt expects baudrate to be set to that value.

johnhaywood, 2023-03-17:
FIXED; was 115200 (sorry) now 9600
Thank you.
I stupidly changed it to 115200 to match openwrt default thinking that would allow me to see both uboot and openwrt messages.
would it not be better for openwrt to output at its default speed of 115200, and just advise users to change uboot baudrate to 115200 in [I] menu.
this is how it works for 99% of other devices with uboot menus?

musashino205, 2023-03-17:
IMO, baudrate of device support in OpenWrt should match the default on stock firmware of the device.
In this case, I believe the problem was caused by the user (maybe not you) intentionally changing baudrate on stock firmware from the default value of 50E.

BTW, the bootmenu of 50E is the original of Fortinet, not that of U-Boot.