openwrt/openwrt #12192: mpc85xx: HiveAP 330: Ethernet PHYs no longer detect link-up as of 5.10 -> 5.15 snapshot

Hurricos, 2023-03-16:
### Describe the bug

Hi,

```console
root@aerohive:~# ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN qlen 1000
    link/ether 08:ea:44:15:67:40 brd ff:ff:ff:ff:ff:ff
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN qlen 1000
    link/ether 08:ea:44:15:67:41 brd ff:ff:ff:ff:ff:ff
17: labb2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 00:02:6f:d5:fe:dd brd ff:ff:ff:ff:ff:ff
```

... despite that the link is detected as up on the other end:

```console
[948854.229587] br-lan: port 4(lan4) entered blocking state
[948854.234917] br-lan: port 4(lan4) entered forwarding state
[948944.354285] mv88e6085 f1072004.mdio-mii:00 lan4: Link is Down
[948944.360221] br-lan: port 4(lan4) entered disabled state
[949419.146554] mv88e6085 f1072004.mdio-mii:00 lan4: Link is Up - 1Gbps/Full - flow control off
[949419.155073] br-lan: port 4(lan4) entered blocking state
[949419.160405] br-lan: port 4(lan4) entered forwarding state
```

I'm almost certain this is a regression due to 5.15.

`ethtool` agrees that the links are down.

Working on pulling a 5.10 snapshot down to show the exact dmesg regression, but there doesn't appear to be one.

### OpenWrt version

r22278-7dceef5ee5

### OpenWrt target/subtarget

mpc85xx/p1020

### Device

Aerohive HiveAP-330

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Hurricos, 2023-03-19:
Each link comes up when `ethtool -s eth0 autoneg on` is run, but autonegotiation is supposedly already configured.

Before this `ethtool -s` command is run, `ethtool eth0` reports:

```
Settings for eth0:
        Supported ports: [ MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: Unknown!
        Duplex: Unknown! (255)
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Current message level: 0x0000003f (63)
                               drv probe link timer ifdown ifup
        Link detected: no
```

After `ethtool` is run:

```
Settings for eth0:
        Supported ports: [ MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: Symmetric
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: on
        MDI-X: off (auto)
        Current message level: 0x0000003f (63)
                               drv probe link timer ifdown ifup
        Link detected: yes
```

This might be the same issue as was fixed in 8b3c313515c1ae37be9d61a4e92c4bf4b43f605f ("mpc85xx: define reset-delay for WS-AP3825i eth PHY"); the two boards use the same PHYs. 

There are differences (for example, `/etc/init.d/network restart` fixes the issue on the AP3825i but not here). Perhaps the reason for the difference is that ifup/ifdown makes use of the reset pins; if I add the GPIO reset pins to the DTS (as well as the `reset-assert-us` properties) we can see whether the problem goes away.

Hurricos, 2023-03-19:
I've mapped the GPIOs which reset the PHYs; they're identical - pins 6 and 7 - as on the WS-AP3825i (thankfully).

Hurricos, 2023-03-20:
Interestingly enough, mapping the PHY reset GPIOs and asserting a reset doesn't solve the issue.

The WS-AP3825i snapshot currently doesn't boot with CONFIG_SMP enabled, so it's tough to say whether other nearly-identical platforms have this issue.

I doubt he'll see it, but @blocktrron was recently working on a sister platform, the WS-AP3715i -- perhaps he'll have some insight as to what's going on here.

Hurricos, 2023-03-20:
> Have you checked if the interrupt is lost within the stack?

I am not sure what this means. I expect the PHY issues an interrupt when configuration changes, and?

> I assume polling the link-state fixes the issue here.

Maybe - I was only able to get the link up with `ethtool --change` / `-s`. I would think that `ethtool eth0` alone would poll the PHY, but maybe there's some caching going on.

The PHY has autonegotiated, but but that the OS is wrongly tracking link state. So, if I'm interpreting "lost in the stack" correctly, then you're saying the PHY's interrupt never makes it to the CPU.

I think PHY interrupts happen on MDIO. On 5.15, I see no interrupts on `mdio` before or after `ethtool -s` is issued:

```txt
root@aerohive:~# cat /proc/interrupts 
           CPU0       CPU1       
:
 24:          0          0   OpenPIC     3 Level     mdio@ffe24000:01
 25:          0          0   OpenPIC     2 Level     mdio@ffe24000:02
 :
```

I seem to recall that 5.10 shows no interrupts at all here. Let me sysupgrade and confirm.

Hurricos, 2023-03-20:
> I'm fine with disabling interrupt handling altogether and either leave it as-is or re-enable it as soon as the bug is fixed.

Do you have a nicer solution than remove the interrupt line from the device trees on each board -- is there a kernel configuration option for this?

blocktrron, 2023-03-20:
This is the nicest solution as it only disables it for the specific board.

Hurricos, 2023-03-20:
OK. I've been assuming that if it's a platform issue, it's impacting other boards. Granted, I have access to almost all of the potentially-impacted boards, so I can go test.

Hurricos, 2023-03-21:
@BKPepe: I certainly can try. Let me find the patches Turris uses to build.

Hurricos, 2023-03-22:
@blocktrron: AP-330 works as of most recent snapshot (tested https://downloads.openwrt.org/snapshots/targets/mpc85xx/p1020/openwrt-mpc85xx-p1020-aerohive_hiveap-330-squashfs-sysupgrade.bin, from r22362-68883278ab).

closed 2023-03-22 by Hurricos

andiohn, 2023-05-10:
> 

Has this been pushed to the most recent snapshot? Sorry that my Github-fu isn't the best :/