openwrt/openwrt #12197: can't mount ntfs3 partition on wndr3800

darwin787, 2023-03-18:
### Describe the bug

snapshot build, can't mount  any ntfs3 partition , invalid argument error.

### OpenWrt version

snapshot

### OpenWrt target/subtarget

ath79

### Device

netgear wndr3800

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Spudz76, 2023-03-18:
does dmesg or logread have anything interesting?  my reflex is that there are missing modules such as NLS or otherwise

Spudz76, 2023-03-18:
also could just be the disk has the dirty bit set, and that will refuse to mount without `ntfsfix -d` or a force option in the mount command

darwin787, 2023-03-19:
not missing modules, not dirty bit

pkgadd, 2023-03-19:
ntfs3 has been new in v5.15 and has probably seen very little testing on big endian (ath79) before, ideally it would have be interesting to test a more current kernel (v6.1) and engaging upstream, but that's going to be difficult with OpenWrt on ath79.

alpha7504, 2024-03-08:
It seems, there's a patch available for this issue. https://lore.kernel.org/lkml/20211207102454.576906-4-thomas.kuehnel@avm.de/T/