openwrt/openwrt #12204: Support Tenda AC11 V2

alenpdev, 2023-03-18:
### Describe the bug

Please can you add model tenda ac11 v2  and xiaomi ax5400. Thank you

### OpenWrt version

1

### OpenWrt target/subtarget

1

### Device

Tenda Ac11 v2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

alenpdev, 2023-03-18:
The Tenda AC11 (AC11) router released in 2018. It is powered by Realtek RTL8197FH @ 1 GHz chipset, 128 MB RAM and 8 MB flash. The AC1200 router is compatible with IEEE 802.11a/b/g/n/ac standards, and the maximum speed that you can get with AC11 is 300 + 866 Mbps.

obsy, 2023-03-18:
RTL8197FH - not supported by OpenWrt.

brada4, 2023-03-19:
Xiaomi AX5400 SOC IPQ5XXX not supported either.

SirLouen, 2023-03-24:
May I ask. Why there are Support requests both in the forum and here in the Github at the same time?