openwrt/openwrt #12208: ramips/mt7621: OpenWrt 22.03.3 will not boot

BetTD, 2023-03-18:
### Describe the bug

My Afoundry EW1200 was running fine on an old stable release. Today, I went to update it with a sysupgrade image and now it's stuck in a boot loop. After hooking it up via a serial interface, I can see the following:

```
===================================================================
                MT7621   stage1 code 13:14:00 (ASIC)
                CPU=50000000 HZ BUS=16666666 HZ
==================================================================
Change MPLL source from XTAL to CR...
do MEMPLL setting..
MEMPLL Config : 0x11000000
3PLL mode + External loopback
=== XTAL-40Mhz === DDR-1200Mhz ===
PLL2 FB_DL: 0x12, 1/0 = 543/481 49000000
PLL3 FB_DL: 0x14, 1/0 = 623/401 51000000
MEMPLL 3PLL mode calibration fail
do DDR setting..[00320000]
Apply DDR3 Setting...(use default AC)
          0    8   16   24   32   40   48   56   64   72   80   88   96  104  112  120
      --------------------------------------------------------------------------------
0000:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0001:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0002:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0003:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0004:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0005:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0006:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0007:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0008:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0009:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
000A:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
000B:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
000C:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
000D:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
000E:|    0    0    0    0    0    0    0    0    0    0    1    1    1    1    1    1
000F:|    0    0    0    0    0    1    1    1    1    1    1    1    1    1    1    1
0010:|    1    1    1    1    1    1    1    1    1    1    0    0    0    0    0    0
0011:|    1    1    1    1    0    0    0    0    0    0    0    0    0    0    0    0
0012:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0013:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0014:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0015:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0016:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0017:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0018:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
0019:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
001A:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
001B:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
001C:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
001D:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
001E:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
001F:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0
rank 0 coarse = 15
rank 0 fine = 80
B:|    0    0    0    0    0    0    0    0    1    1    1    0    0    0    0    0
opt_dle value:9
DRAMC_R0DELDLY[018]=00001E1E
==================================================================
                RX      DQS perbit delay software calibration
==================================================================
1.0-15 bit dq delay value
==================================================================
bit|     0  1  2  3  4  5  6  7  8  9
--------------------------------------
0 |    13 13 12 15 12 12 13 9 11 11
10 |    14 15 12 13 12 13
--------------------------------------

==================================================================
[... 184 lines cut ...]

### Device

Afoundry EW1200

### Image kind

Official downloaded image

### Steps to reproduce

1. Flash OpenWrt 22.03.3 (or latest snapshot).
2. System won't boot

### Actual behaviour

The system never boots, it fails with ``LZMA ERROR 1`` when attempting to decompress the kernel image.

### Expected behaviour

The system should decompress the image fine and complete the boot process.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Spudz76, 2023-03-18:
I think this usually occurs when the kernel has gotten larger than the original kernel partition, thus trying to read 5+MB from a let's say 4MB partition will truncate the read and fail decompression.

However,
```
find flash: MX25L12805D
*** Warning - bad CRC, using default environment
```
This bothers me maybe you have actually bad flash, or the clock rate is too high, etc.  It's only 16MB also which alludes it won't have a kernel partition larger than 4MB.  Would bet the 21.02.5 kernel ends up below 4MB or whatever the kernel partition size is.  Even if it were 5MB it would still be too big.

I would hack around at the DTS for your device since that's where both of the above are defined.

Spudz76, 2023-03-18:
Then again the ram-load through TFTP should work regardless of flash situation.

BetTD, 2023-03-18:
> I think this usually occurs when the kernel has gotten larger than the original kernel partition, thus trying to read 5+MB from a let's say 4MB partition will truncate the read and fail decompression.

Wouldn't this result in an error **during** flashing and not after? You may be right in that the kernel is now larger than 4 MB. What options do I have now if I want to stay up to date?

> However,
>
> ```
> find flash: MX25L12805D
> *** Warning - bad CRC, using default environment
> ```
>
> This bothers me maybe you have actually bad flash, or the clock rate is too high, etc. It's only 16MB also [...]

This may be a dumb question, but what do you mean by DTS?

It may just have bad flash as well, I bought this second-hand for quite cheap, and the 2.4 GHz band doesn't even work, but as a cheap 5 GHz-only access point it gets the job done (after replacing the stock firmware with OpenWrt, that is). I'd much rather keep the system up to date though.

Spudz76, 2023-03-18:
DTS is the device tree source file, and each "board" (platform/model) has one

Yours is [target/linux/ramips/dts/mt7621_afoundry_ew1200.dts](https://github.com/openwrt/openwrt/blob/master/target/linux/ramips/dts/mt7621_afoundry_ew1200.dts)

Reading through there it seems the partition is unified (uimage) and has 16449536 bytes (uboot+env+factory take up 0x50000 ir 327680 bytes of the total 16MB flash) therefore kernel size should not be a factor.  Must be some LZMA format incompatibility, which matches the TFTP result.

Flash clock speed is the `spi-max-frequency` on line 54 of the DTS, although again since TFTP fails too and that happens all in RAM, it's probably also not the clock speed (although I'd prefer if the env-checksum message went away).

To some extent this is all a guessing game since nobody knows exactly what uboot+mods are on the device or what bugs/quirks it might support other than to try stuff like a black-box situation.  There have been several "flavors" of LZMA, maybe it wants the oldest version.  Lots of uboots didn't even support LZMA, on those platforms the lzma-loader is used and the uImage has the kernel marked as "uncompressed" and the lzma-loader does its own decompression (using current LZMA code) rather than hoping the builtin LZMA decompressor isn't broken, or using gzip.

Spudz76, 2023-03-18:
Confirmed your device is not using `uimage-lzma-loader` like most of the other MT7621's do.  Thus you are stuck at guessing what the builtin LZMA decompressor likes.  Some other models use special lzma compression flags, maybe copying some of those over would help.

Also possibly uboot is not allocating enough destination space in RAM.  Error "1" is generic failure and tells nobody anything really other than something went wrong.

Trying to find a reason the older version works, but maybe it uses older lzma compressor (different versions of host-tools? different tactics in the Makefiles?).

Also just noticed after re-reading your serial outputs that you didn't RAM-load the kernel from TFTP but rather flashed it, so flash *is* still involved (not just RAM-only).  So I'd next turn down the flash speed in the DTS and then see if it gets rid of corruption.

You could try option 1 (load to SDRAM) and see if that launches, if so then definitely flash speed or bad flash.

Mijzelf, 2023-03-19:
This looks like a bug I had on my ZyXEL P2812HNU-F1. Read [url=https://forum.openwrt.org/t/zyxel-p2812hnu-f1-u-boot/100281/8]here[/url]. It turned out u-boot only reserved 8MiB target space for decompressing the kernel.

xabolcs, 2023-03-19:
Hi @BetTD, 

did you try earlier 22.03 releases or even `-rc`s?
Please try to flash and boot 22.03.0 too! I think it will throw `LZMA ERROR 1` too.

May I ask for a serial log from the latest 21.02 release (e.g. 21.02.5)? Please paste only the relevant part, starting from the bootmenu, and please include a few lines about OpenWrt booting too!

I'm sure that the fix will be that oneliner what @Spudz76 wrote.
For examples see [commit history](https://github.com/openwrt/openwrt/commits/master/target/linux/ramips/image/mt7621.mk), e.g.:
 * 25ede5170d5fbfd233ba21f7a294ec3a2ddeb635
 * b63d6d4730fd0dc30ce6707338c398e8b9d61d86

As you can see it would be nice to check if the initramfs boot (boot menu 1) too!

Spudz76, 2023-03-19:
I guess if you go back to the last tested working 21.02 version, while you've got it up you could dump `mtd0` and attach it to a reply, so it can be disassembled and checked for misbehavior like only allocating 8MB of RAM for the expanded destination or such.

But also enabling the lzma-loader might work around both issues, since uboot will copy the ~5.3MB into the 8MB region and then launch the loader which should then reallocate proper amounts of RAM for the expansion (beyond whatever uboot did).  As long as the compressed image never gets above 8MB it should "just work" without replacing uboot with a patched version.

So an `mtd0` dump would mostly be for curiosity.  Working on a patch (based on snapshot/master) to enable lzma-loader.  If that is cool then it can possibly be backported to the releases that don't work in case anyone that also has one of these prefers releases over snapshots.  Or at least drop the non-working ones so they don't exist (users without serial capabilities would have sort of a hassle un-bricking, probably).

Spudz76, 2023-03-19:
Patch was actually a one-liner, now available in the above PR

@BetTD or anyone with one of these devices that could test it.

You could apply by checking out `master` and then from the base directory:
```
wget -qO- https://github.com/openwrt/openwrt/pull/12211.patch | patch -p1
```

And then configure/build/test, please.

BetTD, 2023-03-19:
Hey, apologies for the delay.

@Spudz76 - Good news! The patch you made seems to have worked, thanks a lot!

Checked out ``master`` and applied the patch, configured, built, and flashed.

```
Please choose the operation:
   1: Load system code to SDRAM via TFTP.
   2: Load system code then write to Flash via TFTP.
   3: Boot system code via Flash (default).
   4: Entr boot command line interface.
   7: Load Boot Loader code then write to Flash via Serial.
   9: Load Boot Loader code then write to Flash via TFTP.                                                                                                  4
You choosed 2
                                                                                                                                                           0
Upgrade mode read(1:2:0)

2: System Load Linux Kernel then write to Flash via TFTP.
 Warning!! Erase Linux in Flash then burn new one. Are you sure?(Y/N)
 Please Input new ones /or Ctrl-C to discard
        Input device IP (10.10.10.123) ==:192.168.0.204
        Input server IP (10.10.10.3) ==:192.168.0.10
        Input Linux Kernel filename () ==:initramfs3.bin

 netboot_common, argc= 3

 NetTxPacket = 0x87FE4E40

 KSEG1ADDR(NetTxPacket) = 0xA7FE4E40

 NetLoop,call eth_halt !

 NetLoop,call eth_init !
Trying Eth0 (10/100-M)

 Waitting for RX_DMA_BUSY status Start... done

 ETH_STATE_ACTIVE!!
TFTP from server 192.168.0.10; our IP address is 192.168.0.204
Filename 'initramfs3.bin'.

 TIMEOUT_COUNT=10,Load address: 0x80100000
Loading: checksum bad
checksum bad
checksum bad
checksum bad
checksum bad
checksum bad
checksum bad
checksum bad
checksum bad
checksum bad
checksum bad
T checksum bad
checksum bad
T Got ARP REPLY, set server/gtwy eth addr (2c:4d:54:e9:65:59)
Got it
#################################################################
         ####checksum bad
#checksum bad
############################################################
         #################################################################
         #################################################################
         ########checksum bad
################checksum bad
#########################################
         #################################################################
         #################################################################
         #################################################################
[... 47 lines cut ...]
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s15632 r8192 d21232 u45056
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 115628K/131072K available (7106K kernel code, 628K rwdata, 1444K rodata, 4504K init, 242K bss, 15444K reserved, 0K cma-reserved, 0K highmem)
[...]
```

Now my question is, am I good to continue running OpenWrt ``SNAPSHOT`` with this patch until a new release is out that includes it? Or should I stay in 21.02.5 until then?

Spudz76, 2023-03-20:
Yes, also snapshot has some non-backport changes to ramips/mt7621 that might help something, like the broken 2.4G wifi.

BetTD, 2023-03-20:
Unfortunately, it seems like 2.4 GHz wifi is still broken on my device, but personally, I don't really mind it. I can open a separate issue for it if you'd like, but it's a well-known problem with this router.

Spudz76, 2023-03-20:
There were some other models based on mt7621 which simply had some wrong things with PCIe detection, or only seem to work with delays inserted (which have not been accepted since static delays are kind of a bad workaround method).

Such as #9374

myapple, 2023-03-20:
mark

CastleCaptain, 2023-03-20:
I applied the latest RAMIPS MT7621 DIR3060 update release to a DIR3040 today. At boot, the 2.4 Ghz radio starts for a couple of minutes with the 5 Ghz remaining offline and then it switches with the 5Ghz radios running and the 2.4 radio offline. All else seems to work as expected.

Djfe, 2023-03-23:
The output of `logread` and `iw lift` in snapshot could already help in identifying some issues regarding wifi 2.4ghz

Spudz76, 2023-03-23:
The patch has been merged into `master`

Spudz76, 2023-03-23:
Made the backport patch for 22.03

BetTD, 2023-03-25:
> Made the backport patch for 22.03

I think this one requires your attention (#12248)

Spudz76, 2023-03-26:
Yeah, I'll get around to it.

BetTD, 2023-04-19:
Latest release 22.03.4 includes the patch and I can confirm the device upgraded fine. Thanks, guys!

Djfe, 2023-04-19:
are there still issues with wifi?

CastleCaptain, 2023-04-19:
At present i am unable to respond to your question regarding wifi issues. The unit has been running fine as a router for the past 2-3 weeks. However, I have not been at that residence since I installed the latest release at that time and cannot comment on its wifi performance. Immediately after the firmware install the wifi seemed fine. I hope to go there this weekend and will check and respond back here then.

BetTD, 2023-04-19:
> are there still issues with wifi?

I did test that out of curiosity, haven’t gotten that to work unfortunately. Any known debugging steps I could try?

Djfe, 2023-04-19:
> The output of `logread` and `iw list` in snapshot could already help in identifying some issues regarding wifi 2.4ghz

BetTD, 2023-04-21:
Output of ``logread`` and ``iw list``: https://gist.github.com/BetTD/15401ed9428704a427d00ddc24b39918

Spudz76, 2023-04-21:
Wifi issues may also be related to #12359 although that one generally fails to init the 5GHz (may be related causation, as various models wire their devices to different "slots")

Either way since the main topic here is lzma-loader (and could be closed since it's fixed), please move to one of the [other mt7621 PCI/wifi threads](https://github.com/openwrt/openwrt/issues?q=is%3Aissue+is%3Aopen+mt7621+wifi).

xabolcs, 2026-03-20:
> This issue was fixed by 889bbf89bb679f0c5b0fa432e27a3e0dd8940a4e, can be closed.

... and was backported to 22.03 by 0170bc1af1bc7efe4c26d8eeb11db6d3fe4cfcb7 which has been released as 22.03.4