openwrt/openwrt #12216: mtd write /tmp/factory.bin /dev/mtd2 Could not open mtd device: mtd2

shripadrayewar, 2023-03-20:
### Describe the bug

In the openwrt19 , mtd write /tmp/factory.bin /dev/mtd2 can be used, but in openwrt21.05 it will report an error

mtd write /tmp/factory.bin /dev/mtd2
Could not open mtd device: mtd2

as well as https://github.com/openwrt/openwrt/issues/9849 referenced from this issues,
i tried  mtd -r write /tmp/factory.bin factory
but it is also not working

### OpenWrt version

r0-f7c460685b

### OpenWrt target/subtarget

ramips/mt7621

### Device

PBR-M1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-20:
Once openwrt is installed you should use sysupgrade.

Factory image is with vendor headers to load via vendor upgrade tool like web or tftp.

shripadrayewar, 2023-03-20:
Hi Brada,
Im using sysupgrade for firmware upgrade, but need to write mac address on factory bock.
our vendor is providing 1 script to write mac address on factory block, but while writing its giving error.
same script to write mac on factory block is working perfectly on wrt 18 & 19. 
attaching script for reference
looking forward for your help on the issue.
[write_mac.txt](https://github.com/openwrt/openwrt/files/11021293/write_mac.txt)

brada4, 2023-03-20:
Look for`brick` [here](https://openwrt.org/toh/meraki/mr16)

shripadrayewar, 2023-03-20:
Thanks, 
Will test it out tomorrow

shripadrayewar, 2023-04-16:
Issue resolved
Issue was with DTS file, on DTS file, permission was read only for mtd block

closed 2023-04-16 by shripadrayewar