openwrt/openwrt #12217: RROR: toolchain/gcc/initial failed to build.

dd-bpir3, 2023-03-20:
### Describe the bug

 make[3] -C toolchain/gcc/initial compile
    ERROR: toolchain/gcc/initial failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/yph/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/home/yph/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/yph/openwrt/scripts/config'
 make[1] world
 make[2] toolchain/compile
 make[3] -C toolchain/gdb compile
 make[3] -C toolchain/binutils compile
 make[3] -C toolchain/gcc/initial compile
    ERROR: toolchain/gcc/initial failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/yph/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/home/yph/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/yph/openwrt/scripts/config'
make[1]: Entering directory '/home/yph/openwrt'
make[2]: Entering directory '/home/yph/openwrt'
+ mkdir -p /home/yph/openwrt/staging_dir/target-aarch64_cortex-a53_musl
+ cd /home/yph/openwrt/staging_dir/target-aarch64_cortex-a53_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/yph/openwrt/build_dir/target-aarch64_cortex-a53_musl/stamp
touch /home/yph/openwrt/staging_dir/target-aarch64_cortex-a53_musl/.prepared
make[3]: Entering directory '/home/yph/openwrt/toolchain/gdb'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/yph/openwrt/toolchain/gdb'
time: toolchain/gdb/compile#0.00#0.00#0.01
make[3]: Entering directory '/home/yph/openwrt/toolchain/binutils'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/yph/openwrt/toolchain/binutils'
time: toolchain/binutils/compile#0.00#0.00#0.01
make[3]: Entering directory '/home/yph/openwrt/toolchain/gcc/initial'
export SHELL="bash"; make  -C /home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial all-build-libiberty all-gcc all-target-libgcc
make[4]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial'
make[5]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/build-x86_64-pc-linux-gnu/libiberty'
make[6]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/build-x86_64-pc-linux-gnu/libiberty/testsuite'
make[5]: Leaving directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/build-x86_64-pc-linux-gnu/libiberty'
make[5]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/libiberty'
make[6]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/libiberty/testsuite'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/libiberty/testsuite'
make[5]: Leaving directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/libiberty'
make[5]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/lto-plugin'
make  all-am
make[6]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/lto-plugin'
make[6]: Leaving directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/lto-plugin'
make[5]: Leaving directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/lto-plugin'
make[5]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/intl'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/intl'
make[5]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/build-x86_64-pc-linux-gnu/fixincludes'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/build-x86_64-pc-linux-gnu/fixincludes'
make[5]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/build-x86_64-pc-linux-gnu/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[5]: Leaving directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/build-x86_64-pc-linux-gnu/libcpp'
make[5]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/zlib'
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CFLAGS=-O2 -I/home/yph/openwrt/staging_dir/host/include  -pipe  " "CXXFLAGS=-O2 -I/home/yph/openwrt/staging_dir/host/include  -pipe  " "CFLAGS_FOR_BUILD=-O2 -I/home/yph/openwrt/staging_dir/host/include  -pipe" "CFLAGS_FOR_TARGET=-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -g -Os" "INSTALL=/home/yph/openwrt/staging_dir/host/bin/install -c" "INSTALL_DATA=/home/yph/openwrt/staging_dir/host/bin/install -c -m 644" "INSTALL_PROGRAM=/home/yph/openwrt/staging_dir/host/bin/install -c" "INSTALL_SCRIPT=/home/yph/openwrt/staging_dir/host/bin/install -c" "LDFLAGS=-static-libstdc++ -static-libgcc " "LIBCFLAGS=-O2 -I/home/yph/openwrt/staging_dir/host/include  -pipe  " "LIBCFLAGS_FOR_TARGET=-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -g -Os" "MAKE=make" "MAKEINFO=/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0/missing makeinfo --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/home/yph/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl" "infodir=/home/yph/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/share/info" "libdir=/home/yph/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib" "prefix=/home/yph/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl" "tooldir=/home/yph/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/aarch64-openwrt-linux-musl" "AR=ar" "AS=as" "CC=gcc" "CXX=g++" "LD=ld" "LIBCFLAGS=-O2 -I/home/yph/openwrt/staging_dir/host/include  -pipe  " "NM=nm" "PICFLAG=" "RANLIB=ranlib" "DESTDIR=" DO=all multi-do # make
make[5]: Leaving directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/zlib'
make[5]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/libbacktrace'
make  all-am
make[6]: Entering directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/libbacktrace'
true  DO=all multi-do # make
make[6]: Leaving directory '/home/yph/openwrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/gcc-12.2.0-initial/libbacktrace'
[... 66 lines cut ...]
make[2]: *** [toolchain/Makefile:93: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/yph/openwrt'
make[1]: *** [toolchain/Makefile:89: /home/yph/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/yph/openwrt'
make: *** [/home/yph/openwrt/include/toplevel.mk:231: world] Error 2

### OpenWrt version

R23.3.3

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

1248289414, 2023-05-05:
I got the same problem, I found that it was caused by a lack of memory and when I closed the browser, it disappeared.
(PS: I used WSL to build OpenWRT and it made memory even more limited.

anonymous-one, 2023-07-20:
I can confirm. I had a VM with 4gb allocated and was seeing the same. As soon as I bumped it up to 8gb this disappreared.

Had to do a make clean first though.

This can be closed.