openwrt/openwrt #12223: mpc85xx: WS-AP3825i does not boot since 5.10 -> 5.15, unless CONFIG_SMP is disabled

Hurricos, 2023-03-20:
### Describe the bug

[Original forum post](https://forum.openwrt.org/t/adding-openwrt-support-for-ws-ap3825i/101168/138)

Long story short, as of the 5.10 -> 5.15 mpc85xx kernel bump, OpenWrt does not boot on the WS-AP3825i.

Of note, I did hear from @blocktrron that they have the board working on 5.15 -- I need to figure out what's different between our boards and builds.

I usually debug early-boot issues on this board using the WDT to reset and recover the kernel's __log_buf region -- to do this, I need to recompile with CONFIG_SMP disabled to get around some issues. As it turns out, disabling CONFIG_SMP fixes the issue, so I suspect the problem is during second-CPU bringup.

I'd like to bisect the kernel to do this; I have instructions from @stintel about using an external kernel tree (using linux-stable), but I got stuck trying to set up headers so the tree can compile.

### OpenWrt version

r22278-7dceef5ee5

### OpenWrt target/subtarget

mpc85xx/p1020

### Device

Extreme Networks WS-AP3825i

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

blocktrron, 2023-03-20:
While i can not say where things went south - The board is working when using the pre-populated device-tree with the legacy PPC bootwrapper. See https://git.openwrt.org/?p=openwrt/staging/blocktrron.git;a=commitdiff;h=dde920b2ec5dd316f2cf2770f7ba5cedbdc24e8f;hp=ba2cfb83cc71e8b7d77a1913a9d7472c14ef7af4

I prefer this solution instead of depending on the botched size-dependent boot-procedure it is currently using. This is also backwards-compatible with existing boot-procedures.

Hurricos, 2023-03-20:
Wonderful. I'm fine with any solution which ensures the kernel is allowed to read the memory where its device tree resides. From [the kernel docs](https://www.kernel.org/doc/html/latest/powerpc/bootwrapper.html), it sounds like U-Boot will be given a "decoy DTB" in the FIT image to patch, which the kernel will then ignore in favor of its own DTB when it starts.

AFAIK the only loss would be the MAC address setup, which (if it *does* actually break MAC addressing here) we can fix by just pulling it from the u-boot-env.

Hurricos, 2023-03-20:
I will add that the AP3825i's u-boot-env is writable by default, so if we absolutely needed to we could provide a migration script.

Hurricos, 2023-03-22:
I can confirm that commit ed82189339b8bcecfe709f1ce8f3500fb4aedab1 fixes this issue, even without 

```console
root@ap3825i0:~# dmesg | head -2 | tail -1
[    0.000000] Linux version 5.15.102 (builder@buildhost) (powerpc-openwrt-linux-musl-gcc (OpenWrt GCC 12.2.0 r22362-68883278ab) 12.2.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Tue Mar 21 19:49:21 2023
root@ap3825i0:~# strings /dev/mtd7 | grep boot_openwrt
boot_openwrt=cp.b 0xEC000000 0x2000000 0x2000000; interrupts off; bootm start 0x2000000; bootm loados; fdt resize; fdt boardsetup; fdt resize; fdt boardsetup; fdt chosen; fdt resize; fdt chosen; bootm prep; fdt resize; bootm prep; bootm go;
bootcmd=run boot_openwrt
root@ap3825i0:~# cat /etc/os-release | grep RELEASE
OPENWRT_RELEASE="OpenWrt SNAPSHOT r22362-68883278ab"
```

closed 2023-03-22 by Hurricos

Hurricos, 2023-03-26:
> I see you've filled out the CPU details already -- AFAIK the only other loss would be the MAC address labeling / setup, which (if it does actually break MAC addressing here) we can fix by just pulling it from the u-boot-env.

Unfortunately, there are other details U-Boot provides that we lose by feeding it a dummy-image. This change breaks SMP:

```console
root@ap3825i0:/# dmesg | grep -i cpu-release-addr -C 1
[    0.001061] smp: Bringing up secondary CPUs ...
[    0.001263] No cpu-release-addr for cpu 1
[    0.001269] smp: failed starting cpu 1 (rc -2)
```

... because U-Boot prepares the secondary CPU for bringup, but we no longer refer to U-Boot's patched device tree.

Before:

```dts

/memreserve/	0x00000000022d1b3c 0x00000000000034c4;
:
/ {
:
	cpus {
:
		PowerPC,P1020@1 {
:
			enable-method = "spin-table";
			cpu-release-addr = <0x00 0xffff2a0>;
			status = "disabled";
:
		};
	};
:
```

... now, with this fix:

```dts
:
/ {
:
	cpus {
:
:
		PowerPC,P1010@1 {
			bus-frequency = <0x17d783fc>;
			timebase-frequency = <0x2faf080>;
			clock-frequency = <0x2faf07f8>;
			d-cache-block-size = <0x20>;
			d-cache-size = <0x8000>;
			d-cache-sets = <0x80>;
			i-cache-block-size = <0x20>;
			i-cache-size = <0x8000>;
			i-cache-sets = <0x80>;
		};
	};
:
```

... causing the above kernel message.

Hurricos, 2023-03-26:
Filling out the exact details does not help either ... oh well.

Since avoiding FDT relocation does not work, and ignoring the FDT means we can never boot the second CPU, the next fix will have to involve a u-boot replacement. Oh well.

blocktrron, 2023-04-02:
Please don't immediately jump to the conclusion a U-Boot replacement / patch is the fix for any problem.

The issue here is the device overwriting the bootloader-provided bootpage in the zImage extraction phase. Please see my staging-tree for a potential fix.

Hurricos, 2023-04-15:
Hey @blocktrron: I have compiled based on [your staging tree](https://git.openwrt.org/?p=openwrt/staging/blocktrron.git;a=commitdiff;h=9caaa1d5ca1118cdf0271b94857b7cd9b857f9b2) and can confirm that AP3825i returns to 2CPU. It is reduced to 192M RAM due to how the CAM mapping works, but it's fine.