openwrt/openwrt #12228: port forwarding rule is rolled back as soon as it is applied.

m4ri0g, 2023-03-21:
### Describe the bug

create a port forwarding rule in luci. then save and apply. as soon as the rule is applied it dissapear from the list of port forwarding rules. 

[image]

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT3200ACM

### Image kind

Official downloaded image

### Steps to reproduce

add a new port forwarding rule in network > firewall > port forwards 

### Actual behaviour

when the rule is saved. it dissapears without any error message 

### Expected behaviour

the port forward should not be removed when applied. 

### Additional info

```
`root@OpenWrt:~# cat /etc/config/firewall 

config defaults
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option synflood_protect '1'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config zone
	option name 'vpn'
	option output 'ACCEPT'
	option forward 'REJECT'
	list network 'wg0'
	option input 'REJECT'
	option masq '1'

config forwarding
	option src 'lan'
	option dest 'vpn'

config include 'pbr'
	option fw4_compatible '1'
	option type 'script'
	option path '/usr/share/pbr/pbr.firewall.include'`
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-03-21:
Can't reproduce:
[image]
[image]
[image]

closed 2023-03-21 by jow-

m4ri0g, 2023-03-21:
Hello 
thank you for this. 
i have the package pbr installed. could this be the cause the the port forwarding being rolled back?  
[image]

HiDanielHere, 2023-08-06:
Hi, I think the bug @m4ri0g has described is real. I have the same router and also kind of the same port forwarding + pbr issue. The strange thing is that when I apply a new port forwarding rule it is active kind of a second and then rolls back despite being showed in the UI. But the result is the same the rule does not getting applied in combination with pbr. Can anyone give support on how to fix this?

m4ri0g, 2023-08-06:
> Hi, I think the bug @m4ri0g has described is real and I ask you to please open the issue again @jow-. I have the same router and also kind of the same port forwarding + pbr issue. The strange thing is that when I apply a new port forwarding rule it is active kind of a second and then rolls back despite being showed in the UI. But the result is the same the rule does not getting applied in combination with pbr. Can anyone give support on how to fix this?

the problem was indeed PBR. I uninstalled the pbr and could create the port forwarding rule...
so there is something messing with the port forwaring rules when PBR is installed. unfortunately i don't know if its a openwrt issue or have to be taken over with PBR developers. they seems to playing a chicken egg problem blaming each other.

HiDanielHere, 2023-08-06:
Thank you for your answer @m4ri0g. How did you replace the pbr service then? I want to make a VPN split tunneling so that my mail server does have an exception for not routing via default gateway WireGuard VPN..

m4ri0g, 2023-08-06:
> Thank you for your answer @m4ri0g. How did you replace the pbr service then? I want to make a VPN split tunneling so that my mail server does have an exception for not routing via default gateway WireGuard VPN..

I did not replace the PBR service but I had to remove it to be able to create a port forwarding rule and i reinstalled it.

TuncTaylan, 2024-12-25:
I too have the same issue, after installing and configuring PBR the port forwards stop working.