openwrt/openwrt #12229: mac80211: ath11k: sync with ath-next: Xiaomi AX9000 QCN9024 160MHz issue

dick902, 2023-03-21:
### Describe the bug

After apply mac80211: ath11k sync with ath-next update on Xiaomi AX9000
The PCI Wifi QCN9024 is missing 160MHz Wifi option.

 

### OpenWrt version

r22344-ca330cac92

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX9000

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Gost6, 2023-03-23:
The qnap qhora-301w also is affected, missing option for 160mhz  channels width.

robimarko, 2023-03-28:
This should fix it #12274

Soromeister, 2023-04-01:
Confirmed fixed in build `r22475-936df715de`

dick902, 2023-04-01:
Confirm fixed r22475-936df715de

closed 2023-04-01 by dick902