openwrt/openwrt #12232: snapshot builds do not install correctly - Mar 2023 builds

infantvin, 2023-03-21:
### Describe the bug

I am flashing the snapshot build using the LUCI GUI on my MR8300 and whenever the flash is done, the router does not boot. In fact, it never reboots post the flash. I have waited for more than 5-10 minutes in cases and found that the router never responds.

I have to turn off power 3 times to get back to the previous firmware that was installed. This means that whatever version is being installed does not work. I am not sure what logs need to be collected to help you analyze the issue, but if you need please send me information on how to gather this and I will send it across to you.

All snapshot builds are downloaded from here
https://downloads.openwrt.org/snapshots/targets/ipq40xx/generic/

File name is linksys_mr8300-squashfs-sysupgrade.bin

I have tried with the files since Mar 18 until Mar 21 2023 and faced the same problem with all of them.

Thanks.

### OpenWrt version

NA as the router cannot be reached post the flash

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys MR8300 (Dallas)

### Image kind

Official downloaded image

### Steps to reproduce

Run a MR8300 with any existing openwrt image from say Feb or so and then flash using the snapshot images from any day from Mar 18 until today. Router will not boot and the wired connection from the PC to the router will keep saying its trying to connect and fails. This can happen indefinitely and does not correct over time.

### Actual behaviour

orange LED flashes signifying that the flash is happening. However, the router never reboots like it is supposed to and it just stays with the blue LED steady and does not respond to pings to the 192.168.1.1 address. DHCP assignment to a wired device will fail as the device will fail to reach the router which indicates that its not responding.

### Expected behaviour

Normally, the image flash will cause the router to reboot, the blue LED to blink and then it becomes steady indicating the router has booted.

The pings to 192.168.1.1 should be successful.

### Additional info

Please let me know if you need anything. I can get it for you. However the flash itself has failed and I cannot reach the router when this issue happens, so I really dont know what information to gather and give to you.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

schm0, 2023-03-22:
Something is broken.
Everything comes up as normal but nothing network related works.
Simple ping results in: Operation not permitted.
But the wan interface gets an IP through DHCP just fine.
So maybe this is nftables/firewall related?
There was an update to nftables in master.

infantvin, 2023-03-22:
Thanks for verifying. So, is there any action needed from my side or is this something that you guys will fix and rollout in the next build?

anomeome, 2023-03-22:
[forum thread](https://forum.openwrt.org/t/openwrt-snapshot-r22302-on-mr8300/154878)

andyleadbetter, 2023-03-23:
Also happening on RT3200 Mediatek builds.  Tried snapshots on Tuesday, exactly the same issue.  2nd AP obtained DHCP from the first router running 22.03, but no traffic passed from the AP, also wireless did not come up either.  Tried reset to factory on the snapshot still no access, ended up going back to 22.03.03 release.

infantvin, 2023-03-23:
@andyleadbetter  thanks for adding your findings.
@schm0 , if you have some dump of the log where you saw the messages of what is not working, can you please add or attach here.

Thank you

infantvin, 2023-03-28:
Hi

Is this fixed in the builds released now or do we have to wait? Please confirm

Thanks

clauded, 2023-03-29:
I tested yesterday's build (MR8300) and the problem is still present.

Rygle, 2023-04-01:
I've had all sorts of problems with recent snapshot builds including lost DNS or Wifi but no internet with Netgear WAX206. The only way to fix is to revert to an earlier snapshot. Have tried resetting everything and installing factory image but no good. Something fundamental has broken in the last few days and only reverting can fix it.

roblad, 2023-04-02:
any news, I have the same problem

DJMalachite, 2023-04-02:
After seeing this i should keep my router on SNAPSHOT r22253-89390a9e50 but what does this mean for the future is it fixable?

clauded, 2023-04-02:
A serial log has been published on the forum: [https://forum.openwrt.org/t/openwrt-snapshot-r22302-on-mr8300/154878/33](https://forum.openwrt.org/t/openwrt-snapshot-r22302-on-mr8300/154878/33).

The last 3 lines of the log are:

`
[    2.273568] ubi0: attaching mtd11
[    2.273612] ubi0 error: ubi_attach_mtd_dev: VID header offset 2048 too large.
[    2.284388] Waiting for root device /dev/ubiblock0_0...
`

badulesia, 2023-04-02:
Hi @clauded 
I can read on your env variables.
> kernsize=300000
In order to install and boot a snapshot you need kernsize set to 500000.

roblad, 2023-04-02:
Hi, that was on both settings, currently I have

root@MR8300:~# fw_printenv
altkern=5f80000
auto_recovery=yes
baudrate=115200
boot_part=1
boot_part_ready=3
boot_ver=1.3.3
bootcmd=if test $auto_recovery = no; then bootipq; elif test $boot_part = 1; then run bootpart1; else run bootpart2; fi
bootdelay=2
bootpart1=set bootargs $partbootargs && nand read $loadaddr $prikern $kernsize && bootm $loadaddr
bootpart2=set bootargs $partbootargs2 && nand read $loadaddr $altkern $kernsize && bootm $loadaddr
ethact=eth0
ethaddr=00:03:7f:ba:db:ad
flash_type=2
flashimg=tftp $loadaddr $image && nand erase $prikern $imgsize && nand write $loadaddr $prikern $filesize
flashimg2=tftp $loadaddr $image && nand erase $altkern $imgsize && nand write $loadaddr $altkern $filesize
image=rogue.img
imgsize=5800000
ipaddr=192.168.1.1
kernsize=500000
loadaddr=84000000
machid=8010006
netmask=255.255.255.0
partbootargs=init=/sbin/init rootfstype=ubifs ubi.mtd=11,2048 root=ubi0:ubifs rootwait rw
partbootargs2=init=/sbin/init rootfstype=ubifs ubi.mtd=13,2048 root=ubi0:ubifs rootwait rw
prikern=780000
serverip=192.168.1.254
stderr=serial
stdin=serial
stdout=serial

efect was the same

infantvin, 2023-04-03:
I can confirm that this is happening even with the kernsize variable set. Nothing had been changed on my router when this started happening and a previous snapshot flash (before Mar 18) worked as well. My router model is MR8300.

roblad, 2023-04-03:
I restored to stock, re-flashed couple of times in both partitions, I uploaded new build r20090-c2331038b2-ipq40xx factory.pin and the 01 partition hung as previously, did not boot. Does not mater if 3GB or 5GB kernel partition - effect is the same.

look into:

https://eko.one.pl/forum/viewtopic.php?id=22098&p=2

unfortunately it is in Polish, but google can manage :-)

I re-flashed to stock again and uploaded factory bin before 18 Martch,  r20068-788a0cf15c-ipq40xx, and everything is OK, I think that the change of USB or the sequence in the sysupgrade could cause that, the change was missed previously, but I can be wrong, the lib is:

/lib/upgrade/linksys.sh

cur_boot_part="$(/usr/sbin/fw_printenv -n boot_part)"
    if [ -z "${cur_boot_part}" ]; then
        mtd_ubi0=$(cat /sys/devices/virtual/ubi/ubi0/mtd_num)
        case "$(grep -E "^mtd${mtd_ubi0}:" /proc/mtd | cut -d '"' -f 2)" in
        kernel|rootfs)
            cur_boot_part=1
            ;;
        alt_kernel|alt_rootfs)
            cur_boot_part=2
            ;;
        esac
        >&2 printf "Current boot_part='%s' selected from ubi0/mtd_num='%s'" \
            "${cur_boot_part}" "${mtd_ubi0}"
    fi

    # OEM U-Boot for EA6350v3, EA8300 and MR8300; bootcmd=
    #  if test $auto_recovery = no;
    #      then bootipq;
    #  elif test $boot_part = 1;
    #      then run bootpart1;
    #      else run bootpart2;
    #  fi

    case "$cur_boot_part" in
    1)
        fw_setenv -s - <<-EOF
            boot_part 2
            auto_recovery yes
        EOF
        printf "alt_kernel"
        return
        ;;
    2)
        fw_setenv -s - <<-EOF
            boot_part 1
            auto_recovery yes
        EOF
        printf "kernel"
        return
        ;;
    *)
        return
        ;;
    esac
}

badulesia, 2023-04-03:
Hi.
You can't boot any snapshot, valid or invalid unless you set kernel size to 5MB. So keep it to 5MB.
I have always flashed factory image from Linksys OEM, so the sysupgrade process is not involved. Result is always invalid, so the issue is within the OpenWrt image.
As of an USB related issue, I doubt about this but ... who knows, you may be right, I've seen so many weird things. What about to create an image without any USB related files to check? The error log I posted show an error mounting mtd11, before any related USB file load or setup had occured.
I will also try to flash OpenWrt on mtd13 to see it the same partition boot error occurs.

badulesia, 2023-04-03:
Hi again
Just tested to boot a snapshot with any USB related file, on mtd13. Same result.

roblad, 2023-04-04:
it means that something more was changed and was undocumented

clauded, 2023-04-04:
Could it be kernel related?
https://lore.kernel.org/lkml/20230306013308.3884777-1-chengzhihao1@huawei.com/t/

dseven, 2023-04-10:
Has anyone narrowed down the openwrt commit where this problem started occurring? I just did a build from master today, and ran into it for the first time. My previous build, which did not have the problem, was from 2023-02-23. Narrowing it down is a tedious process, but may be the only way forward. I may work on it if no one else has, but my only MR8300 is my "daily driver". 

BTW, https://github.com/openwrt/openwrt/pull/11692 was merged recently. I don't know if it has anything to do with this issue or not (probably not?), but it may be something to rule out....

infantvin, 2023-04-10:
Please see the first post from user [schm0](https://github.com/schm0)
He said it may have something to do with nftables as disabling nftables seemed to have worked to help the boot process complete.
Maybe that is the cause here and if someone has access to that serial console, they can test if disabling nftables actually helps here to get the flashed image working normally. I dont have the serial cable, so I am unable to help.

badulesia, 2023-04-10:
Have you seen the boot logs I published ? This is a partition detection issue due to a change in kernel. nftables caused an issue a few weeks ago but is not involved here.

roblad, 2023-04-10:
Probably it was not applied yet: v3 https://lore.kernel.org/lkml/20230324161923.1456371-1-nschichan@freebox.fr/t/" : "

Re: [PATCH -next v3] ubi: Fix failure attaching when vid_hdr offset equals to (sub)page size
  2023-03-29 15:54     ` [Miquel Raynal](https://lore.kernel.org/lkml/20230306013308.3884777-1-chengzhihao1@huawei.com/t/#me8d637d5b9e98b30d4a1ef63910304b7a8f07ff4)
@ 2023-03-29 21:33       ` Richard Weinberger
  [0 siblings, 0 replies; 5+ messages in thread](https://lore.kernel.org/lkml/20230306013308.3884777-1-chengzhihao1@huawei.com/t/#rf70bf5ffebf2188c3597feb9d977e2fd49797349)
From: Richard Weinberger @ 2023-03-29 21:33 UTC ([permalink](https://lore.kernel.org/lkml/53992166.434647.1680125620704.JavaMail.zimbra@nod.at/) / [raw](https://lore.kernel.org/lkml/53992166.434647.1680125620704.JavaMail.zimbra@nod.at/raw))
  To: Miquel Raynal
  Cc: chengzhihao1, Nicolas Schichan, George Kennedy, linux-kernel,
	linux-mtd, Sascha Hauer, yi zhang

----- Ursprüngliche Mail -----
> Von: "Miquel Raynal" <miquel.raynal@bootlin.com>
>> Thanks for testing.
>> 
>> > Tested-by: Nicolas Schichan <nschichan@freebox.fr>
> 
> Same here.
> 
> Tested-by: Miquel Raynal <miquel.raynal@bootlin.com> # v5.10, v4.19

**### Applied to next, PR will follow soon.**

Thanks everyone!
//richard

infantvin, 2023-04-10:
> Have you seen the boot logs I published ? This is a partition detection issue due to a change in kernel. nftables caused an issue a few weeks ago but is not involved here.
> 
> Here is a commit that may be the origin of the issue https://lore.kernel.org/lkml/20230306013308.3884777-1-chengzhihao1@huawei.com/t/

I did not notice this. If that is the case, then I think I will wait until that partition problem is fixed.

infantvin, 2023-04-10:
> Probably it was not applied yet: v3 https://lore.kernel.org/lkml/20230324161923.1456371-1-nschichan@freebox.fr/t/ :
> 
> Re: [PATCH -next v3] ubi: Fix failure attaching when vid_hdr offset equals to (sub)page size 2023-03-29 15:54 `[Miquel Raynal](https://lore.kernel.org/lkml/20230306013308.3884777-1-chengzhihao1@huawei.com/t/#me8d637d5b9e98b30d4a1ef63910304b7a8f07ff4) @ 2023-03-29 21:33 ` Richard Weinberger [0 siblings, 0 replies; 5+ messages in thread](https://lore.kernel.org/lkml/20230306013308.3884777-1-chengzhihao1@huawei.com/t/#rf70bf5ffebf2188c3597feb9d977e2fd49797349) From: Richard Weinberger @ 2023-03-29 21:33 UTC ([permalink](https://lore.kernel.org/lkml/53992166.434647.1680125620704.JavaMail.zimbra@nod.at/) / [raw](https://lore.kernel.org/lkml/53992166.434647.1680125620704.JavaMail.zimbra@nod.at/raw)) To: Miquel Raynal Cc: chengzhihao1, Nicolas Schichan, George Kennedy, linux-kernel, linux-mtd, Sascha Hauer, yi zhang
> 
> ----- Ursprüngliche Mail -----
> 
> > Von: "Miquel Raynal" [miquel.raynal@bootlin.com](mailto:miquel.raynal@bootlin.com)
> > > Thanks for testing.
> > > > Tested-by: Nicolas Schichan [nschichan@freebox.fr](mailto:nschichan@freebox.fr)
> > 
> > 
> > Same here.
> > Tested-by: Miquel Raynal [miquel.raynal@bootlin.com](mailto:miquel.raynal@bootlin.com) # v5.10, v4.19
> 
> **### Applied to next, PR will follow soon.**
> 
> Thanks everyone! //richard

One doubt, this looks like this thread speaks of the 5.10 and 4.x kernel. The snapshot builds seem to be using 5.15.x of the kernel. Does this apply to that too? Just a doubt I had.

infantvin, 2023-04-10:
Ok that makes sense. Thanks. We will have to wait till them till its implemented I think.

dseven, 2023-04-10:
I've applied the "v3" patch mentioned above to my build, and confirm that it boots successfully. I suppose it has to become part of an upstream kernel release, and that release has to be adopted by OpenWRT. Hopefully it won't be too long....

roblad, 2023-04-10:
I hope that both threads 5.10.x and 5.15.x

badulesia, 2023-04-11:
Good news all, thank you.

roblad, 2023-04-12:
may be someone can try to build  his own image and test

https://firmware-selector.openwrt.org/?version=22.03.3&target=ipq40xx%2Fgeneric&id=linksys_mr8300

badulesia, 2023-04-12:
That is the latest stable version, it works since january!

roblad, 2023-04-12:
Ok, on that with option installed packages to add 

https://firmware-selector.openwrt.org/?version=22.03.4&target=ipq40xx%2Fgeneric&id=linksys_mr8300

that does not work

badulesia, 2023-04-12:
It won't work since the kernel version has not the patch. We must wait for a future 22.03 snapshot or/and 22.03.5.

closed 2023-04-15 by openwrt-bot

roblad, 2023-04-15:
I confirm, it works now

Nimrod-666, 2023-04-16:
Got the same issue trying to update on my EA8300. So it seems not only MR8300 is affected. Maybe someone should update the release guido of that release and include the warning for EA8300 too. Or remove the images for these targets as well.

hhasert, 2023-04-16:
It is fixed in the latest snapshot build at 
https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/ipq40xx/generic/

Tested on EA8300 and works fine now.

badulesia, 2023-04-16:
Same for the MR8300.

commit aad34818 (2023-04-15): kernel: backport fix for recently introduced UBI bug

Import commit "ubi: Fix failure attaching when vid_hdr offset equals to
(sub)page size" which did not yet make it to stable upstream Linux trees.

Fixes: #12232
Fixes: #12339
PR #12398 (2023-04-15): kernel: backport fix for recently introduced UBI bug
Import commit "ubi: Fix failure attaching when vid_hdr offset equals to (sub)page size" which did not yet make it to stable upstream Linux trees.

Fixes: #12232
Fixes: #12339
commit 074db579 (2023-04-15): kernel: backport fix for recently introduced UBI bug

Import commit "ubi: Fix failure attaching when vid_hdr offset equals to
(sub)page size" which did not yet make it to stable upstream Linux trees.

Fixes: #12232
Fixes: #12339

(cherry picked from commit aad34818b50029e07ed9221ae46f9770d6e29785)
PR #12399 (2023-04-15): [22.03] kernel: backport fix for recently introduced UBI bug
Import commit "ubi: Fix failure attaching when vid_hdr offset equals to (sub)page size" which did not yet make it to stable upstream Linux trees.

Fixes: #12232
Fixes: #12339

(cherry picked from commit aad34818b50029e07ed9221ae46f9770d6e29785)