openwrt/openwrt #12239: xdp-tools: compile failure due to reference to host asm/types.h

mcbridematt, 2023-03-22:
### Describe the bug

xdp-tools 1.2.9 fails to compile when the compile host does not have a `/usr/include/asm` directory.

An example of such a system could be Debian without the `gcc-multilib` package installed:
```
$ readlink -f /usr/include/asm
ls: cannot access '/usr/include/asm': No such file or directory
$ apt-get install -y gcc-multilib
$ readlink -f /usr/include/asm
/usr/include/x86_64-linux-gnu/asm
```

I have reproduced on another distribution (openSUSE Tumbleweed) by temporarily renaming /usr/include/asm to /usr/include/asm_

```
$ make package/xdp-tools/compile V=s
Applying ./patches/010-configure-respect-LDFLAGS.patch using plaintext:
patching file configure
touch /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/.prepared_c5ea8befdd5bf98b701a960ee0b812b2_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/.configured_*
rm -f /home/matt/openwrt-vanilla/staging_dir/target-aarch64_cortex-a53_musl/stamp/.xdp-tools_installed
(cd /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/./; if [ -x ./configure ]; then find /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/ -name config.guess | xargs -r chmod u+w; find /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/ -name config.guess | xargs -r -n1 cp --remove-destination /home/matt/openwrt-vanilla/scripts/config.guess; find /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/ -name config.sub | xargs -r chmod u+w; find /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/ -name config.sub | xargs -r -n1 cp --remove-destination /home/matt/openwrt-vanilla/scripts/config.sub; AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9=xdp-tools-1.2.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  " LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9=xdp-tools-1.2.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9=xdp-tools-1.2.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " CPPFLAGS="-I/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/include -I/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include/fortify -I/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include   " LDFLAGS="-L/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/lib -L/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib -znow -zrelro    "  FORCE_SYSTEM_LIBBPF=1 CC="aarch64-openwrt-linux-musl-gcc" CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9=xdp-tools-1.2.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  " LDFLAGS="-L/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/lib -L/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib -znow -zrelro   " CLANG="/home/matt/openwrt-vanilla/llvm-bpf/bin/clang" BPF_TARGET="bpfel" LLC="/home/matt/openwrt-vanilla/llvm-bpf/bin//llc"  ./configure --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
Found clang binary '/home/matt/openwrt-vanilla/llvm-bpf/bin/clang' with version 15 (from 'clang version 15.0.7 (https://git.openwrt.org/openwrt/openwrt.git 00b40ef00d42b9b384e72aea604bc53e39e24ac3)')
libbpf support: system v1.1.0
  perf_buffer__consume support: yes
  btf__load_from_kernel_by_id support: yes
  btf__type_cnt support: yes
  bpf_object__next_map support: yes
  bpf_object__next_program support: yes
  bpf_program__insn_cnt support: yes
  bpf_map_create support: yes
  perf_buffer__new_raw support: yes
  bpf_xdp_attach support: yes
secure_getenv support: yes
echo "BPF_CFLAGS += -I/home/matt/openwrt-vanilla/staging_dir/target-aarch64_cortex-a53_musl/bpf-headers/tools/lib" >> /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/config.mk
touch /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/.configured_a17fb5ef857664f03cd0ce37cc5ea591
rm -f /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/.built
touch /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/.built_check
CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9=xdp-tools-1.2.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/include -I/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include/fortify -I/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include   " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9=xdp-tools-1.2.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/include -I/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include/fortify -I/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include   " LDFLAGS="-L/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/lib -L/home/matt/openwrt-vanilla/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib -znow -zrelro    " PREFIX=/usr RUNDIR=/tmp/run make  -C /home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/matt/openwrt-vanilla/build_dir/target-aarch64_cortex-a53_musl/xdp-tools-1.2.9=xdp-tools-1.2.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  " LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;

lib

  libxdp
    CC       staticobjs/libxdp.o
    CC       staticobjs/xsk.o
    M4       xdp-dispatcher.c
    CLANG    xdp-dispatcher.o
In file included from xdp-dispatcher.c:3:
In file included from ../../headers/linux/bpf.h:11:
[... cut ...]
CONFIG_TARGET_armvirt=y
CONFIG_TARGET_armvirt_64=y
CONFIG_TARGET_armvirt_64_Default=y
CONFIG_PACKAGE_kmod=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libxdp=y
CONFIG_PACKAGE_xdp-filter=y
CONFIG_PACKAGE_xdp-loader=y
CONFIG_PACKAGE_xdpdump=y
CONFIG_PACKAGE_zlib=y
# CONFIG_TARGET_ROOTFS_CPIOGZ is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

APCCV, 2023-06-14:
Same issue. Unable to build (tag) v23.05.0-rc1 for ipq806x. Using default configuration taken from OpenWrt repo.
Installing gcc-multilib on Ubuntu 22.04 build machine gets build to succeed.
I agree with OP that build should not be relying on host machine includes but be self contained.

MaZderMind, 2023-09-03:
Still fails on current `master`.

credomane, 2023-09-26:
Just ran into this myself trying to compile the current stable 22.03.5 while working on support for Ubiquiti EdgeRouter 12.

Installed gcc-multilib per APCCV's comment. Waiting for the compile to fail/succeed now.

donmr, 2023-09-27:
Still in v23.05.0-rc3  (cd14b17cb00cda45819739aa63422a090e3f80e1)

taylorkline, 2024-03-17:
Same problem here, and cannot simply install `gcc-multilib`, because `gcc-multilib` is [not available for `aarm64`](https://packages.debian.org/search?keywords=gcc-multilib).

dudeitssm, 2024-06-03:
> Same issue. Unable to build (tag) v23.05.0-rc1 for ipq806x. Using default configuration taken from OpenWrt repo. Installing gcc-multilib on Ubuntu 22.04 build machine gets build to succeed. I agree with OP that build should not be relying on host machine includes but be self contained.

Thanks for that suggestion! :+1: 
Installing `gcc-multilib` on Debian 12.5 allowed my `v23.05.3` build to compile successfully.

mikegleasonjr, 2024-07-24:
Same for `gcc-multilib` with  `v23.05.4`

marcalbert74, 2024-08-13:
I cannot install  gcc-multilib as it is not available for arm64. Has anyone found a workaround to get this to compile?

diegonc, 2024-09-17:
Is xdp-tools being cross compiled here?

It's not clear to me since it uses clang from $(STAGING_DIR)/host and I'm not very familiar with openwrt builds.
But if it is, installing gcc-multilib makes no sense, I believe. Instead, shouldn't it be instructed to use the  target headers?

okaestne, 2024-10-06:
I sent a patch to the ML. Maybe one of you has time to test this?
http://lists.openwrt.org/pipermail/openwrt-devel/2024-October/043232.html

diegonc, 2024-10-08:
> I sent a patch to the ML. Maybe one of you has time to test this? http://lists.openwrt.org/pipermail/openwrt-devel/2024-October/043232.html

I'm installing a clean Debian x86_64 system. I'll test this as soon as I can get a clone of OpenWrt.

diegonc, 2024-10-09:
I finally checkout the main branch but I couldn't reproduce the exact error. I found instead:

```
xdp-filter
make[5]: Entering directory '/home/diego/dev/openwrt/src/build_dir/target-mips_24kc_musl/xdp-tools-1.4.2/xdp-filter'
/home/diego/dev/openwrt/src/staging_dir/host/llvm-bpf/bin/clang -S \
    -target mips-linux-gnu \
    -D __BPF_TRACING__ \
    -Wall \
    -Wno-unused-value \
    -Wno-pointer-sign \
    -Wno-compare-distinct-pointer-types \
    -Werror \
    -Wno-visibility -fno-stack-protector -nostdinc -isystem /home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/lib/gcc/*/*/include -isystem/home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/usr/include -isystem/home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/include -isystem/home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/include/fortify -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/arch/mips/include -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/arch/mips/include/asm/mach-generic -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/arch/mips/include/generated -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/include -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/arch/mips/include/uapi -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/arch/mips/include/generated/uapi -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/include/uapi -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/include/generated/uapi -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/tools/lib -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/tools/testing/selftests -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/samples/bpf -include linux/kconfig.h -include asm_goto_workaround.h -I/home/diego/dev/openwrt/src/build_dir/target-mips_24kc_musl/xdp-tools-1.4.2 -D__KERNEL__ -D__BPF_TRACING__ -DCONFIG_GENERIC_CSUM -D__TARGET_ARCH_mips -mbig-endian -fno-stack-protector -Wall -Wno-unused-value -Wno-pointer-sign -Wno-compare-distinct-pointer-types -Wno-gnu-variable-sized-type-not-at-end -Wno-address-of-packed-member -Wno-tautological-compare -Wno-unknown-warning-option -fno-asynchronous-unwind-tables -Wno-uninitialized -Wno-unused-variable -Wno-unused-label -O2 -emit-llvm -Xclang -disable-llvm-passes -Wno-error -fno-stack-protector -DBPF_DIR_MNT=\"/sys/fs/bpf\" -DBPF_OBJECT_PATH=\"/usr/lib/bpf\" -DMAX_DISPATCHER_ACTIONS=10 -DTOOLS_VERSION=\""1.4.2"\" -DLIBBPF_VERSION=\"1.4.5\" -DRUNDIR=\"/tmp/run\" -DHAVE_LIBBPF_PERF_BUFFER__CONSUME -DHAVE_LIBBPF_BTF__LOAD_FROM_KERNEL_BY_ID -DHAVE_LIBBPF_BTF__TYPE_CNT -DHAVE_LIBBPF_BPF_OBJECT__NEXT_MAP -DHAVE_LIBBPF_BPF_OBJECT__NEXT_PROGRAM -DHAVE_LIBBPF_BPF_PROGRAM__INSN_CNT -DHAVE_LIBBPF_BPF_PROGRAM__TYPE -DHAVE_LIBBPF_BPF_PROGRAM__FLAGS -DHAVE_LIBBPF_BPF_PROGRAM__EXPECTED_ATTACH_TYPE -DHAVE_LIBBPF_BPF_MAP_CREATE -DHAVE_LIBBPF_PERF_BUFFER__NEW_RAW -DHAVE_LIBBPF_BPF_XDP_ATTACH -DHAVE_LIBBPF_BPF_MAP__SET_AUTOCREATE -DHAVE_LIBBPF_BPF_PROG_TEST_RUN_OPTS -DHAVE_LIBBPF_BPF_XDP_QUERY -DHAVE_SECURE_GETENV -DDEBUG -DLIBBPF_DYNAMIC -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64   -nostdinc -isystem /home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/lib/gcc/*/*/include -isystem/home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/usr/include -isystem/home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/include -isystem/home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/include/fortify -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/arch/mips/include -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/arch/mips/include/asm/mach-generic -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/arch/mips/include/generated -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/include -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/arch/mips/include/uapi -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/arch/mips/include/generated/uapi -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/include/uapi -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/include/generated/uapi -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/tools/lib -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/tools/testing/selftests -I/home/diego/dev/openwrt/src/staging_dir/target-mips_24kc_musl/bpf-headers/samples/bpf -include linux/kconfig.h -include asm_goto_workaround.h -I/home/diego/dev/openwrt/src/build_dir/target-mips_24kc_musl/xdp-tools-1.4.2 -D__KERNEL__ -D__BPF_TRACING__ -DCONFIG_GENERIC_CSUM -D__TARGET_ARCH_mips -mbig-endian -fno-stack-protector -Wall -Wno-unused-value -Wno-pointer-sign -Wno-compare-distinct-pointer-types -Wno-gnu-variable-sized-type-not-at-end -Wno-address-of-packed-member -Wno-tautological-compare -Wno-unknown-warning-option -fno-asynchronous-unwind-tables -Wno-uninitialized -Wno-unused-variable -Wno-unused-label -O2 -emit-llvm -Xclang -disable-llvm-passes -Wno-error -fno-stack-protector -DBPF_DIR_MNT=\"/sys/fs/bpf\" -DBPF_OBJECT_PATH=\"/usr [... cut ...]
CONFIG_TARGET_ath79_generic_DEVICE_8dev_carambola2=y
CONFIG_DEVEL=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_PACKAGE_kmod=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libxdp=y
CONFIG_PACKAGE_xdp-filter=y
CONFIG_PACKAGE_xdp-loader=y
CONFIG_PACKAGE_xdpdump=y
CONFIG_PACKAGE_zlib=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
CONFIG_USE_LLVM_BUILD=y```

any ideas?

okaestne, 2024-10-09:
@diegonc Thanks for testing! My patch is for the openwrt-23.05 branch, but I need to look into getting this running on main as well.

diegonc, 2024-10-09:
Ok, good I switched to 23.05.5 then. After seeing it fail I'll apply the patch and rebuild.

diegonc, 2024-10-10:
@okaestne After applying the patch I got the following erorr:

```
make[4]: Entering directory '/home/diego/dev/openwrt/src/build_dir/target-mips_24kc_musl/xdp-tools-1.2.9'

lib

  libxdp
    CLANG    xdp-dispatcher.o
In file included from xdp-dispatcher.c:4:
In file included from /home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-12.3.0_musl/lib/gcc/mips-openwrt-linux-musl/12.3.0/../../../../mips-openwrt-linux-musl/sys-include/linux/in.h:335:
/home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-12.3.0_musl/lib/gcc/mips-openwrt-linux-musl/12.3.0/../../../../mips-openwrt-linux-musl/sys-include/asm/byteorder.h:17:3: error: "MIPS, but neither __MIPSEB__, nor __MIPSEL__???"
# error "MIPS, but neither __MIPSEB__, nor __MIPSEL__???"
  ^
1 error generated.
make[6]: *** [Makefile:139: xdp-dispatcher.o] Error 1
make[5]: *** [Makefile:20: libxdp] Error 2
make[4]: *** [Makefile:25: lib] Error 2
```

What should I change for either __MIPSEB__ or __MIPSEL__ to be defined?

okaestne, 2024-10-10:
> @okaestne After applying the patch I got the following erorr:
> 
> ```
> make[4]: Entering directory '/home/diego/dev/openwrt/src/build_dir/target-mips_24kc_musl/xdp-tools-1.2.9'
> 
> lib
> 
>   libxdp
>     CLANG    xdp-dispatcher.o
> In file included from xdp-dispatcher.c:4:
> In file included from /home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-12.3.0_musl/lib/gcc/mips-openwrt-linux-musl/12.3.0/../../../../mips-openwrt-linux-musl/sys-include/linux/in.h:335:
> /home/diego/dev/openwrt/src/staging_dir/toolchain-mips_24kc_gcc-12.3.0_musl/lib/gcc/mips-openwrt-linux-musl/12.3.0/../../../../mips-openwrt-linux-musl/sys-include/asm/byteorder.h:17:3: error: "MIPS, but neither __MIPSEB__, nor __MIPSEL__???"
> # error "MIPS, but neither __MIPSEB__, nor __MIPSEL__???"
>   ^
> 1 error generated.
> make[6]: *** [Makefile:139: xdp-dispatcher.o] Error 1
> make[5]: *** [Makefile:20: libxdp] Error 2
> make[4]: *** [Makefile:25: lib] Error 2
> ```
> 
> What should I change for either `__MIPSEB__` or `__MIPSEL__ `to be defined? Maybe I should change the board?
> 
> The asm/types.h include error is gone!

For this error there should be a patch in there for the xdp-tools package Makefile.

```patch
diff --git a/package/network/utils/xdp-tools/Makefile b/package/network/utils/xdp-tools/Makefile
index 3cc7980cf6..9f9a20b10b 100644
--- a/package/network/utils/xdp-tools/Makefile
+++ b/package/network/utils/xdp-tools/Makefile
@@ -95,6 +95,9 @@ MAKE_VARS += \
 define Build/Configure
 	$(call Build/Configure/Default)
 	echo "BPF_CFLAGS += -I$(BPF_HEADERS_DIR)/tools/lib" >> $(PKG_BUILD_DIR)/config.mk
+ifeq ($(ARCH),mipsel)
+	echo "BPF_CFLAGS += -D__MIPSEL__" >> $(PKG_BUILD_DIR)/config.mk
+endif
 endef
 
 define Build/InstallDev
```

Has it been applied?

Thanks again!

diegonc, 2024-10-10:
>
> Has it been applied?
> 

Yes, the `ifeq` is there. It seems I'm failing at setting the ARCH to mipsel 🤔

diegonc, 2024-10-11:
As a side note, there's one of the patches that I don't think work as expected:

<details>
  <summary>012-configure-fix-arch-includes.patch</summary>

```diff
From 6ce30637951949212344f1968c3fd5e5adf1ffc3 Mon Sep 17 00:00:00 2001
From: Sachin Tiptur <coolsachints at gmail.com>
Date: Wed, 13 Sep 2023 12:47:31 +0000
Subject: [PATCH] Fix build on arm64

Fix the arm64 build by adding arch specific header files to
CFLAGS and BPF_CFLAGS. Also, use `gcc -print-multiarch` to get
arch name instead of `uname -m` which is not the right way.

---
 configure      | 7 +++++--
 lib/defines.mk | 4 ++--
 2 files changed, 7 insertions(+), 4 deletions(-)

--- a/configure
+++ b/configure
@@ -65,6 +65,7 @@ check_toolchain()
     : ${M4=m4}
     : ${EMACS=emacs}
     : ${ARCH_INCLUDES=}
+    : ${ARCH_NAME=}
 
     CLANG=$(find_tool clang "$CLANG")
     LLC=$(find_tool llc "$LLC")
@@ -76,6 +77,8 @@ check_toolchain()
         fi;
     done
 
+    ARCH_NAME=$($CC -print-multiarch 2>/dev/null)
+
     clang_version=$($CLANG --version | grep -Po '(?<=clang version )[[:digit:]]+')
     if [ "$?" -ne "0" ]; then
         echo "*** ERROR: Couldn't execute '$CLANG --version'"
@@ -109,7 +112,7 @@ check_toolchain()
     if [ -z "$ARCH_INCLUDES" ]; then
         for dir in $(echo | $CC -Wp,-v -E - 2>&1 | grep '^ '); do
             local idir
-            idir="${dir}/$(uname -m)-linux-gnu/"
+            idir="${dir}/${ARCH_NAME}/"
             [ -d "$idir" ] && ARCH_INCLUDES="-I${idir} $ARCH_INCLUDES"
         done
     fi
--- a/lib/defines.mk
+++ b/lib/defines.mk
@@ -40,8 +40,8 @@ endif
 
 DEFINES += -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64
 
-CFLAGS += -std=gnu11 -Wextra -Werror $(DEFINES)
-BPF_CFLAGS += $(DEFINES) $(filter -ffile-prefix-map=%,$(CFLAGS))
+CFLAGS += -std=gnu11 -Wextra -Werror $(DEFINES) $(ARCH_INCLUDES)
+BPF_CFLAGS += $(DEFINES) $(filter -ffile-prefix-map=%,$(CFLAGS)) $(ARCH_INCLUDES)
 
 CONFIGMK := $(LIB_DIR)/../config.mk
 LIBMK := Makefile $(CONFIGMK) $(LIB_DIR)/defines.mk $(LIB_DIR)/common.mk $(LIB_DIR)/../version.mk
```
</details>

I modified one of the hunks of the patch to look like below:

```diff
@@ -76,6 +77,8 @@ check_toolchain()
         fi;
     done
 
+    ARCH_NAME=$($CC -print-multiarch 2>/dev/null)
+    echo "**** ARCH_NAME: $ARCH_NAME; CC: $CC ****"
     clang_version=$($CLANG --version | grep -Po '(?<=clang version )[[:digit:]]+')
     if [ "$?" -ne "0" ]; then
         echo "*** ERROR: Couldn't execute '$CLANG --version'"
```

and then I run the compilation of xdp-tools which printed:

```
**** ARCH_NAME: ; CC: mips-openwrt-linux-musl-gcc ****
```

`-print-multiarch` doesn't seem to generate an arch name when called on the mips cross-compiler gcc

okaestne, 2024-10-12:
> EDIT3:
> 
> If I choose the `MIPS Pistachio` board then ARCH is mipsel and CONFIG_BIG_ENDIAN is not defined. So, maybe a mix of the two variables is required when setting BPF_CFLAGS (or not, as CONFIG_BIG_ENDIAN not set means then same as arch = mipsel 🤔; but for other archs may be different, so yeah a mix I guess ).
> Pistachio board diffconfig

`target/Config.in` can explain this:

```kconfig
config mips
	select BIG_ENDIAN
	bool

config mipsel
	bool

config mips64
	select BIG_ENDIAN
	select ARCH_64BIT
	bool

config mips64el
	select ARCH_64BIT
	bool
```

So actually, we might need to check even four different cases (mips(64)?(el)?).

> On a side note, there's one of the patches that I don't think work as expected:
> 012-configure-fix-arch-includes.patch

> `-print-multiarch` doesn't seem to generate an arch name when called on the mips cross-compiler gcc

It's correct that it does not print any **multiarch** string for mips(el). Multiarch means that binaries of multiple archs are supported, e.g. amd64 systems that also support i386 binaries. That's why there might be `/usr/lib/x86_64-linux-gnu` and `/usr/lib/i386-linux-gnu`. No multiarch support - no `/usr/include/<multiarch>`, just `/usr/include`.

I'll try to test the other mips variants, when I find some time.

diegonc, 2024-10-13:
> Edit: As well as the main branch, as it seems to be broken even with the newer xdp-tools version where I cherry picked some of the patches from.

BTW, I managed to build the main branch by removing the patch [021-headers-xdp-drop-vlan_hdr-as-already-defined.patch](https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob_plain;f=package/network/utils/xdp-tools/patches/021-headers-xdp-drop-vlan_hdr-as-already-defined.patch;hb=HEAD) which was introduced by commit [5acc4f919c28ee8ea20c98856e1e824168cf2270](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=5acc4f919c28ee8ea20c98856e1e824168cf2270)

Curiously, in the main branch __MIPSEB__ and __MIPSEL__ are no longer required, I'm not sure why. Maybe, it has to do with BPF_CFLAGS including `-D__TARGET_ARCH_mips -mbig-endian`.

<details>
  <summary>diffconfig</summary>

  ```
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_8dev_carambola2=y
CONFIG_DEVEL=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_HAS_BPF_TOOLCHAIN=y
# CONFIG_KERNEL_BTRFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_CIFS_ACL is not set
# CONFIG_KERNEL_EXT4_FS_POSIX_ACL is not set
# CONFIG_KERNEL_EXT4_FS_SECURITY is not set
# CONFIG_KERNEL_F2FS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_F2FS_FS_SECURITY is not set
# CONFIG_KERNEL_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFSPLUS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_HFS_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_POSIX_ACL is not set
# CONFIG_KERNEL_JFFS2_FS_SECURITY is not set
# CONFIG_KERNEL_JFS_POSIX_ACL is not set
# CONFIG_KERNEL_NFS_ACL_SUPPORT is not set
# CONFIG_KERNEL_REISER_FS_POSIX_ACL is not set
# CONFIG_KERNEL_TMPFS_POSIX_ACL is not set
# CONFIG_KERNEL_UBIFS_FS_SECURITY is not set
# CONFIG_KERNEL_XFS_POSIX_ACL is not set
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libxdp=y
CONFIG_PACKAGE_xdp-filter=y
CONFIG_PACKAGE_xdp-loader=y
CONFIG_PACKAGE_xdpdump=y
CONFIG_PACKAGE_zlib=y
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
# CONFIG_USE_FS_ACL_ATTR is not set
CONFIG_USE_LLVM_BUILD=y
# CONFIG_WPA_MBO_SUPPORT is not set
```

knuxyl, 2024-10-20:
xpd-tools fails to compile on my arch install for the nanopi r5c (rockchip rk3568). I can't seem to disable rebuild all packages on every make so I can't make V=sc because it'll be compiling for hours until the error shows up. I've had this problem before like a year or 2 ago and I don't remember how I fixed it.

diegonc, 2024-10-20:
> xpd-tools fails to compile on my arch install for the nanopi r5c (rockchip rk3568). I can't seem to disable rebuild all packages on every make so I can't make V=sc because it'll be compiling for hours until the error shows up. I've had this problem before like a year or 2 ago and I don't remember how I fixed it. Edit : Just piped out to log file and found this
> 
> ```
> In file included from xdpfilt_dny_ip.c:10:
> In file included from ./xdpfilt_prog.h:21:
> ../lib/../headers/xdp/parsing_helpers.h:91:11: error: arithmetic on a pointer to an incomplete type 'struct vlan_hdr'
>    91 |                 if (vlh + 1 > data_end)
>       |                     ~~~ ^
> ../lib/../headers/xdp/parsing_helpers.h:71:9: note: forward declaration of 'struct vlan_hdr'
>    71 |         struct vlan_hdr *vlh;
>       |                ^
> ../lib/../headers/xdp/parsing_helpers.h:94:16: error: incomplete definition of type 'struct vlan_hdr'
>    94 |                 h_proto = vlh->h_vlan_encapsulated_proto;
>       |                           ~~~^
> ../lib/../headers/xdp/parsing_helpers.h:71:9: note: forward declaration of 'struct vlan_hdr'
> --
> make[5]: Leaving directory '/home/knuxyl/Mount/4tb1/system/source/openwrt/build_dir/target-aarch64_generic_musl/xdp-tools-1.4.3/xdp-filter'

This is the I got on main branch too. It can be fixed by removing patch 21, at least for the configuration posted on my last response.

okaestne, 2024-10-20:
Besides all this, can anyone tell me why this package is included in default configs, although it isn't installed by default?

joez, 2024-10-21:
Facing the same "vlan_hdr" issue, I can bypass this issue by disabling the `xdp-tools` entirely:

```
mv package/network/utils/xdp-tools/Makefile package/network/utils/xdp-tools/Makefile.bak
```

Or to modify the file `build_dir/target-x86_64_musl/xdp-tools-1.4.3/headers/xdp/parsing_helpers.h`, add following line:

```
#include <linux/if_vlan.h>
```

diegonc, 2024-10-21:
> #include <linux/if_vlan.h>

@joez this include in a previous version of your post looks fine when applied with the patch I removed

It defines the structure removed by the patch:

https://github.com/torvalds/linux/blob/master/include/linux/if_vlan.h#L35

@Ansuel, as you are the author of 5acc4f919c28ee8ea20c98856e1e824168cf2270, did you meant to include that header along with the removal of the struct (patch 021) or was it included in some other way given the configuration you tested on?

Ansuel, 2024-10-21:
@diegonc as the patch say, the struct was removed as it was already included in other files... probably new version reorganized header and the header that was including the struct doesn't anymore. But we need to be careful with including header as we might introduce inconsistencies with how xdp tools should work...

If it's enough, I would just drop patch 21 and reintroduce the header

diegonc, 2024-10-21:
> @diegonc as the patch say, the struct was removed as it was already included in other files... probably new version reorganized header and the header that was including the struct doesn't anymore. [...]

Yeah, there's a lot of other stuff in the header...

> If it's enough, I would just drop patch 21 and reintroduce the header

... so probably this is the way to go.

Thanks for having a look.

knuxyl, 2024-10-21:
I should mention I looked into what xdp was and realized xdp-tools isn't the package name, it's a bundle of like xdp-filter and others, so I searched menuconfig for those and disabled the 3 modules and the libxdp and the build went fine. At first I thought this was part of the toolchain. They aren't included by default in the kernel, it just builds as modules, so I see no harm in just disabling them. I am curious though as to why these are being built as modules for all targets and not being built in. This is the only place I have even heard of XDP, does it have enough widespread use to justify including building it in all targets as a module by default?
And great job everyone for finding the problem! You are much appreciated