openwrt/openwrt #12242: openssl  cross compile fails for mipsel ramips/mt7621

darwin787, 2023-03-22:
### Describe the bug

```
CC="mipsel-openwrt-linux-musl-gcc" /usr/bin/perl crypto/aes/asm/aes-mips.pl "o32" -Icrypto -I. -Iinclude -Iproviders/common/include -Iproviders/implementations/include -fPIC -pthread -mabi=32 -Wa,--noexecstack -Wall -O3 -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/data/ramips/sdk/build_dir/target-mipsel_24kc_musl/openssl-3.0.8=openssl-3.0.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -znow -zrelro -DOPENSSL_USE_NODELETE -DOPENSSL_PIC -DOPENSSLDIR="\"/etc/ssl\"" -DENGINESDIR="\"/usr/lib/engines-3\"" -DMODULESDIR="\"/usr/lib/ossl-modules\"" -DOPENSSL_BUILDING_OPENSSL -DNDEBUG -DOPENSSL_PREFER_CHACHA_OVER_GCM -DOPENSSL_SMALL_FOOTPRINT -I/data/ramips/sdk/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/usr/include -I/data/ramips/sdk/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include/fortify -I/data/ramips/sdk/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include -DAES_ASM -DOPENSSL_BN_ASM_MONT -DSHA1_ASM -DSHA256_ASM  crypto/aes/aes-mips.S
mipsel-openwrt-linux-musl-gcc  -Icrypto -I. -Iinclude -Iproviders/common/include -Iproviders/implementations/include  -DAES_ASM -DOPENSSL_BN_ASM_MONT -DSHA1_ASM -DSHA256_ASM -fPIC -pthread -mabi=32 -Wa,--noexecstack -Wall -O3 -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -mips16 -minterlink-mips16 -ffile-prefix-map=/data/ramips/sdk/build_dir/target-mipsel_24kc_musl/openssl-3.0.8=openssl-3.0.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -znow -zrelro -DOPENSSL_USE_NODELETE -DOPENSSL_PIC -DOPENSSLDIR="\"/etc/ssl\"" -DENGINESDIR="\"/usr/lib/engines-3\"" -DMODULESDIR="\"/usr/lib/ossl-modules\"" -DOPENSSL_BUILDING_OPENSSL -DNDEBUG -DOPENSSL_PREFER_CHACHA_OVER_GCM -DOPENSSL_SMALL_FOOTPRINT -I/data/ramips/sdk/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/usr/include -I/data/ramips/sdk/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include/fortify -I/data/ramips/sdk/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include -c -o crypto/aes/libcrypto-lib-aes-mips.o crypto/aes/aes-mips.S
crypto/aes/aes-mips.S: Assembler messages:
crypto/aes/aes-mips.S:13: Error: invalid operands `lw $12,0($6)'
crypto/aes/aes-mips.S:14: Error: invalid operands `lw $13,4($6)'
crypto/aes/aes-mips.S:15: Error: invalid operands `lw $14,8($6)'
crypto/aes/aes-mips.S:16: Error: invalid operands `lw $15,12($6)'
crypto/aes/aes-mips.S:17: Error: invalid operands `lw $30,240($6)'
crypto/aes/aes-mips.S:20: Error: invalid operands `xor $8,$12'
crypto/aes/aes-mips.S:21: Error: invalid operands `xor $9,$13'
crypto/aes/aes-mips.S:22: Error: invalid operands `xor $10,$14'
crypto/aes/aes-mips.S:23: Error: invalid operands `xor $11,$15'
crypto/aes/aes-mips.S:25: Error: invalid operands `subu $30,1'
crypto/aes/aes-mips.S:114: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ext $12,$9,8,8'
crypto/aes/aes-mips.S:116: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ext $13,$10,8,8'
crypto/aes/aes-mips.S:117: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ext $14,$11,8,8'
crypto/aes/aes-mips.S:118: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ext $15,$8,8,8'
crypto/aes/aes-mips.S:119: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ins $1,$12,2,8'
crypto/aes/aes-mips.S:120: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ins $2,$13,2,8'
crypto/aes/aes-mips.S:121: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ins $24,$14,2,8'
crypto/aes/aes-mips.S:122: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ins $25,$15,2,8'

omit similar errors

make[4]: *** [Makefile:10063: crypto/aes/libcrypto-lib-aes-mips.o] Error 1
make[4]: Leaving directory '/data/ramips/sdk/build_dir/target-mipsel_24kc_musl/openssl-3.0.8'
make[3]: *** [Makefile:1499: build_sw] Error 2
make[3]: Leaving directory '/data/ramips/sdk/build_dir/target-mipsel_24kc_musl/openssl-3.0.8'
make[2]: *** [Makefile:400: /data/ramips/sdk/build_dir/target-mipsel_24kc_musl/openssl-3.0.8/.built] Error 2
make[2]: Leaving directory '/data/ramips/sdk/feeds/base/package/libs/openssl'
time: package/feeds/base/openssl/compile#5.41#0.50#5.83
    ERROR: package/feeds/base/openssl failed to build.
make[1]: *** [package/Makefile:116: package/feeds/base/openssl/compile] Error 1
make[1]: Leaving directory '/data/ramips/sdk'
make: *** [/data/ramips/sdk/include/toplevel.mk:224: package/openssl/compile] Error 2

```

### OpenWrt version

snapshot

### OpenWrt target/subtarget

ramips/mt7621

### Device

sdk for x64

### Image kind

Self-built image

### Steps to reproduce

cross compile openssl package

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Spudz76, 2023-03-22:
Works here, different specific mt7621 based device (Linksys EA7300v1), compiling on Debian Buster x86_64, cross compiling to mipsel (should all be the same there)

Maybe the SDK toolchain is broken, I build everything.  I haven't rebuilt any toolchain components on purpose from clean in a while but it usually rebuilds whatever needs rebuilt as I pull new changes.  It seems weird to me the local-built vs SDK bot build would end up running different.

darwin787, 2023-03-22:
Maybe your gcc or openssl is not latest.
No issue with gcc 11 and openssl 1.1.1 .

ynezz, 2023-03-22:
I've noticed it as well https://lists.infradead.org/pipermail/openwrt-devel/2023-March/040729.html

ynezz, 2023-03-22:
> The SDK toolchain is too old.

@dhewg thanks for checking, but honestly, I quite don't get this :-)

There is a weekly scheduled build job in [ustream-ssl](https://gitlab.com/openwrt/project/ustream-ssl/-/pipelines) which is supposed to be using latest snapshot SDKs. It was working fine for ages, failed 13 hours ago for both `ath79/generic` and `malta/be` MIPS targets. 

So SDK/toolchain from a week ago (Mar 15, 2023) was [fine](https://gitlab.com/openwrt/project/ustream-ssl/-/pipelines/806600970), there is no `mips16` in there either. So I'm wondering what has changed in the SDK/toolchain area, that suddenly there is a `mips16` being used today?

ynezz, 2023-03-22:
> The SDK toolchain is too old.

BTW inspecting that failed build job for [ath79/generic](https://gitlab.com/openwrt/project/ustream-ssl/-/jobs/3977611619) I can see kernel 5.15.102 and that kernel was commited in e467856cb34, Sat Mar 18, 4 days ago.

dhewg, 2023-03-22:
You need at least a SDK of r22351-5c545bdb36 or newer to use it against a checkout of r22351-5c545bdb36 or newer.

I assume this error is because combining a checkout of >= r22351-5c545bdb36 with a SDK < r22351-5c545bdb36. So it is fixable with either an older checkout or a newer SDK...

ASU has a very related issue because it too uses the SDK. There you can get an image with older components of the SDK but newer packages. Which is impossible to get when rolling everything on your own.

It's absolutely confusing and next to impossible to put this into words so everyone in the community understands and this.

ynezz, 2023-03-22:
> You need at least a SDK of r22351-5c545bdb36 or newer if you're using a checkout of r22351-5c545bdb36 or newer.

Ok, that makes sense to me, thanks for your patience. So closing this as fixed via 5c545bdb36.

closed 2023-03-22 by ynezz

ynezz, 2023-03-22:
> ASU has a very related issue because it too uses the SDK.

It uses ImageBuilder, so consuming prebuilt binaries and this is fast moving target...

> Which is impossible to get when rolling everything on your own.

I'm not sure if that would work, but maybe pinning the feeds in the `feeds.conf.default` in the snapshot SDKs would create a working combination.