openwrt/openwrt #12244: USBIP - usbip binary usermode application needs to be synced with current kernel

anonymous-one, 2023-03-22:
### Describe the bug

/tools/usb/usbip

### OpenWrt version

r22276-8dea8bde2a

### OpenWrt target/subtarget

ipq807x/generic

### Device

QNAP 301w

### Image kind

Official downloaded image

### Steps to reproduce

The binary `usbip` command needs to be updated to match the current kernel.

It is located in package `usbip-client`.

The source for this command is located in all kernel source archives under `/tools/usb/usbip`.

Since this has not been updated in quite some time on the OpenWRT side, it no longer works because of the various changes in the usb vhci driver as well as possibly some changes under /sys.

I unfortunately cannot give you much beyond the following as I have no need for the usbip command on my OpenWRT side (I use it as a server, and usbipd works find), but here is an example of one error:

```
root@OPENWRT-PRIMARY:~# lsmod | grep vhci ; usbip port
usbip_core             16384  2 vhci_hcd,usbip_host
vhci_hcd               28672  0
libusbip: error: no available usb controllers
usbip: error: open vhci_driver
usbip: error: list imported devices
root@OPENWRT-PRIMARY:~#
```

As you can see even though vhci_hcd is loaded, usbip complains.

I had the same issue on my debian mini pc after I manually built a recent kernel. I manually updated my usbip command and it was resolved right away.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dob46, 2023-04-07:
Workaround :
https://github.com/openwrt/openwrt/issues/8102

dob46, 2023-04-22:
Repaired in #12350

zhangsd20, 2023-08-11:
I had the same issue on OpenWRT 23.05, the usbip server worked fine, 
but the client didn't work when I usbip attach got the same error

libusbip: error: no available usb controllers
usbip: error: open vhci_driver
usbip: error: list imported devices

dob46, 2023-08-11:
@zhangsd20 You must compile newer version of libudev-zero (1.0.2 or 1.0.3)