openwrt/openwrt #12249: ipv6lan口分配bug

yk00365, 2023-03-23:
### Describe the bug

1.在一些Android5.0版本上 连接ipv6之后静置一段时间，再去查看是否访问的时候，ipv6掉线，只有该设备等待1分钟之后才能正常访问ipv6
2.在电脑有线或无线连接ipv6之后，等上一天两天之后由于旧的ipv6地址没有消除，同时又来了新的ipv6地址，无法访问ipv6
以上2只有断开重连才能恢复，1需要静置一分钟。

### OpenWrt version

22.10

### OpenWrt target/subtarget

no

### Device

lan口

### Image kind

Official downloaded image

### Steps to reproduce

1.在一些Android5.0版本上 连接ipv6之后静置一段时间，再去查看是否访问的时候，ipv6掉线，只有该设备等待1分钟之后才能正常访问ipv6
2.在电脑有线或无线连接ipv6之后，等上一天两天之后由于旧的ipv6地址没有消除，同时又来了新的ipv6地址，无法访问ipv6
以上2只有断开重连才能恢复，1需要静置一分钟。

### Expected behaviour

希望能够改进lan口ipv6分配问题，中国大陆宽带家宽ipv6是动态的

### Additional info

no

### Diffconfig

```text
no
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

delgh1, 2023-03-30:
Since the OP did not attach an English translation of the post, and it seems that most repo contributors here do not speak Chinese, I will translate it so that most people here can understand. Words in hard brackets ("[]") are added by me to make the sentence fluent and readable.

### Describe the bug
1. On some Android 5.0 versions, after connecting to the ipv6 [network], leave it [the DHCP client] for a while, then check if the internet [is accessible], ipv6 would be offline, only after waiting for [about] one minute, the device can normally access [the] ipv6 [network].
2. After a computer connecting to ipv6 [network] using wired or wireless connection, after waiting for one day or two, because the old ipv6 address does not disappear, while a new ipv6 address comes [is assigned], the ipv6 [network] becomes inaccessible.
Only disconnect and reconnect can solve issue No.2. Waiting for a minute is needed to solve No.1

### OpenWrt version
22.10

Comment: _**不符合要求，应该用 `. /etc/openwrt_release && echo $DISTRIB_REVISION` 表示的版本 
This does not meet the requirement of a bug report, use the value in `. /etc/openwrt_release && echo $DISTRIB_REVISION`**_

### OpenWrt target/subtarget
no

Comment: _**什么叫做no? 使用这个命令 `. /etc/openwrt_release && echo $DISTRIB_TARGET` 
What do you mean no? use this command `. /etc/openwrt_release && echo $DISTRIB_TARGET`**_

### Device
lan port

Comment: _**用这个命令 `cat /tmp/sysinfo/model` 获取路由器的型号
Use this command to get the model of your router `cat /tmp/sysinfo/model`**_

### Image kind
Official downloaded image

### Steps to reproduce
The same as `Describe the bug` section, and I don't think this is appropriate for a bug report

### Actual behaviour
No response

### Expected behaviour
I hope LAN port ipv6 assignment can be improved, [because] in China home broadband's ipv6 [network] is dynamic.

Comment: This is a direct and literal translation from Chinese. I think this person means _DHCPv6 should be improved(?), they receive IPv6 addresses via DHCPv6(-PD?) at a home network environment_

## Really my comments

- 请尽量在确定是bug的情况下再提交bug report，因为这个repo有近1800多issues没有解决，可以去openwrt论坛或者reddit上讨论。
Please, only submit a bug report after you are absolutely sure it is a bug, because there're 1.8k open issues in this repo. Try to get some answers from openwrt forum or reddit.

- 目前看起来可能有设置上的问题。请问有无阅读[这个文档](https://openwrt.org/docs/guide-user/base-system/dhcp)？ 可以把 `/etc/config/network` 及 `/etc/config/dhcp` 的内容贴上来。
For now, it looks like a misconfiguration. Did you read [this doc](https://openwrt.org/docs/guide-user/base-system/dhcp) ? Also, please post your `/etc/config/network` and `/etc/config/dhcp` here.

- 你的ISP使用的是DHCPv6-PD分配地址吗？是否采用IPv4 over IPv6？你如何判断设备不能访问IPv6网("掉线"), 用ping？在"掉线"的时候IPv4网状况如何？同一个网卡上有多个IPv6地址并不影响访问IPv6网，除非ISP更换了/64 prefix。你的电脑是否设置成使用SLAAC?
Does your ISP use DHCPv6-PD to assign addresses? Is IPv4 over IPv6 also used? How do you know if your device cannot access the IPv6 network ("offline"), e.g. using ping? When it is "offline", can it access the IPv4 network? It is OK to have multiple IPv6 addresses on a single interface unless the ISP changed your /64 prefix. Is your PC configured to use SLAAC?

yk00365, 2023-03-30:
您好，来信收悉。
1.这貌似是一个bug，因为我的vivo手机是安卓5.0版本的，在路由器是官方固件且开启ipv6的时候，不会出现掉线的情况，但是自从换openwrt的时候，经常会在该手机连接op设置的WiFi时，刚开始连接，ipv6正常，等过几个小时或者半天之后，再去验证，已经是只能v4不能v6了，静置这个手机就会复现。
2.这俩文件均为openwrt默认设置
3.是dhcp-pd，不是ipv4 over ipv6
一个是ping6返回无法ping6 在一个就是网站验证，例如ipw.cn
发生v6掉线后，只有v4正常，此时只有v4能够访问，v6不可
电脑默认支持。

yk00365, 2023-03-30:
> Since the OP did not attach an English translation of the post, and it seems that most repo contributors here do not speak Chinese, I will translate it so that most people here can understand. Words [...]

您好，来信收悉。
1.这貌似是一个bug，因为我的vivo手机是安卓5.0版本的，在路由器是官方固件且开启ipv6的时候，不会出现掉线的情况，但是自从换openwrt的时候，经常会在该手机连接op设置的WiFi时，刚开始连接，ipv6正常，等过几个小时或者半天之后，再去验证，已经是只能v4不能v6了，静置这个手机就会复现。
2.这俩文件均为openwrt默认设置
3.是dhcp-pd，不是ipv4 over ipv6
一个是ping6返回无法ping6 在一个就是网站验证，例如ipw.cn
发生v6掉线后，只有v4正常，此时只有v4能够访问，v6不可
电脑默认支持。

findlayfeng, 2023-03-31:
之前在家里用 ipv6 的时候也发现了这个问题，观察过一段时间后，这个问题的主要原因是dhcp地址续订时间和isp重置你的链接的时间不匹配造成的。
isp重置链接后，wan口因为感知到接口断开再链接，然后出发重新 dhcp，获取到新地址
内网设备则不会发生接口断开，再链接的过程，设备的ipv6在过期前都不会发生变化，但这个地址不能连到网络，因为上游邻居已经消失
其实ipv4也发生了相似的事情，但是因为一般在路由器会经过nat，内网不重新获取地址也不影响
但是如果你修改了lan口的网络配置，如果你是直连路由器的客户端，因为接口重连，会触发新的dhcp过程，但不是直接链接到路由器的客户端（比如通过交换机）也会有相似的症状，客户端会保留旧的不可用ipv4，直到它过期

据我所知，dhcp报文中没有定义服务器使客户端已经获取到的租约失效的办法，所以从地址重分配的角度是解决不了这个问题的。

我最后总结出来，缓解这个问题两个办法，注意只能缓解

1. 使用单一网络，因为链接重置后，ipv4可用，ipv6不可用，特别是分到的ipv6是公网地址，对很多应用会产生误判，导致不可用，影响使用，但又不能判断断网，改成单一网络后，ipv6一旦不可用，就能判断断网，手动或自动的就会重新尝试获取地址。
2.改小dhcp的租约时长，缓解问题，但会造成一定的网络负担，而且因为ip续期的不及时（比如wifi信号差）也会影响使用体验

yk00365, 2023-03-31:
> 之前在家里用 ipv6 的时候也发现了这个问题，观察过一段时间后，这个问题的主要原因是dhcp地址续订时间和isp重置你的链接的时间不匹配造成的。 isp重置链接后，wan口因为感知到接口断开再链接，然后出发重新 dhcp，获取到新地址 内网设备则不会发生接口断开，再链接的过程，设备的ipv6在过期前都不会发生变化，但这个地址不能连到网络，因为上游邻居已经消失 其实ipv4也发生了相似的事情，但是因为一般在路由器会经过nat，内网不重新获取地址也不影响 在ipv4也复现这个问题的方法是，修改lan口的网络配置，直连路由器的客户端，会因为接口重置，触发新的dhcp过程，不直接链接到路由器的客户端（比如通过交换机）就会出现有ipv4地址，但不可用，直到它过期
> 
> 据我所知，dhcp报文中没有定义服务器使客户端已经获取到的租约失效的办法，所以从路由器主动地址重分配的角度是解决不了这个问题的。 路由器同步的重置一下lan接口也只能影响直接链接到路由器的设备（除非你所有的交换机都是管理型的，支持snmp）。
> 
> 我最后总结出来，缓解这个问题几个办法，注意只能缓解
> 
> 1. 使用单一网络，因为链接重置后，ipv4可用，ipv6不可用，特别是分到的ipv6是公网地址，对很多应用会产生误判，导致不可用，影响使用，但又不能判断断网，改成单一网络后，ipv6一旦不可用，就能判断断网，手动或自动的就会重新尝试获取地址。
> 2. 改小dhcp的租约时长，缓解问题，但会造成一定的网络负担，而且因为ip续期的不及时（比如wifi信号差）也会影响使用体验
> 3. 配置ipv6nat，内网使用本地ipv6，没试过，感觉在使用体验上这个会是最好的，但内网ip的话为什么不直接ipv4呢。

但是华硕官方固件和老毛子就没事，我现在改动ipv6的分配方式了，再试试，毕竟我还是想使用ipv6网络，主要是个别设备不完美，等着官方答复看看

Orchid-Lee, 2024-02-06:
> 之前在家里用 ipv6 的时候也发现了这个问题，观察过一段时间后，这个问题的主要原因是dhcp地址续订时间和isp重置你的链接的时间不匹配造成的。 isp重置链接后，wan口因为感知到接口断开再链接，然后出发重新 dhcp，获取到新地址 内网设备则不会发生接口断开，再链接的过程，设备的ipv6在过期前都不会发生变化，但这个地址不能连到网络，因为上游邻居已经消失 [...]

我也遇到了相同的问题，明明ipv6公网地址已经改变了，openwrt的lan口显示的ipv6地址还是旧的；和你的描述相符，我限制了内网设备获取ipv6地址，并用socat监听lan口ipv6，映射到内网ipv4端口，出现了这个问题，手动点击openwrt的连接才能获取到正确的ipv6地址。
[image]

findlayfeng, 2024-04-12:
> 之前在家里用 ipv6 的时候也发现了这个问题，观察过一段时间后，这个问题的主要原因是dhcp地址续订时间和isp重置你的链接的时间不匹配造成的。
> isp重置链接后，wan口因为感知到接口断开再链接，然后出发重新 dhcp，获取到新地址
> 内网设备则不会发生接口断开，再链接的过程，设备的ipv6在过期前都不会发生变化，但这个地址不能连到网络，因为上游邻居已经消失 [...]

一年后来更正一个描述错误，dhcp服务中存在服务端重置客户端地址的办法，但是openwrt 中没有实现

Orchid-Lee, 2024-04-16:
谢谢大佬，大概了解了， 我目前只让openwrt lan口获取了ipv6地址， 内网通过ipv6转ipv4实现访问， 自从配置了域名ddns后， 没有发现有这种情况了。

ptpt52, 2024-04-20:
我遇到的问题是 Android手机 收到NS包（可能） 不应答回复NA包
路由器发出NS请求包，没有收到手机的NA包
这样android手机发出IPv6包，但是路由器没有办法回复给android手机，这样我的android手机不通。

ptpt52, 2024-04-20:
由于某种原因，路由器发送了NS包，Android不应答NA包，因此路由器不知道Android的MAC地址
如果Android主动发送PING包ping 路由器lan口的IPv6地址，可以ping通，同时让路由器学习到了Android的MAC地址了，这个时候IPv6就通了可以访问外网IPv6地址了。

ptpt52, 2024-04-20:
The issue I encountered is that my Android phone receives NS packets (possibly) but does not reply with NA packets. The router sends out NS request packets, but does not receive NA packets from the phone. Consequently, the Android phone sends out IPv6 packets, but the router cannot reply to the Android phone, resulting in my Android phone being unable to connect.

Due to some reason, the router sends NS packets, but Android does not respond with NA packets, hence the router doesn't know Android's MAC address. If Android proactively sends PING packets to ping the router's LAN interface IPv6 address, it can establish connectivity, allowing the router to learn Android's MAC address. At this point, IPv6 connectivity is established, enabling access to external IPv6 addresses.

138161, 2024-04-24:
wifi掉线，使用wpa3加密试试
有线和无线都掉线，
# 放行入站(wan,lan,docker，单向进入), icmpv6续约
ip6tables -I input_rule           -p udp   --sport 546 --dport 547  -j ACCEPT
ip6tables -I input_rule   -p ipv6-icmp  -m limit --limit 100/s --limit-burst 10  -j ACCEPT

# 解决DHCPv6续约的icmpv6通讯，lan侧到路由器本身。此消息经过中继后再传到上级光猫。
#  过程：Solicit---Advertise---Request---Reply
#  PC端口546，发送请求Solicit至，路由器本身[ff02::1:2]:547端口
#  路由器本身546端口，中继到，光猫547端口
#  光猫547端口，回应一个Advertise给，路由器546端口
#  路由器本身547端口，中继到，PC端口546
#  PC端口546，回应一个Request给路由器547端口
#  路由器本身546端口，中继到，光猫547端口
#  光猫547端口，回应一个Reply给，路由器546端口结束。 
# ip6tables -I zone_lan_input -p udp  --sport 546 --dport 547  -j ACCEPT