openwrt/openwrt #12253: daemon.err hostapd: 160 MHz channel width is not supported in 5 / 6GHz

starraiderx, 2023-03-23:
### Describe the bug

Setting htmode to HE160 doesn't work and produces the following syslog error: "daemon.err hostapd: 160 MHz channel width is not supported in 5 / 6GHz." The router used to support HE160 on previous Snapshot Builds.

### OpenWrt version

r22371-0eebc6f0dd

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

Flash image from https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

abajk, 2023-03-23:
I think the problem is specific to the Qualcomm radios. Yesterday I have been building a snapshot image on Netgear WAX206 (Mediatek Soc) and HE160 works fine.

tmtom, 2023-03-25:
I can confirm the same, on my Xiaomi AX3600, on build r22297-1781e7408a 160MHz channel worked and on r22371-0eebc6f0dd does not work any more.

tmtom, 2023-03-28:
> 

Yes, seems to be an ath11k issue, perhaps 789a0bac35359a3958da0e555231c5bf28abf7c4 ?

Soromeister, 2023-04-01:
Confirmed fixed in build `r22475-936df715de`

Djfe, 2023-07-14:
@starraiderx pls close

closed 2023-07-14 by starraiderx