openwrt/openwrt #12254: With dualstack (IPv4 and IPv6 relay mode) Google Chromecast with Google TVs constantly disconnecting/reconnecting

JiiPee74, 2023-03-24:
### Describe the bug

There is some strange issue with OpenWRT when dualstack is in use. Google Chromecast with Google TV 4K cannot connect to OpenWRT via wifi. Or it can, but it wont stay connected. It will drop connection right after it gets IP's, Usually it gets IPv6 after IPv4 and then it disconnect, but sometimes IPv6 is assigned 1st.

I have to do lot of work to get it connected and stay connected. I have found that sometimes it stay connected when I 1st connect chromecast to another wifi network (example my phone hotspot) and then back to OpenWRT. But it's not working always. Sometimes I have to restart devices to get it working. Only few times I have been able to get chromecast to connect streight away without any network jumping.
When it gets connected, it will stay connected just fine, until one of the devices gets restarted.

Some talk and findings about issue here, thats why I wanted to make bug report because I now know it's not just my router hardware and OpenWRT build causing this:
https://www.reddit.com/r/openwrt/comments/11vtp0j/google_chromecast_with_google_tvs_constantly/

I have this issue and I did some more testing. It seems to be only happening when you are doing relay mode for IPv6. If you do server mode IPv6 (what in my understanding is IPv6 nat) then chromecast connect just fine. Or if you disable IPv6 then connecting via wifi works also fine.

My OpenWRT is self build for unsupported device, but it should not matter because others are reporting exactly same behavior.

Issue started when Chromecast was upgraded to Android 12. But it's not Android 12 issue because I have also tablet connecting to same wifi and it's based on Android 12.

Also my phone does dualstack hotspot where chromecast connect just fine so this really is only issue with OpenWRT and Chromecast with Google TV.

### OpenWrt version

r19160+167-7ea2f3d6e2

### OpenWrt target/subtarget

ramips/mt7620

### Device

Telewell TW-4G (LTE)

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-24:
Your router is NOT SUPPORTED by OpenWRT.
Check support progress here: https://github.com/openwrt/openwrt/pull/9834

JiiPee74, 2023-03-24:
> Your router is NOT SUPPORTED by OpenWRT. Check support progress here: #9834

Did you even bother to read my report?

I think I said my router is NOT supported, but I made report anyway because other people with DIFFERENT routers are facing this EXACTLY same issue, so it's not related to my unsupported router.

And I am well aware of my router support progress because I have made that PR.

brada4, 2023-03-24:
So you are running snapshot of your own produce. Cool. 

Provide separate distinctly named AP-s on 2.4 and 5GHz bands, and compare`logread`  between stable and unstable conection. It could be regulatory domain issue that one end says needing weaker signal in this area and other device moderating TX rate. Android derives regulatory domain from region setting (SIM card in case of phone)

brada4, 2023-03-24:
What I was more into - the disconnection reason from logread. i.e. how it looks there.
Trace options in android developer mode log in /data/misc - maybe thay are smarter.

My wired Android TV acquires IPv6 via router advertisment leaving no trail in any log. Just tried wifi and same - DHCPv4 lease, and no log for v6.

brada4, 2023-03-24:
IPv6 relay mode for autoconfig relays requests to upstream. It is not NAT.

JiiPee74, 2023-03-24:
> My wired Android TV acquires IPv6 via router advertisment leaving no trail in any log. Just tried wifi and same - DHCPv4 lease, and no log for v6.

Wired afaik works fine, however I cannot test it, I have no wired option in my dongle, I would need to buy USB hub with ethernet connection or buy google offered charger what has 100Mbps ethernet port.

> IPv6 relay mode for autoconfig relays requests to upstream. It is not NAT.

And I think I have not said anywhere that relay mode is nat, I have said that server mode is nat.

brada4, 2023-03-25:
It is not a problem with wired or wireless connection. Your provider does not support issuing multiple IPv6 addresses for your delegated subnet in relay more, you need to use your own RADV and DHCPv6 servers. No NAT involved.

JiiPee74, 2023-03-25:
> It is not a problem with wired or wireless connection. Your provider does not support issuing multiple IPv6 addresses for your delegated subnet in relay more, you need to use your own RADV and DHCPv6 servers. No NAT involved.

Eh.. My ISP provides /64 prefix and it works fine on all other devices in my network.
- Linux box
- Windows 11
- Tablet
- Phone

All gets IPv6 and IPv4 address just fine through OpenWRT, IPv6 with relay mode and IPv4 with DHCP. Only this chromecast with google tv 4k dongle have issues with it AND only when using with OpenWRT.
As I said, when I create hotspot with my phone what also offers dualstack, no issues.

And if it was what you claim, then how do you explain that it CAN connect through OpenWRT when I do lot of network jumping and when it gets connected, it will stay connected until one of the devices is rebooted?

brada4, 2023-03-25:
Not finding anything in OpenWRT log - you need to check what Chromecast writes in wifi trace and why it thinks it disconnects.

JiiPee74, 2023-03-25:
Well I dont have knowhow for that. Haven't played with android stuff like that at all.

brada4, 2023-03-25:
The problem is disconnect )?perceived in Android side?) without any log trail in OpenWRT

First try hardest in OpenWRT side:
rising wireless log level https://openwrt.org/docs/guide-user/network/wifi/basic one step at the time, might need to rise
setting private virtual AP for chromecast and looking at traffic dump
(we still dont know if it is wifi or ip addressing issue that makes Android basically switch off radio while protocols are running)

Outline of Android steps:
Find a file manager in play store that does not show lots of ads, grant it full storage access (default one allows only standard folders)

Tap 10x on build number and "developer options" menu appears.
In that menu enable trace log for wifi, supplicant or so.

Mine writes /data/misc/wifi/wpa_supplicant.log, i have to share it via gmail to myself, particular file manager is unable to follow file changes.

brada4, 2023-03-27:
You need logs from chromecasts since it drops radio before anything to blame is logged in OpenWRT.

brada4, 2023-03-28:
Your problem is not related to OpenWRT.
https://www.googlenestcommunity.com/t5/Chromecast/Chromecast-with-Google-TV-4K-Wifi-keeps-reconnecting-after-update-to/m-p/261857

rkuhl84, 2023-03-28:
> Your problem is not related to OpenWRT. https://www.googlenestcommunity.com/t5/Chromecast/Chromecast-with-Google-TV-4K-Wifi-keeps-reconnecting-after-update-to/m-p/261857

That person is using OpenWRT as well so not sure how that proves the problem isn't related to OpenWRT?

I did not have any of these issues until I switched to OpenWRT a couple weeks ago.

brada4, 2023-03-29:
The change was chromecast update Android 10->12.
Android wifi trace should tell what goes wrong, like some unrecognized connection parameter or so.

JiiPee74, 2023-04-06:
I got some very old wifi box what I have connected to my main openwrt router LAN port. With that box between handling wifi traffics, I have no issues with chromast getting both IP's.

brada4, 2023-04-06:
Obviously you need to approach google support and help them to re-create connectivity situation after Android upgrade.
You can try to get that trace log and try to guess - say there is info block in beacon that may contain one attribute less or more than device expects, one bitrate or antenna setting less or more etc.

There is nothing anti-android12 in OpenWRT.

JiiPee74, 2023-10-13:
Adding to this that I bought new phone few months ago. Sony Xperia 10 V what came with android 13. All was working fine until few days ago phone got small system upgrade and now situation is same as with Android TV 12. Phone keeps dropping wifi connection.

So now I have only tablet left what can use wifi and tablet is Xiaomi Pad 5 with Android 13.

brada4, 2023-10-14:
Liaise with relay upstream wrt options they set to break your androids. android works with dnsmasq splicing delegation just fine.

JiiPee74, 2023-10-18:
Adding more info.

I bought new router, Linksys WRT1900ACS in hope that it was my very old router. I have latest official OpenWRT installed in that box and problems continue.
Phone wont stay on wifi if I have IPv6 relay enabled. If it is disabled or if I use server mode (ipv6 nat), then phone will happily stay on wifi.

brada4, 2023-10-18:
Get trace log from your android and packet fump ysing openwrt and decide how to make peace between android and your provider.

JiiPee74, 2023-10-18:
> Get trace log from your android and packet fump ysing openwrt and decide how to make peace between android and your provider.

It's not provider issue. If I use phone as hotspot, dual stack works fine with chromecast with android tv. ISP is same in both cases. Interweb comes to my OpenWRT box via Zte MC7010 5G modem.

I do not know how to get trace log or packet dump.

brada4, 2023-10-18:
Disable ip6 autoconf relays as it was by default and let openwrt hand you out ip6 addresses. there is no sanitisation done on relayed packets and nonody is clairvoyant about packets your upstream uses to crash your tv

JiiPee74, 2023-10-18:
> Disable ip6 autoconf relays as it was by default and let openwrt hand you out ip6 addresses. there is no sanitisation done on relayed packets and nonody is clairvoyant about packets your upstream uses to crash your tv

OpenWRT default IPv6 settings doesn't work. It spit this warning:
`daemon.warn odhcpd[1690]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!`

Clients gets IP from ULA-prefix and IPv6 traffic wont go outside lan. And ULA-prefix addresses is what I really dont want. I get /64 public IPv6 from my operator what I want to share to my devices.

Where I have said anything about my TV is crashing? You are now making up fairytales.

brada4, 2023-10-19:
? tcpdump
? wifi trace during tv's normal operation ... i mean reconnect loop

mohd-akram, 2025-12-27:
What's the update on this? Is there a workaround without disabling IPv6?

mohd-akram, 2025-12-28:
Specifying DNS addresses in "Announced IPv6 DNS servers" in the lan interface seems to resolve the issue.

rscarvalho90, 2025-12-31:
> Specifying DNS addresses in "Announced IPv6 DNS servers" in the lan interface seems to resolve the issue.

Thank you! This solved my problem.
I've set the RA-Service to hybrid and put a fake IPV6 DNS Server. Apparently it didn't break my network.