openwrt/openwrt #12257: Failed to resolve specific DNS request with OpenWrt 22.03

amitguptakiitian, 2023-03-24:
### Describe the bug

Hi,

I am using Asus rt-ax53u router flashed with 22.03.3 OpenWRT firmware. I have enabled stubby for DOT.
dnsmasq is failed to resolve the DNS URL if the DNS response is more than 512 bytes. 
nslookup output for the failed response is as below.
[image]
I tried capturing the lo interface network packets and observed that **message truncated bit is set** in the DNS response. 
Ideally, either edns0 option or dns request must be be sent over tcp to handle such scenario but it is not happening. 
Below is the DNS response snapshot were message truncated bit is set.
[image]

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

1. install Stubby and configure it in strict mode with DOT servers
2. nslookup 512.size.dns.netmeister.org

### Actual behaviour

[image]

### Expected behaviour

#nslookup 512.size.dns.netmeister.org
[image]

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-24:
Because upstream DNS server is out of order.
You can reduce EDNS0 UDP packet size in dnsmasq configuration to something it expects, or use public DNS server like quad9 as "forwarder"

amitguptakiitian, 2023-03-27:
> DNS serv

there is no issue with stubby. seems the issue is with dnsmasq, it forwards the DNS request to stubby at **lo** interface, stubby resolves the DNS request with configured upstream server but while responding the response to dnsmasq, I see the message is truncated. 
EDNS0 packet size is configured at 1232.

when I disable the stubby and configured the dnsmasq to use any DNS server(quad 9, google etc.), it works, but not with stubby.  
Thanks.

amitguptakiitian, 2023-03-27:
I can see the same issue with the Linksys 1900ACS V2 model as well with openwrt version r19803-9a599fee93.

brada4, 2023-03-27:
Default configuration file is in strict mode against cloudflare and it perfectly works over localhost and in the dhcp network.
What did you change in stubby configuration?

amitguptakiitian, 2023-03-27:
I set the **localuse** to 1. 
below is the configuration of my dnsmasq.
[image]

amitguptakiitian, 2023-03-27:
I checked with the lower version 19.07 of openwrt and it works. Below is the response captured at lo interface.
[image]

the below response I received with OpenWrt 22.03.
[image]

brada4, 2023-03-27:
Your setup is identical to mine, i get 900-950 byte responses via localhost and no problem.
ath79 is sort of same mips32 by large.
```
root@OpenWrt:~# opkg list-installed | grep -e stubby -e dnsmasq
dnsmasq - 2.86-15
stubby - 0.4.3-1
```

amitguptakiitian, 2023-03-28:
Yes, I also have the same version of dnsmasq and stubby.
    _root@OpenWrt:~# opkg list-installed | grep -e stubby -e dnsmasq
    dnsmasq - 2.86-15
    stubby - 0.4.3-1_
is **Localuse** flag of dnsmaq set in your case as well? when you do nslookup, is it going through 127.0.0.1#53 or <ISP provided DNS server>#53?

brada4, 2023-03-28:
Mine is as default as possible, nothing consciously changed for dnsmasq and stubby as they worked right away, localuse is 1 by default.
But I trust you have a glitch that is not repeatable in my identically default setup.