openwrt/openwrt #12259: [ASUS RT-AC58U] radio1 goes off (regression)

lesf0, 2023-03-25:
### Describe the bug

After working for several hours radio1 (5ghz wifi) goes off and does not come back until the manual restart of the interface.
[image]

[Kernel log](https://pastebin.com/aCSHxjA1)

I've tried several different versions, r22307-4bfbecbd9a does not work either ( [Kernel log](https://pastebin.com/gGFfJVr9) ), but r19590-042d558536 works fine. I can bisect it further if it helps finding the issue.

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ASUS RT-AC58U

### Image kind

Self-built image

### Diffconfig

```text
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_generic=y
CONFIG_TARGET_ipq40xx_generic_DEVICE_asus_rt-ac58u=y
CONFIG_DEVEL=y
CONFIG_BINARY_FOLDER="/home/lesf0/Dev/openwrt-build-scripts/bin/rt-ac58u"
CONFIG_BINUTILS_VERSION="2.37"
CONFIG_BINUTILS_VERSION_2_37=y
CONFIG_BPF_TOOLCHAIN_HOST=y
CONFIG_BPF_TOOLCHAIN_HOST_PATH=""
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_DOWNLOAD_FOLDER="/home/lesf0/Dev/openwrt-build-scripts/dl"
CONFIG_DRIVER_11N_SUPPORT=y
CONFIG_GCC_VERSION="11.2.0"
CONFIG_HAS_BPF_TOOLCHAIN=y
CONFIG_IMAGEOPT=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
CONFIG_PACKAGE_dnsmasq-full=y
CONFIG_PACKAGE_dnsmasq_full_dhcp=y
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_dnsmasq_full_ipset=y
CONFIG_PACKAGE_dnsmasq_full_noid=y
CONFIG_PACKAGE_https-dns-proxy=y
CONFIG_PACKAGE_ip6tables-nft=y
CONFIG_PACKAGE_iptables-nft=y
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-ifb=y
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-ipset=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nf-ipt6=y
CONFIG_PACKAGE_kmod-nft-compat=y
CONFIG_PACKAGE_kmod-sched-bpf=y
CONFIG_PACKAGE_kmod-sched-cake=y
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_kmod-tcp-bbr=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-usb-printer=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_libbpf=y
CONFIG_PACKAGE_libcares=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libelf=y
CONFIG_PACKAGE_libev=y
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext-nft=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_libmbedtls=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libuci-lua=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-https-dns-proxy=y
CONFIG_PACKAGE_luci-app-p910nd=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-proto-wireguard=y
CONFIG_PACKAGE_luci-theme-openwrt-2020=y
CONFIG_PACKAGE_nano=y
CONFIG_PACKAGE_nanosplit=y
CONFIG_PACKAGE_p910nd=y
CONFIG_PACKAGE_qosify=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_tc-full=y
CONFIG_PACKAGE_tc-mod-iptables=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wireguard-tools=y
CONFIG_PACKAGE_xtables-nft=y
CONFIG_PACKAGE_zlib=y
CONFIG_USE_LLVM_HOST=y
CONFIG_VERSIONOPT=y
CONFIG_VERSION_BUG_URL=""
CONFIG_VERSION_CODE="2023.03.25.07.50.32"
CONFIG_VERSION_DIST="jankWRT"
CONFIG_VERSION_HOME_URL=""
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER=""
CONFIG_VERSION_MANUFACTURER_URL=""
CONFIG_VERSION_NUMBER="owrt-221fbfa2d8+config-bb5e13f"
CONFIG_VERSION_PRODUCT=""
CONFIG_VERSION_REPO="https://downloads.openwrt.org/releases/22.03.3"
CONFIG_VERSION_SUPPORT_URL=""
CONFIG_BINUTILS_VERSION="2.37"
CONFIG_BINUTILS_VERSION_2_37=y
CONFIG_DRIVER_11N_SUPPORT=y
# CONFIG_EXTRA_TARGET_ARCH is not set
# CONFIG_GCC_DEFAULT_PIE is not set
# CONFIG_GCC_DEFAULT_SSP is not set
CONFIG_GCC_VERSION="11.2.0"
# CONFIG_INSTALL_GFORTRAN is not set
# CONFIG_KERNEL_ARM_PMU is not set
# CONFIG_PACKAGE_dnsmasq_full_auth is not set
# CONFIG_PACKAGE_dnsmasq_full_conntrack is not set
# CONFIG_PACKAGE_dnsmasq_full_dnssec is not set
# CONFIG_PACKAGE_dnsmasq_full_tftp is not set
# CONFIG_PACKAGE_kmod-nf-conntrack-netlink is not set
# CONFIG_PACKAGE_libgmp is not set
# CONFIG_PACKAGE_libnetfilter-conntrack is not set
# CONFIG_PACKAGE_libnettle is not set
# CONFIG_PACKAGE_libnfnetlink is not set
# CONFIG_SJLJ_EXCEPTIONS is not set
# CONFIG_VERSION_CODE_FILENAMES is not set
# CONFIG_VERSION_FILENAMES is not set
# CONFIG_WOLFSSL_ASM_CAPABLE is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Djfe, 2023-03-25:
might be related to the wifi driver (ath10k-ct)
https://github.com/greearb/ath10k-ct/issues/131
https://github.com/greearb/ath10k-ct/issues/188

lesf0, 2023-03-29:
The issue seems to stop reproducing on the current trunk

closed 2023-03-29 by lesf0