openwrt/openwrt #12260: rfkill not accessible

oskarirauta, 2023-03-25:
### Describe the bug

/dev/rfkill is not accessible by hostapd/wpad

```
root@gateway:/dev# logread|grep rfkill
Sat Mar 25 16:24:21 2023 daemon.notice hostapd: rfkill: Cannot open RFKILL control device
```

if I do chmod 666 /dev/rfkill and restart wpad, this error doesn't appear.

### OpenWrt version

r22385-4dedcbdc80

### OpenWrt target/subtarget

x86/64

### Device

Cerberus Interceptor Q878GE

### Image kind

Self-built image

### Steps to reproduce

restart pad (full) or boot your system and examine log

### Actual behaviour

wpad/hostapd does not have sufficient permissions over /dev/rfkill

### Additional info

This patch might fix the issue, but haven't had a chance to test it so far; also, not so sure if it's the correct approach.. Should set /dev/rfkill permissions to everyone on dev creation...

```
--- a/net/rfkill/core.c
+++ b/net/rfkill/core.c
@@ -840,6 +840,7 @@
 	u32 state;
 	int error;
 
+	add_uevent_var(env, "DEVMODE=%#o", 0666);
 	error = add_uevent_var(env, "RFKILL_NAME=%s", rfkill->name);
 	if (error)
 		return error;
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-25:
File should be changed to be writable by group `network`
```diff
--- /etc/init.d/wpad
+++ /etc/init.d/wpad
-                chown network:network /var/run/hostapd
+               chown network:network /var/run/hostapd /dev/rfkill
```

oskarirauta, 2023-03-25:
Yes, that patch against kernel didn't do anything.