openwrt/openwrt #12263: Missing port and port mapping of wan on RT-AX1800S

BookerSoft, 2023-03-25:
### Describe the bug

The build installs and only shows three lan ports and one wan this device has four lan ports and fourth one maps to wan and the device’s wan port maps no where. Please help also 160mhz does not work 22.03.3

### OpenWrt version

22.03.3

### OpenWrt target/subtarget

Tamils/mt7621

### Device

RT-AX1800S using RT-AX53U image

### Image kind

Official downloaded image

### Steps to reproduce

Install image via mtd-write 
Configure network

### Actual behaviour

Only four Ethernet ports work all lan

### Expected behaviour

All Ethernet lan and wan port(s) working

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-26:
Both router specs say they do not support 160MHz channels
Port configurations are part of DTS file for particular device and version, after your report expect that to be on fast track. AX53 has four ports. Can you walk ports and tell the name of current WAN port?

BookerSoft, 2023-03-26:
I believe wan is mapping to either lan 1or 4 the left most port, the actual wan port does nothing.

brada4, 2023-03-26:
Its normal, the original device has only 4 ports. Wait for somebody to take on it.

BookerSoft, 2023-03-26:
How hard is that? What does it require?

brada4, 2023-03-27:
You can ask in forums for more assistance on porting.
find dts file for AX53, and add another network port either before or after existing and identify one coloured wan.
(and rebuilding/booting multiple times new kernels)

As for here - no change for existing device needed.

Djfe, 2023-03-29:
You install the image of the Asus RT-AX53U and then you wonder why it won't detect LAN4 😅
Support for RT-AX54 and thereby RT-AX1800S has been added recently to master.
You can get the image [here](https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ramips%2Fmt7621&id=asus_rt-ax54)
Try it out, it will probably work fine :)
If not, pls open a new bug about it. This issue can be closed.

Commit: https://github.com/openwrt/openwrt/commit/92276eef704af0d74d067676de7e8b9966de91ca