openwrt/openwrt #12271: GL-AR150: ethtool reports wrong link speed

Hirogen1, 2023-03-27:
### Describe the bug

ethtool and LUCI report a gigabit link although device has no gigabit port.

### OpenWrt version

22.03.3

### OpenWrt target/subtarget

ath79/generic

### Device

GL.iNet GL-AR150

### Image kind

Official downloaded image

### Steps to reproduce

connect device to WAN-Port
install package ethtool
"ethtool eth0"

### Actual behaviour

ethtool reports:

_Settings for eth0:
        Supported ports: [ TP MII ]
        Supported link modes:   1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: No
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 0
        Transceiver: external
        Auto-negotiation: on
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
        Link detected: yes_

### Expected behaviour

ethtool reports link speed 100Mbit/s or less

### Additional info

WAN and LAN port show the same behaviour.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-03-28:
There is MII port from SoC to switch ASIC, you have gigabit to switch that has other 100Mbps ports. 
[image]

brada4, 2023-03-28:
i.e your ethtool reports correct speed.

Hirogen1, 2023-04-10:
LUCI on the GL-AR150 shows only two network adapters: eth0 and eth1.

[image]

The status of eth1 in LUCI fits to the link status and speed of the WAN port but LUCI always reports 1000 full-duplex for eth0 - no matter if link on LAN port is up or not.
So effectivly with LUCI you cannot distinguish if LAN port is up or down.

Hirogen1, 2023-05-02:
There is no menue item "Network/Switch" on this device.

[image]

brada4, 2023-05-02:
It should be there because ....
> Ok, swconfig reports 100baseT.

i.e there is a switch with gigabit port to SoC and 100Mbit external ports (? and eth1 not connected anywhere?)

What do you have in /etc/config/network supposedly 3 sections - 1 switch and 2 switch_vlan ?

e.g  mine(not on your switch) - one port is tagged, so in OpenWRT 2 subnets are accessed via eth0.1 and eth0.2 (yours are different, certainly not 5 ports external)

```
config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '2 3 4 5 0t'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '1 0t'
```

In principle you should be able to re-create config sections using luci via wifi, but keep config backup and reset pencil ready - i could delete switch_vlan sections from text file, reboot router, and recreate vlan-s but your mileage may vary.